code-iai/iai_kinect2 #29: Build libfreenect2-fork without OpenGL support

masterhui, 2015-01-09:
Hello Thiemo,
I hope you can help me with this. Today I just switched from an older version (wiedemeyer/libfreenect2 branch:old_master commit:22eeda006eedc03f02815809bb2a7d4ce2351d1f: "implemented a better device selection [...]" from 9 Oct. 2014) to the most current one (wiedemeyer/libfreenect2 branch:iai_kinect2 commit:318b52be2353a752628ad4148f089d36fa43573b:"modifications for iai_kinect2").

Obviously, quite a few things have changed. The result is that I cannot get ./Protonect to run on my setup anymore. My setup is:
```
Distribution: Ubuntu 14.04.01 
Kernel: 3.13.0-43-generic
Hardware: Intel Core i7-3840QM CPU with integrated Intel HD Graphics IvyBridge GPU
OpenCL Platform: OpenCL 1.2 beignet 1.0
```

As this is a minimal Ubuntu server installation without x-server components, OpenGL is not available. With the old version (commit:22eeda0) this was no problem as the OpenClDepthPacketProcessor was used. 

However the latest version of wiedemeyer/libfreenect2 the situation is as follows:
* Branch master (commit 5466b57): libfreenect2.so compiles, when I run it I get: 
```
~/libs/libfreenect2/examples/protonect$ bin/Protonect 
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
[OpenGLContext] failed to initialize glew for the current context!
Segmentation fault
```
which is expected (due to non-working OpenGL).

* Branch iai_kinect2 (commit 318b52b): libfreenect2.so compiles, when I run it I get: 
```
~/libs/libfreenect2/examples/protonect/bin$ ./Protonect 
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
./Protonect: symbol lookup error: /home/ros/libs/libfreenect2/examples/protonect/lib/libfreenect2.so: undefined symbol: udev_new
```
This looks like two issues to me: (1) Broken OpenGL; (2) undefined symbol: udev_new

I pretty much spent all day trying to resolve this, but to no avail. Hence, my questions for you are:
1. How to compile libfreenect2 without OpenGL support?
2. Any idea how to resolve the undefined symbol error?

Thanks!

wiedemeyer, 2015-01-12:
I didn't changed anything related to protonect. If you want it to use OpenCL you should give an instance of `OpenCLPacketPipeline` as parameter to [openDevice](https://github.com/wiedemeyer/libfreenect2/blob/iai_kinect2/examples/protonect/Protonect.cpp#L61) in `protonect.cpp`.

But why do you need to run protonect anyway? Don't you use kinect2_bridge? It should use OpenCL by default.

masterhui, 2015-01-12:
I don't actually need to run protonect. But running kinect2_bridge, I get

```
process[kinect2_bridge-15]: started with pid [2449]
~/overlay_ws/devel/lib/kinect2_bridge/kinect2_bridge: symbol lookup error: ~/libs/libfreenect2/examples/protonect/lib/libfreenect2.so: undefined symbol: udev_new
[kinect2_bridge-15] process has died
```
As protonect used to work (and throws the same error using branch iai_kinect2), I thought it might be a good idea to get this working first before I try to get kinect2_bridge working again.

But you're right, my problem isn't the two libGL errors (running glxgears throws the same error, but still works). The reason why kinect2_bridge keeps crashing is the "libfreenect2.so: undefined symbol: udev_new" error. I also looked into this, but don't know how to resolve. I get:

```
~/libs/libfreenect2/examples/protonect/lib$ nm libfreenect2.so | grep udev                                                   
000000000004c8e4 t linux_udev_event_thread_main
000000000004cd8d t linux_udev_hotplug_poll
000000000004cc0a t linux_udev_scan_devices
000000000004c497 t linux_udev_start_event_monitor
000000000004c78d t linux_udev_stop_event_monitor
000000000040f0a0 d udev_control_pipe
000000000040f2b8 b udev_ctx
                 U udev_device_get_action
                 U udev_device_get_devnode
                 U udev_device_get_sysname
000000000004ca30 t udev_device_info
                 U udev_device_new_from_syspath
                 U udev_device_unref
                 U udev_enumerate_add_match_subsystem
                 U udev_enumerate_get_list_entry
                 U udev_enumerate_new
                 U udev_enumerate_scan_devices
                 U udev_enumerate_unref
000000000004cac0 t udev_hotplug_event
                 U udev_list_entry_get_name
                 U udev_list_entry_get_next
000000000040f2c0 b udev_monitor
                 U udev_monitor_enable_receiving
000000000040f09c d udev_monitor_fd
                 U udev_monitor_filter_add_match_subsystem_devtype
                 U udev_monitor_get_fd
                 U udev_monitor_new_from_netlink
                 U udev_monitor_receive_device
                 U udev_monitor_unref
                 U udev_new
                 U udev_unref
```
Any ideas how to further debug? Cheers

wiedemeyer, 2015-01-12:
you could try `ldd libfreenect2.so` to see if all libraries are found.
Did you run `depends/install_ubunutu.sh` after checking out the iai_kinect2 branch? Because it changes something in the libusb build. Maybe the following helps:
- clean the libfreenect2 build directory
- checkout iai_kinect2 branch
- install depends
- cmake / ccmake libfreenect2
- make && make install
- rebuild iai_kinect2
- try again

masterhui, 2015-01-12:
I rebuilt libfreenect2 and iai_kinect2 as suggested, but I'm still getting the same error. The output of ```ldd libfreenect2.so``` is:

```
~/libs/libfreenect2/examples/protonect/lib$ ldd libfreenect2.so
        linux-vdso.so.1 =>  (0x00007fff3f9af000)
        libopencv_core.so.2.4 => /opt/ros/hydro/lib/libopencv_core.so.2.4 (0x00007f4b51fc1000)
        libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007f4b51d66000)
        libGLEWmx.so.1.10 => /usr/lib/x86_64-linux-gnu/libGLEWmx.so.1.10 (0x00007f4b51ae8000)
        libglfw.so.3 => /home/ros/libs/libfreenect2/examples/protonect/lib/libglfw.so.3 (0x00007f4b518d5000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4b516b6000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f4b514ae000)
        libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007f4b51219000)
        libOpenCL.so.1 => /opt/intel/intel-opencl-1.2-4.6.0.92/opencl-1.2-4.6.0.92/lib64/libOpenCL.so.1 (0x00007f4b51011000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4b50d05000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4b509ff000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4b507e7000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4b50421000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f4b5280e000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f4b50208000)
        libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f4b4fed2000)
        libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f4b4fcc8000)
        libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f4b4fab8000)
        libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f4b4f8b1000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f4b4f687000)
        libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007f4b4f45d000)
        libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f4b4f24a000)
        libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f4b4f047000)
        libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f4b4ee41000)
        libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f4b4ec3e000)
        libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007f4b4ea27000)
        libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007f4b4e822000)
        libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007f4b4e61e000)
        libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007f4b4e41b000)
        libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007f4b4e215000)
        libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f4b4dff5000)
        libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007f4b4ddf3000)
        libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f4b4dbe7000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4b4d9e2000)
        libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f4b4d7d8000)
        libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4b4d5d3000)
        libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4b4d3cd000)

```
The only potential issues I note are:
* Should there be a reference to libusb?
* libopencv_core.so.2.4 points to the Hydro version of libopencv_core.so. Not sure why, but seems to be a different matter.

wiedemeyer, 2015-01-12:
libusb is build as a static library and included in  libfreenect2.so.
was hydro sourced while building libfreenect2? Why is hydro on your system? I thought hydro is not supported on 14.04. With indigo the opencv system library in /usr/lib/... is used.

wiedemeyer, 2015-01-12:
is libudev installed on you system?

masterhui, 2015-01-12:
Q1) Indigo was sourced when libfreenect2.so was built. Hydro is as a remnant: the current installation was originally 13.10/Hydro, I upgraded it last month to 14.04/Indigo. 

Q2) Yes, libudev is installed. See:

```
ros@autonomy:~/libs/libfreenect2/examples/protonect/lib$ dpkg -l *libudev*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                         Version             Architecture        Description
+++-============================-===================-===================-==============================================================
ii  libudev-dev                  204-5ubuntu20.9     amd64               libudev development files
ii  libudev1:amd64               204-5ubuntu20.9     amd64               libudev shared library
```
and
```
ros@autonomy:~/libs/libfreenect2/examples/protonect/lib$ find /usr/lib -iname "*libudev*" -exec ls -l {} \; 
-rw-r--r-- 1 root root 529 Nov 18 16:38 /usr/lib/x86_64-linux-gnu/pkgconfig/libudev.pc
lrwxrwxrwx 1 root root 38 Nov 18 16:39 /usr/lib/x86_64-linux-gnu/libudev.so -> /lib/x86_64-linux-gnu/libudev.so.1.3.5
```

wiedemeyer, 2015-01-12:
If the error also appears on the master branch (which is the same as the original libfreenect2 master), then you could write an issue on the libfreenect2 github from OpenKinect. Maybe there is someone who has a solution.

masterhui, 2015-01-12:
If I switch to branch master in libfreenect2 and make/make install, I get:

```
# rosrun kinect2_bridge kinect2_bridge                                                                             
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
/home/ros/overlay_ws/devel/lib/kinect2_bridge/kinect2_bridge: symbol lookup error: /home/ros/overlay_ws/devel/lib/kinect2_bridge/kinect2_bridge: undefined symbol: _ZN12libfreenect220OpenCLPacketPipelineC1Ei
```

wiedemeyer, 2015-01-12:
You need to build it with opencl enabled and in iai_kinect2 you have to be on the master branch.
But you can simply try to run protonect. If it fails then it is clearly an issue of libfreenect2 (or your system configuration).

masterhui, 2015-01-12:
If I run protonect on libfreenect2/master branch I get:
```
ros@autonomy:~/libs/libfreenect2/examples/protonect/bin$ ./Protonect 
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
[OpenGLContext] failed to initialize glew for the current context!
Segmentation fault
```
So no ```undefined symbol: udev_new``` error in this case. I am out of ideas at this point, reinstalling the whole system (which is based on minimal Ubuntu from scratch) would be very time consuming, which I would like to avoid.

wiedemeyer, 2015-01-12:
change [protonect.cpp line 61](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/Protonect.cpp#L61) to use opencl. enable opencl. try if it fails.
something like:
```
libfreenect2::Freenect2Device *dev = freenect2.openDefaultDevice(new libfreenect2::OpenCLPacketPipeline());
```

masterhui, 2015-01-12:
If I use OpenCLPacketPipeline in Protonect like oyu suggested, I don't get the error:

```
root@autonomy:/home/ros/libs/libfreenect2/examples/protonect/bin# ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 13 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:1 with serial 500637743142
[Freenect2Impl] found 1 devices
[HEIKO] num_devices = 1
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 05 14 01 47 4b 53 39 31 36 2e 31    13......GKS916.1
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                X...........

platforms.size() = 2
[OpenCLDepthPacketProcessor::selectDevice] found platform: Intel Gen OCL Driver vendor: Intel
***** devices.size() = 1
HERE
[OpenCLDepthPacketProcessor::selectDevice] found device: Intel(R) HD Graphics IvyBridge M GT2 vendor: Intel
platforms.size() = 2
[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  0d 26 00 00                                        .&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 500637743142
device firmware: 2.3.3912.0.7
[DepthPacketStreamParser::handleNewData] not all subsequences received 0
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 512
[DepthPacketStreamParser::handleNewData] not all subsequences received 1022
X11 connection rejected because of wrong authentication.
(rgb:2928): Gtk-WARNING **: cannot open display: localhost:10.0
```
It only eventually errors do to no X11 display being available... But the ```undefined symbol: udev_new``` error is gone.

masterhui, 2015-01-12:
Please note the update in my previous comment - I pasted from the wrong terminal. Sorry.

masterhui, 2015-01-13:
What's the conclusion? Is there anything else you can think of for me to try and fix the error on the iai_kinect2 branch of libfreenect2?

wiedemeyer, 2015-01-13:
But on your post on "OpenCLPacketPipeline with Protonect" it has an error getting the opencl devices. Maybe you should first try to solve that issue. maybe they are related.

Or you could try to force it to link to libudev by adding udev to libfreenect2 CMakeLists.txt:
```
TARGET_LINK_LIBRARIES(freenect2 ${LIBRARIES} udev)
```
Yesterday a lot of changes have been pushed to the repos. You could check them out.
But that are just random guesses right now.

masterhui, 2015-01-13:
Sorry, I forgot to mention that the error in the previous post was due to executing without root permission (remember, beignet dosen't play nice if no local display is attached). I tried again as root, and the clGetDeviceIDs disappeared as expected.

With that sorted, I went on to try out your other suggestion: I added udev to the TARGET_LINK_LIBRARIES line - and voila - everything works (libfreenect2/branch:iai_kinect2 and iai_bridge/branch:master, both up-to-date as of today).

Thanks, great job for helping me debug this issue. Please go ahead and close if you wish.

masterhui, 2015-01-13:
FYI
Renamed this issue to "libfreenect2-fork: undefined symbol udev_new"

wiedemeyer, 2015-01-13:
Great!

closed 2015-01-13 by wiedemeyer