code-iai/iai_kinect2 #30: kinect2_bridge SegFault issue

gypsytearss, 2015-01-11:
Hi, I'm running into a Segmentation Fault when trying to launch kinect2_bridge via roslaunch on Ubuntu 14.04 using OpenCLPacketPipeline with Beignet installed.

Trying to follow the known working configuration for Intel GPU (IvyBridge), my system is currently:
* Ubuntu 14.04 (Linux version 3.13.0-43-generic )
* Beignet v1.0

Building the libfreenect2 library was successful and Protonect example works fine with OpenCL, and building the Beignet v1.0 package also went relatively smoothly. 

Though after building the iai_kinect2 library and attempting to run 'roslaunch kinect2_bridge kinect2_bridge.launch' I get the following result:

```
SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.9

NODES
  /
    kinect2_bridge (kinect2_bridge/kinect2_bridge)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2_bridge-1]: started with pid [3771]
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 508862642542
[Freenect2Impl] found 1 devices
[kinect2_bridge-1] process has died [pid 3771, exit code -11, cmd /home/colin/catkin_ws/src/iai_kinect2/kinect2_bridge/devel/lib/kinect2_bridge/kinect2_bridge __name:=kinect2_bridge __log:=/home/colin/.ros/log/a893f52a-99c9-11e4-befa-542696d42fef/kinect2_bridge-1.log].
log file: /home/colin/.ros/log/a893f52a-99c9-11e4-befa-542696d42fef/kinect2_bridge-1*.log
[kinect2_bridge-1] restarting process
```

Where the process just continues to die and and restart until Ctrl+C.

Running kinect2_bridge through gdb produces the following:

```
(gdb) run
Starting program: /home/colin/catkin_ws/src/iai_kinect2/kinect2_bridge/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe19e2700 (LWP 3365)]
[New Thread 0x7fffe0aff700 (LWP 3366)]
[New Thread 0x7fffdbfff700 (LWP 3368)]
[New Thread 0x7fffdb7fe700 (LWP 3369)]
[New Thread 0x7fffdaffd700 (LWP 3370)]
[New Thread 0x7fffda7fc700 (LWP 3375)]
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 508862642542
[Freenect2Impl] found 1 devices

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff5fb0694 in libfreenect2::Freenect2DeviceImpl::Freenect2DeviceImpl (this=0x6ce350, context=<optimized out>, pipeline=
    0x7ffff7f9a890, usb_device=<optimized out>, usb_device_handle=<optimized out>, serial=...)
    at /home/colin/libfreenect2/examples/protonect/src/libfreenect2.cpp:329
329	  rgb_transfer_pool_.setCallback(pipeline_->getRgbPacketParser());
(gdb) bt full
#0  0x00007ffff5fb0694 in libfreenect2::Freenect2DeviceImpl::Freenect2DeviceImpl (this=0x6ce350, context=<optimized out>, 
    pipeline=0x7ffff7f9a890, usb_device=<optimized out>, usb_device_handle=<optimized out>, serial=...)
    at /home/colin/libfreenect2/examples/protonect/src/libfreenect2.cpp:329
No locals.
#1  0x00007ffff5fb094f in libfreenect2::Freenect2::openDevice (this=0x7fffffffd9a8, idx=0, pipeline=0x7ffff7f9a890, 
    attempting_reset=<optimized out>) at /home/colin/libfreenect2/examples/protonect/src/libfreenect2.cpp:706
        device = 0x0
        dev = @0x683880: {dev = 0x6c8dc0, serial = {static npos = <optimized out>, 
            _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
              _M_p = 0x6c91a8 "508862642542"}}}
        dev_handle = 0x6dbbb0
        num_devices = <optimized out>
        r = <optimized out>
#2  0x000000000043402e in Kinect2Bridge::init(std::string const&) ()
No symbol table info available.
#3  0x0000000000430e1f in main ()
No symbol table info available.
```

Just wondering if anyone has run into similar problems in their installations (could be missing something silly.. I'm new at this :)) -- Many thanks for any help!

wiedemeyer, 2015-01-12:
Did you changed the code of `protonect.cpp` to use OpenCL? Because by default it will use openGL.
I would recommend to build libfreenect2 and iai_kinect2 in debug and then start it in gdb.

gypsytearss, 2015-01-21:
Hey Thiemo, thanks a ton for taking the time to respond and advise.. apologies that I've been working on some other things the last few days and haven't had time to debug. Will try to get to it tonight.

In the meantime & to answer your question, I have changed `protonect.cpp` some in conjunction with [this issue and potential solution](https://github.com/OpenKinect/libfreenect2/issues/109) , as diff shows below, but the only place I switched to OpenCL is in `packet_pipeline.h`

protonect.cpp diff:
```
diff --git a/examples/protonect/Protonect.cpp b/examples/protonect/Protonect.cpp
index 01150e8..7f94c6d 100644
--- a/examples/protonect/Protonect.cpp
+++ b/examples/protonect/Protonect.cpp
@@ -36,12 +36,12 @@
 #include <libfreenect2/frame_listener_impl.h>
 #include <libfreenect2/threading.h>
 
-bool shutdown = false;
-
-void sigint_handler(int s)
-{
-  shutdown = true;
-}
+//bool shutdown = false;
+//
+//void sigint_handler(int s)
+//{
+// shutdown = true;
+//}
 
 int main(int argc, char *argv[])
 {
@@ -66,8 +66,8 @@ int main(int argc, char *argv[])
     return -1;
   }
 
-  signal(SIGINT,sigint_handler);
-  shutdown = false;
+//  signal(SIGINT,sigint_handler);
+//  shutdown = false;
 
   libfreenect2::SyncMultiFrameListener listener(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | li
   libfreenect2::FrameMap frames;
@@ -79,7 +79,8 @@ int main(int argc, char *argv[])
   std::cout << "device serial: " << dev->getSerialNumber() << std::endl;
   std::cout << "device firmware: " << dev->getFirmwareVersion() << std::endl;
 
-  while(!shutdown)
+//  while(!shutdown)
+  while(1)
   {
     listener.waitForNewFrame(frames);
     libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];
@@ -91,7 +92,7 @@ int main(int argc, char *argv[])
     cv::imshow("depth", cv::Mat(depth->height, depth->width, CV_32FC1, depth->data) / 4500.0f);
 
     int key = cv::waitKey(1);
-    shutdown = shutdown || (key > 0 && ((key & 0xFF) == 27)); // shutdown on escape
+//    shutdown = shutdown || (key > 0 && ((key & 0xFF) == 27)); // shutdown on escape
 
     listener.release(frames);
     //libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(100));
```

wiedemeyer, 2015-02-13:
@gypsytearss: do you have an update on this one, or can it be closed?

closed 2015-02-18 by wiedemeyer

vishu2287, 2015-09-29:
I am able to run ./Protonect and can view color, ir and depth images with out any error.
When I am trying to run kinect2_bridge from the root, it asks for clients. If I provide kinect2/sd/image_color_rect, it produces the messages but if I provide other topics like kinect2/qhd/image_color, the following error genetares

root@tegra-ubuntu:~/catkin_ws/devel# rosrun kinect2_bridge kinect2_bridge
parameter:
base_name: kinect2
sensor: 
fps_limit: -1
calib_path: /home/ubuntu/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
use_png: false
jpeg_quality: 90
png_level: 1
depth_method: opengl
depth_device: -1
reg_method: default
reg_devive: -1
max_depth: 12
min_depth: 0.1
queue_size: 2
bilateral_filter: true
edge_aware_filter: true
publish_tf: false
base_name_tf: kinect2
worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:5 with serial 017275643547
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
0: 017275643547 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000: 00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41 ............C..A
0x0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x0020: 0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00 .!3U... ........
0x0030: 00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00 .......... .....
0x0040: 31 33 00 00 00 04 0a 0d 47 4d 42 30 31 33 2e 31 13......GMB013.1
0x0050: 58 00 00 00 00 00 00 00 07 00 00 00 X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000: 23 26 00 00 #&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000: 23 26 00 00 #&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[... 328 lines cut ...]
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~127.059Hz (7.87038ms) publishing rate: ~14.6242Hz
Segmentation fault

I am using Jetson TK1 board .

cpu ang gpu performance were altered with maximum efficiency

Why lots of "skipping rgb packet!" and " [TransferPool::submit] failed to submit transfer " ?

why not providing depth data?
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz

Why segmentation fault after producing one message to the topics?