code-iai/iai_kinect2 #33: ROS dependency issues

christiankerl, 2015-01-13:
to compile iai_kinect2 with catkin I had to install ros-indigo-image-transport-plugins and pcl by hand. I thought they should be installed by `rosdep install iai_kinect2`. Am I wrong or are there some dependency entries in the package.xml missing? I'm not yet used to the new ROS dependency management / build system.

wiedemeyer, 2015-01-13:
I'm not sure. It has `compressed_image_transport` and `compressed_depth_image_transport` in the `package.xml` these are the components of `image_transport_plugins` that it uses. Maybe it works if `image_transport_plugins` is in the dependencies instead. Could you try it?
And in ROS Indigo PCL is not part of ROS anymore. ROS uses the PCL system library. PCL is no ROS package, so I didn't mentioned it in the package.xml.

wiedemeyer, 2015-01-13:
I think in general `ros-indigo-desktop-full` is installed and that one comes with `libpcl` and ros-`indigo-image-transport-plugins`.

christiankerl, 2015-01-13:
I've only installed ros-indigo-desktop 

running `rosdep install -r --from-paths .` in the iai_kinect2 top-level folder installed the `compressed_image_transport`  and `compressed_depth_image_transport` dependencies

after all I'm not quite sure what the right series of commands is after cloning iai_kinect2 into my catkin_ws/src/ folder. maybe you could add this to the README?

wiedemeyer, 2015-01-13:
I have to look into it and add information the the README.
But the easiest solution for users would be to create a debian package.

paulbovbel, 2015-01-13:
Before releasing debs, you'd still have to add libpcl-deb to the package.xml dependencies. And of course, libfreenect2 would need to be released as well, either into ubuntu or packages.ros.org. @christiankerl, is a release on the horizon at all?

wiedemeyer, 2015-01-14:
Hi,
I just added the correct dependencies to the `package.xml` files and added detail instructions to the README. You can see these in the devel branch. I tested them on a virtual machine with a ros-indigo-ros-base installation.

wiedemeyer, 2015-02-13:
I think the ros dependency issue is fixed.

closed 2015-02-13 by wiedemeyer