code-iai/iai_kinect2 #35: Can't build wiedemeyer fork of libfreenect2

brhemes, 2015-01-15:
Using the linked fork from the readme or making the modifications manually I get a bunch of undefined referenced to udev when building Protonect.cpp.  The errors seem to originate from "Change line 102 from 'usb-1.0' to 'usb-1.0.a'".  If I leave this as usb-1.0 it builds and runs.  However, when changed to usb-1.0.a I get the output below.  Thanks, Brett.

...
Scanning dependencies of target Protonect
[ 97%] Building CXX object CMakeFiles/Protonect.dir/Protonect.cpp.o
Linking CXX executable bin/Protonect
lib/libfreenect2.so: undefined reference to `udev_monitor_unref'
lib/libfreenect2.so: undefined reference to `udev_enumerate_scan_devices'
lib/libfreenect2.so: undefined reference to `udev_new'
...
lib/libfreenect2.so: undefined reference to `udev_enumerate_add_match_subsystem'
collect2: ld returned 1 exit status
make[2]: *** [bin/Protonect] Error 1
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
make: *** [all] Error 2

wiedemeyer, 2015-01-16:
Hi,
this is the same error than issue #29. The solution there should also work in your case.

brhemes, 2015-01-16:
As suggested in issue issue #29, added udev to libfreenect2 CMakeLists.txt:
```
TARGET_LINK_LIBRARIES(freenect2 ${LIBRARIES} udev)
```
Protonect now compiles.

closed 2015-01-16 by brhemes