code-iai/iai_kinect2 #36: clGetPlatformIDs(-1001) error after a driver update

bassofil, 2015-01-16:
I re-installed the Nvidia drivers in my laptop and then `kinect2_bridge` started not working anymore.
I tried to re-install everything from stratch but nothing.
The fact that astonish me is that the example program provided by `freenect2` (`freeenect2/examples/protonec/bin/Protonec`) is perfectly working.
Any suggestion?

wiedemeyer, 2015-01-16:
Hi,
Protonect is by default not using OpenCL but OpenGL. So if you change it to use OpenCL you will get the same error.
```
libfreenect2::Freenect2Device *dev = freenect2.openDefaultDevice(new libfreenect2::OpenCLPacketPipeline());
```
I had the same issue today. I read in another issue that starting `sudo ./Protonect` (with OpenCL pipeline) once solved it and afterwards it is possible to start it without root. It also solved the problem for me, no clue why.

Other than that you could check if a file `/etc/OpenCL/vendors/nvidia.icd` exists and if the library mentioned there is found somewhere in `/usr/lib`.

bassofil, 2015-01-19:
Thanks very much for the information.
I tried to run the `kinect2_bridge` node once with `sudo` and now, magically, everything seem to work.
That is, I tried to run
```
roscd
cd lib/kinect2_bridge/
sudo ./kinect2_bridge
```
but, as expected, the node could not connect to the ros master, dying immediately.
Anyway, now lunching `roslaunch kinect2_bridge kinec2_bridge.launch` works.

closed 2015-01-19 by wiedemeyer