code-iai/iai_kinect2 #39: Trouble building with Ubuntu 12.04 and ROS Hydro (cxx11 support and boost issues)

brhemes, 2015-01-16:
I am not sure if this is actually an issue or just a result of my Linux ineptitude so please be gentle :)
Here is the progress I have made...

After some minor hiccups I was able to get a build going that made it all the way to kinect2_bridge before failing.  The error I got was
```
[100%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
/home/hemes/ros_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘void Kinect2Bridge::run()’:
/home/hemes/ros_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:372:62: error: no matching function for call to ‘libfreenect2::SyncMultiFrameListener::waitForNewFrame(libfreenect2::FrameMap&, int)’
/home/hemes/ros_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:372:62: note: candidate is:
/usr/local/include/libfreenect2/frame_listener_impl.h:54:8: note: void libfreenect2::SyncMultiFrameListener::waitForNewFrame(libfreenect2::FrameMap&)
/usr/local/include/libfreenect2/frame_listener_impl.h:54:8: note:   candidate expects 1 argument, 2 provided
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed
```

After digging some I realized that libfreenect2 was compiling without cxx11 support and thus the member function causing the error was never defined.  See libfreenect2/frame_listener_impl.h:50
```
...
	#ifdef LIBFREENECT2_THREADING_STDLIB
	  bool waitForNewFrame(FrameMap &frame, int milliseconds);
...
```

Turns out that the gcc that ships with 12.04 (Version 4.6) does not have cxx11 support so I tried upgrading to 4.8.  This solved my cxx11 support issue and I was able to build libfreenect2 again and got rid of the above error.

However, my shiny new version of gcc now causes some issues with the version of boost in 12.04 (Version 1.48.0.2).  I now get a plethora of errors regarding use of deleted functions.  Some representative output is as follows:
```
[ 80%] Built target depth_registration
[100%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
In file included from /home/hemes/ros_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:37:0:
/opt/ros/hydro/include/cv_bridge/cv_bridge.h: In constructor ‘cv_bridge::CvImage::CvImage(const Header&, const string&, const cv::Mat&)’:
/opt/ros/hydro/include/cv_bridge/cv_bridge.h:77:54: error: use of deleted function ‘std_msgs::Header_<std::allocator<void> >::Header_(const std_msgs::Header_<std::allocator<void> >&)’
     : header(header), encoding(encoding), image(image)
                                                      ^
In file included from /opt/ros/hydro/include/sensor_msgs/CameraInfo.h:51:0,
                 from /home/hemes/ros_ws/src/iai_kinect2/kinect2_calibration/src/kinect2_calibration.cpp:34:
/opt/ros/hydro/include/std_msgs/Header.h:55:8: note: ‘std_msgs::Header_<std::allocator<void> >::Header_(const std_msgs::Header_<std::allocator<void> >&)’ is implicitly deleted because the default definition would be ill-formed:
 struct Header_
        ^
/opt/ros/hydro/include/std_msgs/Header.h:55:8: error: use of deleted function ‘boost::shared_ptr<std::map<std::basic_string<char>, std::basic_string<char> > >::shared_ptr(const boost::shared_ptr<std::map<std::basic_string<char>, std::basic_string<char> > >&)’
...
```
I saw references that this issue was addressed in revision 1.48.0.3.  If I try to update boost, however, apt-get wants to remove all of ROS Hydro...  is there any hope besides building ROS from source with newer versions of gcc/boost?

I am in this predicament because ROS-Industrial is still in Hydro.  I believe they have an Indigo release scheduled for some time before April of this year but I would like to get this going before that.

wiedemeyer, 2015-01-19:
Have you tried gcc 4.7? I think I used that one on Hydro.
If it is a dedicated machine that runs the Kinect 2, you could try to run Indigo.

wiedemeyer, 2015-01-19:
With commit b58979eb7bade5836f9826f64fd665df35cc54d2 I added code that should make it possible to run `kinect2_bridge` even if libfreenect2 is not compiled with CXX11. So it should work with gcc 4.6.

brhemes, 2015-01-19:
I tried with gcc 4.7 and am getting the same boost errors.  I can try sans CXX11 support and see what I get but I am thinking I might also just make the jump to Indigo at this point and possibly try to use the indigo development branch for all of the ROS Industrial stuff... might be esasier that way and then I'll be totally ready for when ROS-I releases in Indigo.

brhemes, 2015-01-19:
OK, so I rebuilt everything using gcc 4.6 using your modified fork of libfreenect2 (still have to make the revision in issue #29 to libfreenect2 CMakeLists.txt) and the devel branch of iai_kinect2.  Everything builds but FYI the output of `rosdep install -r --from-paths .` is now
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
registration_viewer: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_bridge: Cannot locate rosdep definition for [depth_registration]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
The protonect executable runs but when I run `rosrun kinect2_bridge kinect2_bridge` I get a segfault with the following output:
```
hemes@jake:~/ros_ws/src/iai_kinect2$ rosrun kinect2_bridge kinect2_bridge 
[DepthRegistration::New] OpenCL registration method not available!
[DepthRegistration::New] OpenCL registration method not available!
[OpenGLContext] failed to initialize glew for the current context!
Segmentation fault (core dumped)
```

Here is the backtrace:
```
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007ffff6264e1b in bindToUnit (unit=33984, this=<optimized out>)
    at /home/hemes/src/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:216
#2  allocate (new_height=4240, new_width=352, this=0x67ede8)
    at /home/hemes/src/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:228
#3  libfreenect2::OpenGLDepthPacketProcessorImpl::initialize (this=0x67ec90)
    at /home/hemes/src/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:375
#4  0x00007ffff62638b9 in libfreenect2::OpenGLDepthPacketProcessor::OpenGLDepthPacketProcessor (this=0x689a80, parent_opengl_context_ptr=<optimized out>, 
    debug=false)
    at /home/hemes/src/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:683
#5  0x00007ffff62527cc in libfreenect2::OpenGLPacketPipeline::createDepthPacketProcessor (this=0x679ea0)
    at /home/hemes/src/libfreenect2/examples/protonect/src/packet_pipeline.cpp:108
#6  0x00007ffff62528f1 in libfreenect2::BasePacketPipeline::initialize (
    this=0x679ea0)
    at /home/hemes/src/libfreenect2/examples/protonect/src/packet_pipeline.cpp:43
#7  0x00007ffff6252aaf in libfreenect2::OpenGLPacketPipeline::OpenGLPacketPipeline (this=0x679ea0, debug=<optimized out>)
    at /home/hemes/src/libfreenect2/examples/protonect/src/packet_pipeline.cpp:101
#8  0x0000000000410551 in Kinect2Bridge::Kinect2Bridge(double, bool, int, int)
    ()
#9  0x0000000000407a13 in main ()
```

I am going to start working on an Indigo build but will keep my Hydro install around in case you care to see this through...

Thanks,
Brett

brhemes, 2015-01-19:
I am getting this same last error with my Indigo build:
```
hemes@cake:~/ros_ws/build/iai_kinect2$ rosrun kinect2_bridge kinect2_bridge
[DepthRegistration::New] OpenCL registration method not available!
[DepthRegistration::New] OpenCL registration method not available!
[OpenGLContext] failed to initialize glew for the current context!
Segmentation fault (core dumped)
```

It seems that the OpenCL depth registraion method is getting called despite me haveing Eigen.  The catkin_make output looks OK...
```
hemes@cake:~/ros_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/hemes/ros_ws
Source space: /home/hemes/ros_ws/src
Build space: /home/hemes/ros_ws/build
Devel space: /home/hemes/ros_ws/devel
Install space: /home/hemes/ros_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/hemes/ros_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/hemes/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/hemes/ros_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/hemes/ros_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/hemes/ros_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.11
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - depth_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - registration_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'depth_registration'
-- ==> add_subdirectory(iai_kinect2/depth_registration)
-- Eigen found (include: /usr/include/eigen3)
-- Could NOT find OpenCL (missing:  OPENCL_LIBRARIES OPENCL_INCLUDE_DIRS) 
-- CPU based depth registration enabled
-- OpenCL based depth registration disabled
-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'kinect2_calibration'
-- ==> add_subdirectory(iai_kinect2/kinect2_calibration)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'registration_viewer'
-- ==> add_subdirectory(iai_kinect2/registration_viewer)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
[... 61 lines cut ...]
/usr/include/eigen3/Eigen/src/Core/Assign.h:520:123:   required from ‘static Derived& Eigen::internal::assign_selector<Derived, OtherDerived, false, false>::run(Derived&, const OtherDerived&) [with Derived = Eigen::Matrix<double, 4, 1>; OtherDerived = Eigen::Matrix<double, 4, 1>]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:618:105:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set_noalias(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<double, 4, 1>; Derived = Eigen::Matrix<double, 4, 1>]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:288:31:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>&) [with _Scalar = double; int _Rows = 4; int _Cols = 1; int _Options = 0; int _MaxRows = 4; int _MaxCols = 1]’
/home/hemes/ros_ws/src/iai_kinect2/depth_registration/src/depth_registration_cpu.cpp:155:39:   required from here
/usr/include/eigen3/Eigen/src/Core/Assign.h:49:19: warning: logical ‘and’ applied to non-boolean constant [-Wlogical-op]
Linking CXX shared library /home/hemes/ros_ws/devel/lib/libdepth_registration.so
[ 80%] Built target depth_registration
Scanning dependencies of target kinect2_bridge
[100%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
In file included from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:52:0,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/hemes/ros_ws/src/iai_kinect2/registration_viewer/src/viewer.cpp:32:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
/usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (nb_points_ == -1)
                              ^
Linking CXX executable /home/hemes/ros_ws/devel/lib/kinect2_bridge/kinect2_bridge
Linking CXX executable /home/hemes/ros_ws/devel/lib/kinect2_calibration/kinect2_calibration
[100%] Built target kinect2_bridge
[100%] Built target kinect2_calibration
Linking CXX executable /home/hemes/ros_ws/devel/lib/registration_viewer/viewer
[100%] Built target viewer
```

I am out of time for today, otherwise I would have dug a bit more.  Hope this at least gives you an idea of what is going wrong.

wiedemeyer, 2015-01-20:
The segfault comes from the OpenGL processor, but in `Protonect` it is working. I will try it on my system later.
And the other problem is that the OpenCL registration is used even if it is not enabled in `depth_registration`. I will look at this too.

wiedemeyer, 2015-01-20:
I found out why `glewInit()` is not initializing. `glfwInit()` has to be called before `glewInit()`. This is done in `Protonect` but not in libfreenect2.
The depth_registration problem should be fixed with 999b30cce61d1ed753228bee1b5ac75b61f1fc92.
@christiankerl: should is be like this, that the caller has to execute `glfwInit()` before using libfreenect2, or should the opengl_depth_packet_processor / pipeline instantiation be responsible for this?

brhemes, 2015-01-20:
Indeed 999b30cce61d1ed753228bee1b5ac75b61f1fc92 fixes the depth_registration problem.

As for the glfwInit issue, I went ahead and added a `glfwInit()` call in kinect2_bridge.cpp (right before the first libfreenect2 calls) just to try and test the original Hydro issues I was having and everything ended up working.  I am able to run the bridge, viewer, and calibration in both Hydro and Indigo.

So in conclusion it seems b58979eb7bade5836f9826f64fd665df35cc54d2 solves the original  12.04+Hydro / no CXX11 issue when the distro-default gcc 4.6 is used.

Feel free to close or rename this issue however you see fit depending on how you end up handling the remaining glfwInit() issue.

Thanks so much for your work; this package will be very useful for my purposes.
Brett

wiedemeyer, 2015-01-20:
In the devel branch in commit efd26792504ffe88aae8f7f6d53698e8b4af5a8f I added `glfwInit()` to kinect2_bridge.

closed 2015-01-21 by wiedemeyer