code-iai/iai_kinect2 #41: OS X C++11 support detection fails

jmtatsch, 2015-01-23:
```
ERRORThe compiler /usr/bin/c++ has no C++11 support. Please use a different C++ compiler.
```
In my case  /usr/bin/c++ is:

```
Apple LLVM version 6.0 (clang-600.0.56) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin14.0.0
Thread model: posix
```

Later in the compilation run I get a ton of issues but i guess those are not the underlying issue.

wiedemeyer, 2015-01-23:
You could look into the cmake error log in `catkin_ws/build/CMakeFiles/CMakeError.log` to find out what went wrong.
The CMakeLists.txt files are using a macro to check for c++11 support:
```
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
```

jmtatsch, 2015-01-23:
Funny, the check for c++11 works once I comment out 

```# Unused warnings
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wuninitialized -Winit-self -Wunused-function -Wunused-label -Wunused-variable -Wunused-but-set-variable -Wunused-but-set-parameter")
# Additional warnings
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Warray-bounds -Wtype-limits -Wreturn-type -Wsequence-point -Wparentheses -Wmissing-braces -Wchar-subscripts -Wswitch -Wwrite-strings -Wenum-compare -Wempty-body -Wlogical-op")```

Apparently clang now treats the warnings that some flags are not supported  as errors?

```warning: unknown warning option '-Wunused-but-set-variable'; did you mean '-Wunused-const-variable'? [-Wunknown-warning-option]
warning: unknown warning option '-Wunused-but-set-parameter'; did you mean '-Wunused-parameter'? [-Wunknown-warning-option]
warning: unknown warning option '-Wlogical-op'; did you mean '-Wlong-long'? [-Wunknown-warning-option]```

However I still have trouble with this error:

```
/Users/tatsch/ros_catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:372:58: error: too many arguments to function call, expected single argument
      'frame', have 2 arguments
      bool newFrames = listener->waitForNewFrame(frames, 1000);
                       ~~~~~~~~~~~~~~~~~~~~~~~~~         ^~~~
/usr/local/include/libfreenect2/frame_listener_impl.h:54:3: note: 'waitForNewFrame' declared here
  void waitForNewFrame(FrameMap &frame);
  ^
3 warnings and 1 error generated.

```

wiedemeyer, 2015-01-23:
That is the same error as in issue #39. Should be fixed in devel branch.
libfreenect2 is build without C++11, so it uses tinythread. But with tinythread there is no non blocking implementation of waitForNewFrame. So you could either use devel or compile libfreenect2 with C++11.

jmtatsch, 2015-01-23:
Using devel worked, thank you.

closed 2015-02-18 by wiedemeyer