code-iai/iai_kinect2 #42: Reducing CPU load in Ubuntu 12.04 with ROS hydro

avidbot1, 2015-01-23:
Hi,

I've built this in Ubuntu 12.04 for ROS hydro. I've got an intel core i7-4790 and an nvidia GeForce GTX 745. When I run kinect2_bridge it only seems to be detecting the cpu.

    [DepthRegistration::New] Using OpenCL registration method!
    [DepthRegistration::New] Using OpenCL registration method!
    [Freenect2Impl] enumerating devices...
    [Freenect2Impl] 16 usb devices connected
    [Freenect2Impl] found valid Kinect v2 @4:8 with serial 500902543042
    [Freenect2Impl] found valid Kinect v2 @4:10 with serial 503614242542
    [Freenect2Impl] found 2 devices
    Kinect2 devices found: 
      0: 500902543042 (selected)
      1: 503614242542
    [Freenect2DeviceImpl] opening...
    [Freenect2DeviceImpl] opened

    starting kinect2

    [Freenect2DeviceImpl] starting...
    [Freenect2DeviceImpl] ReadData0x14 response
    92 bytes of raw data
    0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
    0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
    0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    .!3U ...........
    0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
    0x0040:  31 33 00 00 00 06 10 08 47 4d 42 30 31 34 2e 31    13......GMB014.1
    0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                X...........

    [OpenCLDepthPacketProcessor::listDevice]  devices:
      0: Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz (CPU)[Intel(R) Corporation]
    [OpenCLDepthPacketProcessor::init]  selected device: Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz 
    (CPU)[Intel(R) Corporation]

Here's what I'm getting as my cpu usage.

[image]

My goal is to get 3 Kinect2 sensors running simultaneously on an intel core i5. Is there a way to reduce CPU load while staying with Ubuntu 12.04 and ROS hydro?

Andrew

wiedemeyer, 2015-01-25:
Well, you could use your nvidia GPU for OpenCL. If the GPU is not detected, then the OpenCL drivers for the GPU are not installed correctly.

avidbot1, 2015-01-27:
Ok, I was running into [Issue 13](/code-iai/iai_kinect2/issues/13) and you suggested installing ocl-icd-opencl-dev for Nvidia instead of nvidia-opencl-dev, but that package isn't available in 12.04. I got past the issue by installing [Intel CPU OpenCL](https://software.intel.com/en-us/intel-opencl) so it makes sense that it's using the CPU and not the GPU. At the end of the day it's better that I'm not using the GPU as the application I'm designing for doesn't have one. I don't need RGB data and I don't need to sample as quickly so I'll try cutting out the RGB data and sampling at a lower frequency and see what that does for me.

wiedemeyer, 2015-01-27:
If you just need the depth, you could change the code to just receive those frames. That would reduce the CPU load by not running libturbojpeg to decompress the color images. But the depth processing is much more demanding, so a GPU would help, either a dedicated or an integrated.
But the dependencies for the package ocl-icd-opencl-dev from 14.04 are also met in 12.04. We have a 12.04 with an nvidia, where opencl is working just fine. The following packages are installed `ocl-icd-libopencl1`, `ocl-icd-opencl-dev`, `opencl-headers` and `nvidia-opencl-icd-340`.

closed 2015-01-29 by avidbot1