code-iai/iai_kinect2 #43: Support for multiple devices

avidbot1, 2015-01-23:
Is there currently a way to get this working with multiple devices? As far as I can tell right now when kinect2_bridge is run, one device is selected by default and that's the only one that runs.

Andrew

wiedemeyer, 2015-01-25:
There is a pull-request #40 that implemented the support. You could try it out if you want.

avidbot1, 2015-01-27:
Thanks. It does allow me to select which sensor to run but when I try to run 2 sensors at one I get a libusb busy error. Any idea how to fix it?

    process[sensor_two_kinect2_bridge-1]: started with pid [28388]
    [DepthRegistration::New] Using OpenCL registration method!
    [DepthRegistration::New] Using OpenCL registration method!
    [Freenect2Impl] enumerating devices...
    [Freenect2Impl] 16 usb devices connected
    [Freenect2Impl] found valid Kinect v2 @4:6 with serial 500902543042
    [Freenect2Impl] found valid Kinect v2 @4:4 with serial 503614242542
    [Freenect2Impl] found 2 devices
    Kinect2 devices found: 
      0: 500902543042
      1: 503614242542 (selected)
    process[sensor_two_points_xyzrgb_lowres-2]: started with pid [28408]
    process[sensor_two_points_xyzrgb_highres-3]: started with pid [28429]
    [Freenect2Impl] re-enumerating devices after reset
    [Freenect2Impl] enumerating devices...
    [Freenect2Impl] 16 usb devices connected
    [Freenect2Impl] found valid Kinect v2 @4:7 with serial 503614242542
    [Freenect2Impl] found valid Kinect v2 @4:6 with serial 500902543042
    [Freenect2Impl] found 2 devices
    [Freenect2DeviceImpl] opening...
    [UsbControl::claimInterfaces(ControlAndRgbInterfaceId)] failed! libusb error -6: LIBUSB_ERROR_BUSY
    [Freenect2DeviceImpl] closing...
    [Freenect2DeviceImpl] deallocating usb transfer pools...
    [Freenect2DeviceImpl] closing usb device...
    [Freenect2DeviceImpl] closed
    [Freenect2DeviceImpl] failed to open Kinect v2 @4:6!
    no device connected or failure opening the default one!
    OpenCV Error: Assertion failed (!fixedSize() || ((Mat*)obj)->size.operator()() == _sz) in create, file         /tmp/buildd/ros-hydro-opencv2-2.4.9-2precise-20140819-1745/modules/core/src/matrix.cpp, line 1364
    terminate called after throwing an instance of 'cv::Exception'
      what():  /tmp/buildd/ros-hydro-opencv2-2.4.9-2precise-20140819-        1745/modules/core/src/matrix.cpp:1364: error: (-215) !fixedSize() || ((Mat*)obj)->size.operator()() ==     _sz in function create
    
    [sensor_two_kinect2_bridge-1] process has died [pid 28388, exit code -6, cmd         /home/avidbot/kinect2_catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge     __name:=sensor_two_kinect2_bridge __log:=/home/avidbot/.ros/log/6b6c7520-a665-11e4-b73c-    ec0ec46a4a4b/sensor_two_kinect2_bridge-1.log].
    log file: /home/avidbot/.ros/log/6b6c7520-a665-11e4-b73c-    ec0ec46a4a4b/sensor_two_kinect2_bridge-1*.log
    respawning...

JordiMM, 2015-01-28:
Could it be that you are using USB 2.0 instead of 3.0?
https://github.com/OpenKinect/libfreenect2/issues/132

Also, maybe it's not your current problem, but to achieve myself the multiple cameras running I had to follow the instructions in this post, along with installing a 3.16 linux kernel version.
https://github.com/christiankerl/libfreenect2/issues/7#issuecomment-47422005

avidbot1, 2015-01-29:
Installing the 3.16 kernel got it working. Thanks!

wiedemeyer, 2015-02-03:
I added support for multiple devices to the devel branch. Feel free to try them out.

avidbot1, 2015-02-04:
When I launched the second device the first one shut down with this message:

    [ WARN] [1423078763.978455162]: Shutdown request received.
    [ WARN] [1423078763.978532359]: Reason given for shutdown: [new node registered with same name]

I was able to resolve it by changing `name="kinect2_bridge"` to `name="$(arg base_name)_kinect2_bridge"` in kinect2_bridge.launch.

avidbot1, 2015-02-04:
When I try to run a second device on my intel NUC it crashes with this output:

```
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    .!3U ...........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 06 10 08 47 4d 42 30 31 34 2e 31    13......GMB014.1
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                X...........

[CommandTransaction::receive] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
*** glibc detected *** /home/avidbot/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: double free or corruption (top): 0x0000000002b3f870 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x7e846)[0x7f672ed8d846]
/usr/local/lib/libfreenect2.so(_ZN12libfreenect28protocol18CommandTransaction6Result10deallocateEv+0x12)[0x7f672f906352]
/usr/local/lib/libfreenect2.so(_ZN12libfreenect28protocol18CommandTransaction6Result8allocateEm+0x22)[0x7f672f906392]
/usr/local/lib/libfreenect2.so(_ZN12libfreenect28protocol18CommandTransaction7executeERKNS0_11CommandBaseERNS1_6ResultE+0x39)[0x7f672f906919]
/usr/local/lib/libfreenect2.so(_ZN12libfreenect219Freenect2DeviceImpl5startEv+0xb52)[0x7f672f909322]
/home/avidbot/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge(_ZN13Kinect2Bridge10initDeviceERSs+0x3bd)[0x43638f]
/home/avidbot/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge(_ZN13Kinect2Bridge10initializeEv+0xa8b)[0x434d2b]
/home/avidbot/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge(_ZN13Kinect2Bridge3runEv+0x22)[0x433a48]
/home/avidbot/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge(main+0x284)[0x42f932]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xed)[0x7f672ed3076d]
/home/avidbot/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge[0x42d3f9]
```
As far as I can tell I've done everything the same on my main computer and the NUC so I'm wondering if it's an issue with the hardware. The NUC has an intel core i5, integrated graphics, and 4Gb of RAM whereas my main computer has an intel core i7, nvidia geforce gtx 745, and 12Gb of RAM. 

Any ideas on how to get this working on the NUC?

wiedemeyer, 2015-02-05:
I fixed it in f42d5e0. I replaced the name of the launch file node to just `$(arg base_name)`. Another solution would be to use `$(anon kinect2_bridge)`, but this results in very long names and the `base_name` should be unique as well.
But the other problem seems to be just libfreenect2 related. So you connected 2 devices to the NUC, right? Did you connected both to an USB 3.0 port?

jeppewalther, 2015-02-05:
I can start both of my sensors but not at the same time. 

This happens when I start the second kinect. I am on 3.13.0-44 kernel
```
➜  workspace git:(master) ✗  rosrun kinect2_bridge kinect2_bridge _sensor:=501224741942 _base_name:=kblue
parameter:
        base_name: kblue
           sensor: 501224741942
        fps_limit: -1
       calib_path: /home/juw/checkout/tangible_3d_model/workspace/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: false
edge_aware_filter: false
       publish_tf: false
     base_name_tf: kblue

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: GeForce GTX 980 (GPU)[NVIDIA Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: GeForce GTX 980 (GPU)[NVIDIA Corporation]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 25 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:8 with serial 501224741942
[Freenect2Impl] found valid Kinect v2 @4:6 with serial 500560540242
[Freenect2Impl] found 2 devices
Kinect2 devices found: 
  0: 501224741942 (selected)
  1: 500560540242
[Freenect2DeviceImpl] opening...
[UsbControl::setConfiguration(set)] failed! libusb error -99: LIBUSB_ERROR_OTHER
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @4:8!
no device connected or failure opening the default one!
using sensor defaults.
using sensor defaults.
using defaults.
using defaults.
[1]    6377 segmentation fault (core dumped)  rosrun kinect2_bridge kinect2_bridge _sensor:=501224741942 _base_name:=kblue
➜  workspace git:(master) ✗ uname -a
Linux d-main 3.13.0-44-generic #73-Ubuntu SMP Tue Dec 16 00:22:43 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
 
```

avidbot1, 2015-02-05:
I had 2 devices connected to the NUC, both connected to USB 3.0 ports. I'll post the issue in libfreenect2.

jeppewalther, 2015-02-05:
Same problem on 
Linux d-main 3.16.0-29-generic #39-Ubuntu SMP Tue Dec 16 20:54:13 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

pmolina7, 2015-02-22:
Hi @jeppewalther 
What computer system are you using? We only see this failure with the intel NUC and not with the Dell computer - this might have to do with HW.

jeppewalther, 2015-02-23:
Hi it is stationary computer with a Asus X99-S Socket 2011 V3 motherboard.

dvonhausen, 2015-05-13:
If we use the master branch of IAI Kinect2, some of us will always get a LIBUSB_ERROR_BUSY error unless we apply fix #81 to the master branch. This is because the serial number gets converted to a value that will not match the correct serial number of one (or both) kinect2 on the system and the libusb will try to claim a kinect2 port that was already claimed by the first node producing the error. To get around this using the master branch, apply fix #81. Then write launch files to start 2 kinects supplying the proper serial numbers. I have not been able to get rosrun to work so I use roslaunch which works fine with multiple kinects. You still need to apply other fixes that have already been identified (update kernel, etc) to make it work.

wiedemeyer, 2015-05-18:
I merged #81 to the devel branch.

vedanshu95, 2015-05-21:
hi,

I am trying to implement two kinect on ubuntu 14.04. Can some one help me out? ITs detecting two  kinects but i cant operate them simultaneously

wiedemeyer, 2015-05-26:
@vedanshu95 you should try to run two instances of `Protonect` at the same time. One for each device:
`Protonect <serialnumber 1>` and `Protonect <serialnumber 2>`

If that does not work, look at libfreenect2 issues for help. If it works, and only kinect2_bridge is not working, then provide some more information on the problem.

wiedemeyer, 2015-06-01:
Any update or can I close this issue?

closed 2015-06-05 by wiedemeyer