code-iai/iai_kinect2 #44: 15 Hz Limit

jeppewalther, 2015-01-26:
I am on a K7 5820K with a GTX980
 However i seem to get a rather slow update of 15 Hz

Output from kinect2_bridge semmes to be fine

```
[OpenCLDepthPacketProcessor] avg. time: 2.58556ms -> ~386.764Hz
[OpenCLDepthPacketProcessor] avg. time: 2.60462ms -> ~383.933Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.4375ms -> ~44.5682Hz
[OpenCLDepthPacketProcessor] avg. time: 2.59951ms -> ~384.688Hz
[OpenCLDepthPacketProcessor] avg. time: 2.61145ms -> ~382.929Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.5549ms -> ~44.3363Hz
[OpenCLDepthPacketProcessor] avg. time: 2.61456ms -> ~382.473Hz
```

But when subscribing to topics i get. 
```
➜  workspace git:(master) ✗ rostopic hz /kinect2_head/depth/image     
subscribed to [/kinect2_head/depth/image]
average rate: 15.000
        min: 0.057s max: 0.076s std dev: 0.00460s window: 75
average rate: 15.012
        min: 0.056s max: 0.076s std dev: 0.00473s window: 90
average rate: 14.993
        min: 0.056s max: 0.076s std dev: 0.00459s window: 105
average rate: 15.002
        min: 0.056s max: 0.076s std dev: 0.00460s window: 120
average rate: 15.005
        min: 0.056s max: 0.076s std dev: 0.00469s window: 135
Caverage rate: 14.999
        min: 0.056s max: 0.076s std dev: 0.00477s window: 147
```

Any ideas ?

jeppewalther, 2015-01-26:
This also seems to be the case on my Mac Book pro and a secondary computer (15 FPS). Seems like i am missing something obvious.

wiedemeyer, 2015-01-27:
Just to be sure, you are running `rostopic hz` on the same machine, right?
I experienced the same behavior randomly, but it is not reproducible. Sometimes it sticks to 15 hz and then it is again 30. I don't know where the problem comes from, but I guess it's libusb, which is just receiving 15 fps. Could you check how many fps kinect2_bridge receives? I just added fps infos to the output in commit 4ef189d2d655bf8308516121cf2792f8c26cb8dc.
If kinect2_bridge just receives 15 fps, then you should open an issue on libfreenect2.

jeppewalther, 2015-01-27:
Yes all the test are run locally. 

That is funny i am only able to repouce it on my macbook now. Stationary computer is back at 30 Hz. Only difference being my computer had been set to sleep in between. Yesterday restarting diddent fix it. 

**Output from my Macbook Pro**
```
[DepthPacketStreamParser::handleNewData] not all subsequences received 0
[DepthPacketStreamParser::handleNewData] not all subsequences received 1022
[DepthRegistrationOpenCL::listDevice]  devices:
  0: GeForce GT 750M (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::init]  selected device: GeForce GT 750M (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: GeForce GT 750M (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::init]  selected device: GeForce GT 750M (GPU)[NVIDIA Corporation]
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1011
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 999
[DepthPacketStreamParser::handleNewData] not all subsequences received 831
set process thread priority to: 19
set process thread priority to: 19
set process thread priority to: 19
set process thread priority to: 19
set process thread priority to: 19
set process thread priority to: 19
starting main loop
set process thread priority to: 19
set process thread priority to: 19
[kinect2_bridge] avg. time: 0.0719461 -> ~13.8993 Hz
[OpenCLDepthPacketProcessor] avg. time: 7.83431ms -> ~127.644Hz
[kinect2_bridge] avg. time: 0.0667598 -> ~14.9791 Hz
[OpenCLDepthPacketProcessor] avg. time: 6.92005ms -> ~144.508Hz
[TurboJpegRgbPacketProcessor] avg. time: 21.7866ms -> ~45.8997Hz
[kinect2_bridge] avg. time: 0.066664 -> ~15.0006 Hz
[OpenCLDepthPacketProcessor] avg. time: 6.91995ms -> ~144.51Hz
[kinect2_bridge] avg. time: 0.0665985 -> ~15.0153 Hz
[OpenCLDepthPacketProcessor] avg. time: 7.04432ms -> ~141.958Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.1661ms -> ~43.1666Hz
[kinect2_bridge] avg. time: 0.0667096 -> ~14.9903 Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1019
[OpenCLDepthPacketProcessor] avg. time: 6.96358ms -> ~143.604Hz
[kinect2_bridge] avg. time: 0.0666067 -> ~15.0135 Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1020
[OpenCLDepthPacketProcessor] avg. time: 7.0089ms -> ~142.676Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1015
[TurboJpegRgbPacketProcessor] avg. time: 23.7497ms -> ~42.1058Hz
[kinect2_bridge] avg. time: 0.0665758 -> ~15.0205 Hz
[OpenCLDepthPacketProcessor] avg. time: 6.85674ms -> ~145.842Hz
[kinect2_bridge] avg. time: 0.0667502 -> ~14.9812 Hz
[OpenCLDepthPacketProcessor] avg. time: 6.98812ms -> ~143.1Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.6253ms -> ~42.3274Hz
[kinect2_bridge] avg. time: 0.0666197 -> ~15.0106 Hz
[OpenCLDepthPacketProcessor] avg. time: 7.04442ms -> ~141.956Hz
[kinect2_bridge] avg. time: 0.0668577 -> ~14.9571 Hz
[OpenCLDepthPacketProcessor] avg. time: 6.84096ms -> ~146.178Hz
[kinect2_bridge] avg. time: 0.0664569 -> ~15.0474 Hz
[TurboJpegRgbPacketProcessor] avg. time: 21.6179ms -> ~46.258Hz
```

wiedemeyer, 2015-01-27:
You may want to open an issue on libfreenect2.

jeppewalther, 2015-01-27:
Sure should it be on the official one or on your branch?

wiedemeyer, 2015-01-27:
On the official.

wiedemeyer, 2015-01-30:
I think with the last commit de5360f5f31aba4601c62a0f5f84342c03c2cf14 to the devel branch this issue is fixed.

jeppewalther, 2015-01-30:
Yes .. it is running here. Closing issue now.

closed 2015-01-30 by jeppewalther