code-iai/iai_kinect2 #45: OpenCL registration method not available

jihoonl, 2015-01-27:
I tried to run kinect2 in ubuntu 14.04 and got the following error.

```
jihoonl@whoola:~/research/ros/freenect2/devel$ rosrun kinect2_bridge kinect2_bridge 
^[[A[DepthRegistration::New] OpenCL registration method not available!
[DepthRegistration::New] OpenCL registration method not available!
[OpenGLContext] failed to initialize glew for the current context!
Segmentation fault (core dumped)
```

wiedemeyer, 2015-01-27:
That is because libfreenect2 does not initialize glfw. I added a fix for this in the devel branch.

jihoonl, 2015-01-27:
I see thanks.

closed 2015-01-27 by jihoonl

zchenpds, 2016-06-03:
Hi @wiedemeyer,
Could you please point out which commit fixed this issue? I tried but couldn't find it.