code-iai/iai_kinect2 #47: Sensor serial id does not get passed trough

jeppewalther, 2015-02-04:
The serial number does not seem to be parsed trough. 
Tested some of the other arguments they seem to be working fine. 

```
➜  workspace git:(master) ✗ rosrun kinect2_bridge kinect2_bridge _sensor:=500560540242                  
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: GeForce GTX 980 (GPU)[NVIDIA Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: GeForce GTX 980 (GPU)[NVIDIA Corporation]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 25 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:7 with serial 501224741942
[Freenect2Impl] found valid Kinect v2 @4:5 with serial 500560540242
[Freenect2Impl] found 2 devices
Kinect2 devices found: 
  0: 501224741942 (selected)
  1: 500560540242
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
```

wiedemeyer, 2015-02-05:
I fixed it in f42d5e086d9ca8607ce16914b69826ccf725cb97.
The problem is strange, if a parameter contains just numbers it is converted to an int or double and you can't read it so a string. I changed the type to double and convert it to a string later. this should fix it.

jeppewalther, 2015-02-05:
I had a short look at it yesterday and wondered what might be happening. Glad you figured it out. It is working now.

closed 2015-02-05 by jeppewalther

SilvioGiancola, 2015-10-16:
Hi everyone,
I have met the same problem with a new machine I had to configure. I have installed the latest version of libfreenect2 and iai_kinect2 and when I try to pass a sensor serial in argument it is not taken into consideration. Here's an example :

sineco@sineco-X9SPV-M4:~$ rosrun kinect2_bridge kinect2_bridge _sensor:=507040542542
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/sineco/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 15 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 507991442542
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 507040542542
[Info] [Freenect2Impl] found 2 devices
Kinect2 devices found: 
  0: 507991442542 (selected)
  1: 507040542542

As you can see the second kinect2 has not been selected. I have tried with "" but without success. Both Kinects are seen by my system, Protonect works well with 2 instances. I use a 3.16 Kernel in a ubuntu 14.04.1 distribution.  

Thank you!

wiedemeyer, 2015-10-16:
Specifying the serial only works with the launch. With rosrun it is converted to a number and can't be read in as a string.