code-iai/iai_kinect2 #49: No publication of depth images in Kinect Bridge

andrewadare, 2015-02-05:
Hey there,
Really enjoying the Kinect bridge, thanks for your hard work. I've unfortunately run into a problem where everything seems to work fine, except the depth data isn't captured and published. This happens regardless of using CPU or the OpenCL acceleration on my GPU.

What we see is ...

starting 8 worker threads
starting main loop

[kinect2_bridge] depth avg. time: inf -> ~0 Hz
[kinect2_bridge] color avg. time: 0.0354116 -> ~28.2394 Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.8243ms -> ~59.4378Hz
[kinect2_bridge] depth avg. time: inf -> ~0 Hz
[kinect2_bridge] color avg. time: 0.0330776 -> ~30.2319 Hz

As the depth avg. time is 0, we don't get any depth images output.

Any guesses?

wiedemeyer, 2015-02-06:
It could be that your kernel is too old, or the USB 3.0 controller is not supported. The terminal output earlier might give a clue.
You could also look up the [issues posted at libfreenect2](https://github.com/OpenKinect/libfreenect2/issues), there are multiple entries to this problem.

andrewadare, 2015-02-11:
Thanks for the reply, but we have great performance when using libfreenect2. No problem at all with depth, USB controllers, etc. It's only when we try to use the IAI software that we have this issue.

wiedemeyer, 2015-02-11:
Could you check out the recent version and than post also the terminal output before `starting main loop`?

frederico00, 2015-02-12:
Hi,

I'm new with this, but i think that i have a depth registration with the ./Protonect example using OpenGL. With the bridge i'm using the CPU and i can have a depth registration. Is it cause of the USB controller?

sudo ./Protonect
[OpenGLDepthPacketProcessor] avg. time: 16.4282ms -> ~60.8708Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.5919ms -> ~40.6638Hz

rosrun kinect2_bridge kinect2_bridge 
[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!

[TurboJpegRgbPacketProcessor] avg. time: 25.7458ms -> ~38.8413Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~181.367Hz (5.51367ms) publishing rate: ~14.978Hz

and cause of that the when i run: rosrun registration_viewer viewer kinect2 sd cloud

topic color: /kinect2/rgb_lowres/image
topic depth: /kinect2/depth_lowres/image
starting receiver...

After this nothing apears ......

Regards,
FC

frederico00, 2015-02-12:
I post the terminal output before main loop:

      base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/frederico/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:3 with serial 000778644447
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 000778644447 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    .!3U...@........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 02 0a 19 47 4d 41 39 38 35 2e 31    13......GMA985.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        .&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[Freenect2DeviceImpl] started

device serial: 000778644447
device firmware: 4.3.3913.0.7

default ir camera parameters: 
fx 365.453, fy 365.453, cx 259.876, cy 205.669
k1 0.0907246, k2 -0.270171, p1 0, p2 0, k3 0.0958409

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults.
using sensor defaults.
using defaults.
using defaults.
^C
camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.4533996582031, 0, 259.8763122558594;
  0, 365.4533996582031, 205.6690979003906;
  0, 0, 1]
distortion coefficients ir:
[0.09072461724281311, -0.2701708972454071, 0, 0, 0.09584092348814011]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[0; 0; 0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
starting main loop

wiedemeyer, 2015-02-12:
`[TransferPool::submit] failed to submit transfer` this means that the transfer of depth images could not be initialized and this an error message from libfreenect2.
I noticed that you started protonect with sudo. So I guess you need root rights for it to work. And for the kinect2_bridge it is the same. Did you change the permissions to access the sensor like [described in the instructions](https://github.com/code-iai/iai_kinect2#permissions-to-access-the-kinect-one)?

frederico00, 2015-02-12:
Hey Thiemo,

Yes I changed the permisions. I also tried without the permisions but the
result is: 0 devices found.

So i guess it isn't a permisions problem ...

Melhores Cumprimentos,
--
Frederico Cabeço
Licenciado em Eng. Electrótecnica
Telm: (+351) 912376259

wiedemeyer, 2015-02-12:
Did you added the modifications to the CMakelists.txt from libfreenect2? please check with `ldd catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge` if the link entry for libusb is pointing to the one inside the `libfreenect2/depends` folder.
You could also try to run kinect2_bridge as root by switching to root, sourcing your ws and running kinect2_bridge.

wiedemeyer, 2015-02-12:
Is protonect running also without sudo?

frederico00, 2015-02-12:
The problem was with opencl. At the beginning i had installed nvidia-libopencl1-340 (340.76-0ubuntu1~xedgers14.04.1) to put libfreenect to work. When i was trying to to install Beignet, program in the list I installed ocl-icd-libopencl1. That was a mistake... So what i did was basicaly:

Removed the following packages:
ocl-icd-libopencl1

Installed the following packages:
nvidia-libopencl1-340 (340.76-0ubuntu1~xedgers14.04.1)

and rebuild everything again

Maybe it was the same problem of andrewadare ...

lehnert, 2015-02-13:
Hi,

I'm getting a similar result where the kinect2_bridge node slows down to 0 Hz whenever I try and use another node which subscribes to a kinect2 depth topic.

This is the output before the "starting main loop" message

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Caicos (GPU)[Advanced Micro Devices, Inc.]
  1: Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz (CPU)[GenuineIntel]
[DepthRegistrationOpenCL::init]  selected device: Caicos (GPU)[Advanced Micro Devices, Inc.]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Caicos (GPU)[Advanced Micro Devices, Inc.]
  1: Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz (CPU)[GenuineIntel]
[DepthRegistrationOpenCL::init]  selected device: Caicos (GPU)[Advanced Micro Devices, Inc.]
starting main loop
------------------------------------------------------------------------------------------------------------------------------

If I then use rviz to try and visualise the point clouds or even the depth image the publishing rate slowly drops to 0Hz and looks like this

------------------------------------------------------------------------------------------------------------------------------
[kinect2_bridge] depth processing: ~643.602Hz (1.55376ms) publishing rate: ~29.6362Hz
[kinect2_bridge] color processing: ~210.41Hz (4.75263ms) publishing rate: ~30.3022Hz
[OpenGLDepthPacketProcessor] avg. time: 18.0526ms -> ~55.3937Hz
[TurboJpegRgbPacketProcessor] avg. time: 18.4748ms -> ~54.1277Hz
[kinect2_bridge] depth processing: ~720.064Hz (1.38876ms) publishing rate: ~30.2376Hz
[kinect2_bridge] color processing: ~212.987Hz (4.69512ms) publishing rate: ~29.9053Hz
[OpenGLDepthPacketProcessor] avg. time: 17.8656ms -> ~55.9736Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.6665ms -> ~50.8478Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 895
[kinect2_bridge] depth processing: ~672.564Hz (1.48685ms) publishing rate: ~29.6603Hz
[kinect2_bridge] color processing: ~217.339Hz (4.60111ms) publishing rate: ~29.9935Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.3804ms -> ~51.5985Hz
[kinect2_bridge] depth processing: ~12.1556Hz (82.2665ms) publishing rate: ~19.3187Hz
[kinect2_bridge] color processing: ~152.455Hz (6.55933ms) publishing rate: ~29.6442Hz
[OpenGLDepthPacketProcessor] avg. time: 31.452ms -> ~31.7945Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.2327ms -> ~51.9949Hz
[kinect2_bridge] depth processing: ~9.86241Hz (101.395ms) publishing rate: ~18.6583Hz
[kinect2_bridge] color processing: ~150.82Hz (6.63041ms) publishing rate: ~29.6533Hz
[kinect2_bridge] depth processing: ~8.47387Hz (118.01ms) publishing rate: ~17.9944Hz
[kinect2_bridge] color processing: ~144.703Hz (6.91071ms) publishing rate: ~28.3246Hz
[TurboJpegRgbPacketProcessor] avg. time: 18.8472ms -> ~53.0583Hz
[OpenGLDepthPacketProcessor] avg. time: 38.9854ms -> ~25.6507Hz
[kinect2_bridge] depth processing: ~7.46039Hz (134.041ms) publishing rate: ~11.6636Hz
[kinect2_bridge] color processing: ~149.541Hz (6.68715ms) publishing rate: ~28.9923Hz
[TurboJpegRgbPacketProcessor] avg. time: 18.839ms -> ~53.0814Hz
[kinect2_bridge] depth processing: ~2.77453Hz (360.421ms) publishing rate: ~9.32949Hz
[kinect2_bridge] color processing: ~154.969Hz (6.45289ms) publishing rate: ~4.33155Hz
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1019
[TurboJpegRgbPacketProcessor] avg. time: 19.5674ms -> ~51.1055Hz
[OpenGLDepthPacketProcessor] avg. time: 66.7277ms -> ~14.9863Hz
[kinect2_bridge] depth processing: ~2.26637Hz (441.234ms) publishing rate: ~9.32776Hz
[kinect2_bridge] color processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.2931ms -> ~49.2778Hz
[kinect2_bridge] depth processing: ~2.38433Hz (419.405ms) publishing rate: ~8.99496Hz
[kinect2_bridge] color processing: ~135.494Hz (7.38041ms) publishing rate: ~0.99944Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.8688ms -> ~50.3302Hz
[kinect2_bridge] depth processing: ~2.21791Hz (450.875ms) publishing rate: ~8.99513Hz
[kinect2_bridge] color processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.261ms -> ~49.3558Hz
[OpenGLDepthPacketProcessor] avg. time: 65.6825ms -> ~15.2247Hz
[kinect2_bridge] depth processing: ~2.35223Hz (425.129ms) publishing rate: ~9.329
------------------------------------------------------------------------------------------------------------------------------

Any ideas what would be causing it to slow down to 0Hz?

wiedemeyer, 2015-02-13:
I think this is a problem I introduced by decoupling depth from color. So each worker thread checks for a depth frame and then for a color frame. if the time for processing a depth frame takes very long (very high CPU load), all worker threads are working on processing the depth images, so no worker thread ever comes to work on the color frames. I will add a fix for this.
One solution would be to reduce CPU load. Are you using compressed image transfer? Compression uses a lot CPU. Does the processing time improve if you also use openCL for depth processor?

wiedemeyer, 2015-02-13:
I added a fix to the devel branch, commit 2b684247a2c1aee9f5f55cd8ae4af823cfa2bd0f. Could you checkout the devel branch and see if it fixes the problem for you?

lehnert, 2015-02-13:
I tried a few different fixes including subscribing to the compressed depth image but that didn't work either. I have tried the code on two desktop machines and a laptop with different gpu's and getting the same result. The work around im using at the moment is by using the forked repo by OpenPTrack.

In regards to your latest message I just left work for the weekend and I don't have a kinect2 at my home. I will have to check your latest changes after the weekend. I really appreciate you taking the time to look into my issue, thanks. I will let you know how it goes.

lehnert, 2015-02-17:
Hi, I tested your changes and it is still producing 0Hz publishing rate for the colour topic. This is testing the node with nothing else running only with RVIZ trying to visualise the point clouds. So I think there is little cpu load when I'm trying to test the system.

wiedemeyer, 2015-02-17:
What did you use in rviz PointCloud2 or DepthImage plugin? And when you close rviz does it start to publish color again? Did you looked at your CPU load with htop?
Right now I have no idea of what could be wrong. You could add some debug output to see whether it ever receives color images and calls the processing and publishing methods. Or start it in a debugger.

closed 2015-04-20 by wiedemeyer