code-iai/iai_kinect2 #50: Noise in data from kinect2_bridge Devel Branch

jeppewalther, 2015-02-06:
I am having a strange issue with noise in the data from bridge. It is only noticeable when I do a integration scan with kinect fusion.  It only happens in the development branch. I have been looking over the code in kinect2_bridge.cpp but I have not been able to figure out what introduces the noise. I also tried to add a depth topic with float resolution to kinect2_bridge, but still same problem. 

I get depth data from 
```
std::string topicDepth = K2_TOPIC_IMAGE_DEPTH K2_TOPIC_RAW;
```
Below is example images from the my scanner application, only  difference is the version of iai_kinect2.

Any ideas on what might introduce this noise?. 

**Master Branch**
[image]

**Devel Branch**
[image]

wiedemeyer, 2015-02-06:
What is the commit you are using on the master and on the devel branch?

wiedemeyer, 2015-02-06:
I think I found the problem. When I changed the parameter handling I added wrong default values to bilateral and edge aware filters, so they were disabled by default.

jeppewalther, 2015-02-06:
Great! That fixed it :+1: 

I noticed that you get float values from libfreenect2 and down sample them to 16 bit. Does that give any quality loss. Or is there another reason for libfreenect2 using floats.

wiedemeyer, 2015-02-06:
The resolution of the 16 bit depth image is 1mm. So not much information is lost, also this is still lower than the noise you will see on flat surfaces. The reason why libfreenect2 returns depth values as floats is probably that the depth processing is based on floating-point operations.
There are a couple of reasons why I am using 16 bit integer, mostly memory usage and performance. In ROS the images are often transfered over the network, so less size means more frames per second. Also depth image compression is only working with 16 bit depth images. And for the depth registration I use OpenCL and the slowest part is to transfer the data to and from the GPU, so less size is better.

jeppewalther, 2015-02-07:
Thanks for the info. Closing issue now.

closed 2015-02-07 by jeppewalther