code-iai/iai_kinect2 #52: Kinect2 Calibration viewer window doesn't get updated

wjchen84, 2015-02-10:
I just started to use your package and kinect2. Apologize if I am asking a stupid question. I managed to get almost all the commands to run. But when running 'rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color', Kinect2 Calibration viewer window doesn't get updated, so that I can only record one picture (or multiple same ones). I am not sure if it was designed like this, but I was expecting to record a bunch of different images for calibration. From your example on README, it seems you were using ~400 images for calibration. Could you please let me know if I am wrong and where I should check?

Thank you in advance!

wiedemeyer, 2015-02-10:
I have never experienced this. So the image shown is always the same? What is the kinect2_bridge output while the calibration tool is running?
Please try to run kinect2_bridge with a frame limit to reduce cpu load. `rosrun kinect2_bridge kinect2_bridge _fps_limit:=1` should limit it to 1 frame per second.

wjchen84, 2015-02-12:
Thank you Thiemo for the suggestion. I tried with reduced frame rate as you suggested and it worked now (with up to 15fps). So it seems my onboard intel gpu cannot handle the default computation load? I actually has Nvidia Quadro K2100M available but I couldn't make the Nvidia driver work with this package so I am using the Ubuntu driver (Nouveau) now. 

If without reducing frame rate, the image shown is always the same, and the <code>kinect2_bridage</code> output is:
<pre><code>
chin@chin-ThinkPad-W540:~$ rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/chin/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: false
edge_aware_filter: false
       publish_tf: false
     base_name_tf: kinect2

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 511488142542
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 511488142542 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 0b 09 10 47 4d 46 31 35 31 2e 31    13......GMF151.1
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        .&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 511488142542
device firmware: 2.3.3912.0.7

default ir camera parameters: 
fx 365.192, fy 365.192, cx 259.865, cy 205.231
k1 0.0904933, k2 -0.271391, p1 0, p2 0, k3 0.0980318

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
[... 57 lines cut ...]
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[OpenCLDepthPacketProcessor] avg. time: 13.7366ms -> ~72.7984Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: 28.9298ms -> ~34.5664Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth avg. time: 0.0333422 -> ~29.992 Hz
[kinect2_bridge] color avg. time: 0.0468875 -> ~21.3277 Hz
[OpenCLDepthPacketProcessor] avg. time: 11.6963ms -> ~85.4973Hz
^C[Freenect2DeviceImpl] stopping...
</code></pre>

wiedemeyer, 2015-02-12:
I guess it is not the GPU but the CPU load. The calibration tool tries to detect the pattern in each image it receives, so if the rate in which it receives images is as high or higher than the time it takes to detect the pattern, then it will use as much CPU as possible. For calibration there is no need for a high frame rate, because images are taken slowly.

wjchen84, 2015-02-12:
Thank you Thiemo for the explanation. I think that makes sense to me. I can feel the heat from the CPU when running for long. I will just use low frame rate for calibration. I think we can close this issue now.

closed 2015-02-12 by wjchen84