code-iai/iai_kinect2 #53: Provide a sample calibration

xlz, 2015-02-17:
Can you provide a sample calibration file as default for testing?

wiedemeyer, 2015-02-17:
Hi,
there are two folders with example calibrations. Or do you mean kinect2_bridge should load a default calibration if no calibration for the specific sensor is provided?

xlz, 2015-02-17:
Ahh, I didn't look hard enough.

Your idea is good though.

closed 2015-02-17 by xlz

wiedemeyer, 2015-02-17:
They are located in [kinect2_bridge/data/<serialnumber>](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge/data/299150235147).
Right now the defaults are the intrinsics reported by the sensor, with no extrinsics. Becauce I don't have default values for the extrinsic parameters that work well for all sensors.