code-iai/iai_kinect2 #54: Support for rosbag

jeppewalther, 2015-02-19:
I can as it is record with rosbag from topics posted by kinect2_bridge. However it would be nice if I only needed to record raw depth, ir and rgb. So if kinect2_bridge could have an option to grab input from topics instead of from libfreenect2.

Maybe it would be cleaner to make a kinect2_bag package. Or possibly to split kinect2_bridge into a grabber and a processor.

wiedemeyer, 2015-02-24:
Hi,
yes it is a good idea. I will look into it.

wiedemeyer, 2015-03-05:
I divided the single parts into multiple nodelets. One nodelet that just receives the data from libfreenect2 and published it through ROS topics, and others for the processing, for example the registration, etc. But it seems to create too much overhead, even with zero-copy pointer transfers between the nodelets. It was not able to provide 30 fps for compressed registered depth images, while the CPU load was between 70 to 80 percent. The current implementation manages 30 fps on compressed depth and rgb images while having a CPU of ~40 percent.
But it's still possible to just record the topics that your nodes listen too. And recording the compressed images reduces the bag file size.

closed 2015-06-01 by wiedemeyer