code-iai/iai_kinect2 #55: What am I doing wrong?

Parth-Mehrotra, 2015-02-20:
Hi all, here is what I did on Ubuntu 14.04 LTS running on a Toshiba Satellite, intel i7, nvidia with usb 3.0 and 2.0 ports.

1. Setup ros by running this [install-ros.sh script] (https://github.com/Parth-Mehrotra/Tools/tree/master/InstallScripts)

2. Setup opencv by running this [install-opencv.sh script] (https://github.com/Parth-Mehrotra/Tools/tree/master/InstallScripts)

3. Setup PCL by running this [install-pcl.sh script] (https://github.com/Parth-Mehrotra/Tools/tree/master/InstallScripts)

4. Installed  libfreenect2 via the instructions at the master branch

5. Made the changes and installed

6. Cloned the repository into an empty catkin workspace

7. I sourced the respective setup.zsh files from my /opt/ros/... and from the devel/ folders.

At this point I have encountered no issues

I tried running 
```zsh
rosrun kinect2_bridge kinect2_bridge
```

and I get the following message:
```
[ERROR] [1424391698.413758209]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying..
```

So I assume I need to run roscore or something like that. So if I run 
```zsh
roscore
```
in one terminal and
```zsh
rosrun kinect2_bridge kinect2_bridge
```

in another terminal, I get the following segmentation fault.
```
[ERROR] [1424393345.496836066]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
[ INFO] [1424393446.243884489]: Connected to master at [localhost:11311]
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/parthmehrotra/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[1]    2679 segmentation fault (core dumped)  rosrun kinect2_bridge kinect2_bridge
```

I must be overlooking something really trivial. Thanks for taking the time to help me.

wiedemeyer, 2015-02-20:
Did you add the modifications to libfreenect2 before building and it? You could check if `protonect` from libfreenect2 is running for you.
Why are you using different versions of PCL and openCV? They should already be installed when you install ros. I would recommend to setup everything like described in the README, just to make sure it's not related to your setup.
Also you should check if the libraries kinect2_bridge uses to compile time are the same that it links against run time.

Parth-Mehrotra, 2015-02-20:
How can I check that? Are there any version numbers I can check or something?

wiedemeyer, 2015-02-24:
Please check if `protonect` is working and setup everything as described in the [README](https://github.com/code-iai/iai_kinect2/blob/master/README.md).

andersgb1, 2015-02-26:
Hi, I have the **exact** same situation here on a Ubuntu 14.04 x64 machine. As far as I can see, I did follow your fine tutorial to the point. Can you help use how we check protonect?

On another note, could it be the problem is on an x64 system?

andersgb1, 2015-02-26:
As an addendum, this is the tail of dmesg after the error above:

[165657.980726] kinect2_bridge[25518]: segfault at 1c0 ip 00007fde5da57488 sp 00007fffe56ab250 error 4 in libglfw.so.3.0[7fde5da50000+12000]

andersgb1, 2015-02-26:
And here's what I got from gdb, if it has any relevance. Stack trace with the error at the bottom:

andersgb1@andersgb1-laptop:~/workspace/catkin_ws/devel/lib/kinect2_bridge$ gdb -ex r --args ./kinect2_bridge 
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./kinect2_bridge...(no debugging symbols found)...done.
Starting program: /home/andersgb1/workspace/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdd481700 (LWP 25786)]
[New Thread 0x7fffdc392700 (LWP 25788)]
[New Thread 0x7fffdbb91700 (LWP 25789)]
[New Thread 0x7fffdb390700 (LWP 25790)]
[New Thread 0x7fffdab8f700 (LWP 25795)]
[New Thread 0x7fffda38e700 (LWP 25796)]
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/andersgb1/workspace/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

Program received signal SIGSEGV, Segmentation fault.
glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266)
    at /home/andersgb1/workspace/libfreenect2/depends/glfw_src/src/window.c:560
560	            return window->glMajor;
(gdb) bt
#0  glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266)
    at /home/andersgb1/workspace/libfreenect2/depends/glfw_src/src/window.c:560
#1  0x00007ffff58a9704 in flextInit (window=0x0, bindings=0x6fd390)
    at /home/andersgb1/workspace/libfreenect2/examples/protonect/src/flextGL.c:20
#2  0x00007ffff58acf7b in libfreenect2::OpenGLDepthPacketProcessorImpl::initialize (this=this@entry=0x6fa4a0)
    at /home/andersgb1/workspace/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:469
#3  0x00007ffff58ab019 in libfreenect2::OpenGLDepthPacketProcessor::OpenGLDepthPacketProcessor (this=0x6b7aa0, parent_opengl_context_ptr=<optimized out>, 
    debug=<optimized out>)
    at /home/andersgb1/workspace/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:783
#4  0x00007ffff58992a6 in libfreenect2::OpenGLPacketPipeline::createDepthPacketProcessor (this=0x6b3d90)
    at /home/andersgb1/workspace/libfreenect2/examples/protonect/src/packet_pipeline.cpp:109
#5  0x00007ffff5899576 in libfreenect2::BasePacketPipeline::initialize (
    this=this@entry=0x6b3d90)
    at /home/andersgb1/workspace/libfreenect2/examples/protonect/src/packet_pipeline.cpp:43
#6  0x00007ffff5899943 in libfreenect2::OpenGLPacketPipeline::OpenGLPacketPipeline (this=0x6b3d90, parent_opengl_context=<optimized out>, 
    debug=<optimized out>)
    at /home/andersgb1/workspace/libfreenect2/examples/protonect/src/packet_pipeline.cpp:102
#7  0x000000000042054e in Kinect2Bridge::initialize() ()
#8  0x0000000000420bb9 in Kinect2Bridge::run() ()
#9  0x000000000040acc7 in main ()

andersgb1, 2015-02-26:
Sorry for spamming, but I got it to run by passing **_depth_method:=cpu**

wiedemeyer, 2015-02-26:
Hi,
it looks like it segfaults inside the glfw lib that is used by libfreenect2 for the OpenGL depth processing. If it also happens when running `protonect`, you should open an issue on libfreenect2.
You should try out the OpenCL depth processor, the CPU processor is really slow.

andersgb1, 2015-02-26:
Thanks for reply,

However, I think the problem here is that neither @Parth-Mehrotra or myself would know how to do these tests. How can I run protonect? I can't run the opencl processor because it reports it as unavailable, but I guess the error is on my side for not having installed correctly...

wiedemeyer, 2015-02-26:
`protonect` is an executable inside libfreenect2. After compiling libfreenect2 it should be located in `libfreenect2/examples/protonect/bin/Protonect`.

Parth-Mehrotra, 2015-02-26:
If I follow the steps on the libfreenect2 and execute the ./bin/Protonect I also get a segmentation fault. Should I create an Issue there?

howlong, 2015-03-13:
I encountered the same issue. Have you guys solved it?

wiedemeyer, 2015-03-13:
Please do the following steps:
- build libfreenect2 in `Debug`
- run it in gdb: `gdb <libfreenect2>/examples/protonect/bin/Protonect` then `run`
- if it crashes, do a 'bt' and find out what went wrong

howlong, 2015-03-13:
thanks for your help.
and I tried and here is the output and my laptop config.
what should I do next?
--------------------------- output of protonect ------------
Starting program: /home/lhan/git_lib/libfreenect2/examples/protonect/bin/Protonect 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe4bfe700 (LWP 11134)]
[New Thread 0x7fffe43fd700 (LWP 11135)]
[New Thread 0x7fffe1271700 (LWP 11136)]

Program received signal SIGSEGV, Segmentation fault.
glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266)
    at /home/lhan/git_lib/libfreenect2/depends/glfw_src/src/window.c:560
560	            return window->glMajor;
(gdb) b
Breakpoint 1 at 0x7ffff640c508: file /home/lhan/git_lib/libfreenect2/depends/glfw_src/src/window.c, line 560.
(gdb) t
[Current thread is 1 (Thread 0x7ffff7f9ca40 (LWP 11129))]

------------------------- laptop config ---------------------------
Thinkpad T500:
CPU:	Intel® Core™2 Duo CPU T9400 @ 2.53GHz × 2 
GPU:	Gallium 0.4 on AMD RV635 [Mobility Radeon HD 3650]
RAM:	3.9 GiB

OS:	Ubuntu 14.04 LTS x64
Kernal:	3.13.0-32-generic

wiedemeyer, 2015-03-13:
Could it be that you GPU does not support the GL version needed for libfreenect2? You could try to use OpenCL depth processing instead. Or look for a solution on libfreenect2 issue page, or submt a issue there.
But anyways, Kinect 2 requires a USB 3.0 port. Does your computer has one? I'm just asking because a Core 2 Duo is pretty old.

howlong, 2015-03-13:
what's the lowest GL version for libfreenect2? I couldn't find any description about it.
Also, your guess is right. It does not have usb 3.0. I will use another one with usb 3.0.
thanks a lot.

closed 2015-05-27 by wiedemeyer