code-iai/iai_kinect2 #56: About kinect2_brdige.launch 's queue_size

aginika, 2015-02-22:
Hi!

I am using kinect2_bridge.launch to visualize the point clouds in rviz.
But, when I launched as `roslaunch kinect2_bridge kinect2_bridge.launch`,
it didn't publish any points in /kinect2/depth_{lowres | high_res}/points.

So, I changed the launch file's depth_image_proc queue_size from 2 to 4.
Then, the pointcloud is published correctly.

I'm just curious, but is my environment set up wrongly? If then, what should I do?

Thanks.

wiedemeyer, 2015-02-24:
Hi,
I have no clue what the problem could be, but good to hear that it is working now.

aginika, 2015-02-24:
Hmm, all right.
Thank you.

closed 2015-02-24 by aginika

lywind, 2015-03-12:
I encountered the same problem and fixed it just using your advice.