code-iai/iai_kinect2 #57: libusb acting up?

tysonhammen, 2015-02-26:
I had everything running fine then I had to shut down the machine to move it and when it came back up I now keep getting "[DepthPacketStreamParser::handleNewData] image data too short!

I thought that maybe connecting my v1 kinect had messed up the usb somehow so I completely rebuilt and am still getting the same issue. Any ideas?

tysonhammen, 2015-02-26:
As a note I see these messages in dmesg:
[ 2709.860400] xhci_hcd 0000:00:10.0: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[ 2800.750936] kinect2_bridge[29085]: segfault at 1c0 ip 00007f9c565cf4f8 sp 00007fff90c0fde0 error 4 in libglfw.so.3.0[7f9c565c8000+12000]
[ 2850.897495] usb 7-2.1: reset SuperSpeed USB device number 10 using xhci_hcd
[ 2850.913309] xhci_hcd 0000:00:10.0: xHCI xhci_drop_endpoint called with disabled ep ffff880415ef2c00

wiedemeyer, 2015-02-26:
I guess it is an issue related to USB. Did you changed your kernel or updated the system before restart, or is it the exact same configuration? Did you tried to reconnect the Kinect2? Maybe to a different USB 3 port? If nothing works, try `protonect` from libfreenect2. If that one fails too, it's a problem related to libfreenect2 and you could open an issue on their issue page.

tysonhammen, 2015-02-26:
I did try a different kernel but I completely rebuilt the system from
scratch after that to make sure everything was back to stock. Protonect
does do the exact same thing. I will open up the issue over there.

tysonhammen, 2015-02-26:
I got that issue figured out and everything now connects up but I still have no depth data. when freenect2 builds it says that it finds the OpenGL .so file but when the bridge starts it still says that it is using the CPU method. I am using the fglrx driver but also have the fglrx-updates driver as well. I switched between them but it didn't seem to make any difference. Any ideas what the issue could be here?

tysonhammen, 2015-02-27:
@wiedemeyer so I go the USB issue figured out. Turns out this was the same issue you faced. I do have one other issue that is being a pain though. I don't have any depth data. I get RGB and IR but depth is not being processed. I have tried to use fglrx and fglrx-updates. Rebuilding after the switch. Don't suppose you have any information that may be of help?

wiedemeyer, 2015-02-27:
oh, if you enabled features of libfreenect2 you have to rebuild kinect2_bridge in order to propagate the changes. Either `catkin_make clean` or `rm -rf catkin_ws/build`.

tysonhammen, 2015-02-27:
so I have enabled the OpenCL features and rebuilt both libfreenect2 and kinect2_bridge and now the bridge reports that OpenCL is being used for depth registration but no data is being processed/published.

wiedemeyer, 2015-02-27:
My guess is, that it does receive any depth images from libfreenect2. Did you managed to get `protonect` running?

tysonhammen, 2015-02-27:
@wiedemeyer I did but your correct. It appears that there is no depth data being output by it. Not sure why yet.. I posted the question on the issue I opened previously on their page.

tysonhammen, 2015-02-28:
@wiedemeyer so if I run the kinect2_bridge I don't see the "[DepthPacketStreamParser::handleNewData] image data too short!" errors but I do if I run protonect... is it just ignoring them or do have I just messed something up?

wiedemeyer, 2015-03-12:
Those messages get printed by libfreenect2.

closed 2015-05-27 by wiedemeyer