code-iai/iai_kinect2 #58: Beignet install

tysonhammen, 2015-02-28:
Not sure if you want to add the notes here or not but just installing the lllvm and clang packages will not let Beignet build. I had to add a couple of additional sym links to allow "make" to run. They are below.

sudo ln -s /usr/lib/llvm-3.5/bin/llvm-as /usr/bin/llvm-as
sudo ln -s /usr/lib/llvm-3.5/bin/llvm-link /usr/bin/llvm-link

mitchellwills, 2015-02-28:
I had to do something similar

wiedemeyer, 2015-02-28:
Did you installed llvm 3.5 from the Ubuntu packages? The Ubuntu package has some [issues](https://bugs.launchpad.net/ubuntu/+source/llvm-toolchain-3.5/+bug/1372062), some files are located in wrong directories. Please use the one provided from http://llvm.org.

tysonhammen, 2015-02-28:
Got it. Yes, I just installed through apt.

closed 2015-05-27 by wiedemeyer