code-iai/iai_kinect2 #60: Calibration: depth.png images

cjpedc, 2015-03-13:
Hi,

I am trying to perform the calibration of the Kinect2 and when I execute:

$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate dept

I get the following error message:

OpenCV Error: Assertion failed (points.checkVector(2) >= 0 && (points.depth() == CV_32F || points.depth() == CV_32S)) in boundingRect, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/contours.cpp, line 1895
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/contours.cpp:1895: error: (-215) points.checkVector(2) >= 0 && (points.depth() == CV_32F || points.depth() == CV_32S) in function boundingRect

Do you have any suggestion about what could be the problem and how to solve it?

Another thing which I don't know if is normal or not, is that the ''XXXXXdepth.png'' and ''0000_sync_depth.png'' images are completely black. Are they suppose to be like that or they should be like in gray scale?   

Finally, the maximum frequency for depth is not greater than 3 Hz. This is the message that I am getting when running " rosrun kinect2_bridge kinect2_bridge " :+1: 

parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/tmrcv1/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 026235343747
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 026235343747 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 02 03 0d 47 4d 48 38 32 35 2e 31    13......GMH825.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  1d 26 00 00                                        .&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 026235343747
device firmware: 2.3.3913.0.7

default ir camera parameters: 
fx 366.21, fy 366.21, cx 255.93, cy 203.813
k1 0.0890476, k2 -0.268077, p1 0, p2 0, k3 0.096709

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 0

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[366.2099914550781, 0, 255.9300994873047;
  0, 366.2099914550781, 203.8130035400391;
  0, 0, 1]
distortion coefficients ir:
[0.08904761075973511, -0.2680774033069611, 0, 0, 0.0967089906334877]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
starting main loop

[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~7794.49Hz (0.128296ms) publishing rate: ~2.99179Hz
[kinect2_bridge] color processing: ~4252.88Hz (0.235135ms) publishing rate: ~29.2531Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: 26.3784ms -> ~37.9098Hz
[kinect2_bridge] depth processing: ~9661.82Hz (0.1035ms) publishing rate: ~2.99693Hz
[kinect2_bridge] color processing: ~4246.25Hz (0.235502ms) publishing rate: ~29.9693Hz
[TurboJpegRgbPacketProcessor] avg. time: 21.9903ms -> ~45.4745Hz

----------------------------------------------------------------------------------------------------------------------------------------
Thanks in advance.

wiedemeyer, 2015-03-16:
If you run `Protonect`, which is located inside libfreenect2, does it display depth images?
I would recommend to not use the CPU depth processing. It seems like you have beignet installed. so you could enable opencl depth processing in libfreenect2 cmake and compile it and kinect2_bridge again.

cjpedc, 2015-03-16:
Thanks for your answer.
Protonect doesn't display depth images.
Although, I was able to enable opencl depth processing in libfreenect2 by doing cmake and compiling it again. This also fixed the problem with the depth publishing rate.

-------------------------------------------------------------------------------------------------------------------------------------------
$ rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/tmrcv1/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 026235343747
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 026235343747 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 02 03 0d 47 4d 48 38 32 35 2e 31    13......GMH825.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5d 26 00 00                                        ]&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5f 26 00 00                                        _&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 026235343747
device firmware: 2.3.3913.0.7

default ir camera parameters: 
fx 366.21, fy 366.21, cx 255.93, cy 203.813
k1 0.0890476, k2 -0.268077, p1 0, p2 0, k3 0.096709

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
[DepthPacketStreamParser::handleNewData] not all subsequences received 0

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[366.2099914550781, 0, 255.9300994873047;
  0, 366.2099914550781, 203.8130035400391;
  0, 0, 1]
distortion coefficients ir:
[0.08904761075973511, -0.2680774033069611, 0, 0, 0.0967089906334877]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
starting main loop

[kinect2_bridge] depth processing: ~7085.5Hz (0.141133ms) publishing rate: ~30.3295Hz
[kinect2_bridge] color processing: ~4428.99Hz (0.225785ms) publishing rate: ~29.9962Hz
[OpenCLDepthPacketProcessor] avg. time: 11.3291ms -> ~88.268Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.7299ms -> ~40.437Hz

wiedemeyer, 2015-03-17:
Are the depth images still black, or does it work now? What is the OS and the Kernel that you are using?

cjpedc, 2015-03-17:
I am using Ubuntu 14.04 and the kernel version is 3.16.0-31-generic

The depth images are not completely black right now. It is possible to observe different shadows using a regular image viewer. Although, I can't manage to find a good calibration. I have been collecting sets of 200 images for color (mono), ir and sync; but it takes time to calculate the parameters and when performing the depth calibration, for some images it throws the following error message (which I can overpass deleting the files corresponding to that image):

frame: ./0102_sync_depth.png
frame: ./0103_depth.png
OpenCV Error: Assertion failed (points.checkVector(2) >= 0 && (points.depth() == CV_32F || points.depth() == CV_32S)) in boundingRect, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/contours.cpp, line 1895
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/contours.cpp:1895: error: (-215) points.checkVector(2) >= 0 && (points.depth() == CV_32F || points.depth() == CV_32S) in function boundingRect

Thanks once again for the help.

wiedemeyer, 2015-03-18:
But the IR and Color calibration worked, right?
For each of the depth files there should also be a `.yaml` file with a list of coordinates `points: [...`. Does this file exist and does it contain any points.
Oh, and you need to calibrate IR before running the depth calibration.

cjpedc, 2015-03-18:
Yes, both IR and Color calibration seem to be working. 
I get the following type of .yaml files:
   0248_color_points.yaml
   0230_sync_ir_points.yaml
   0230_sync_color_points.yaml

wiedemeyer, 2015-03-18:
the depth and ir and ir_points with the same number go together. is there a list of points inside the `*_ir_points.yaml` file of one of the frames that failed during calibration? Or is the depth image of that frame zero? Did you removed all files from the calibration folder before you did the new calibration with the OpenCL depth processor, or could it be that some images are still from the previous calibration with the CPU depth processor, where all the depth images were zero?

If the illumination is bad it could be that the pattern is not detected in the image, but you would see that. You could use any pattern you want, but the PR2 checker board is pretty small and does not have so many points. And if you print out a pattern, glue it on something flat, so that the pattern is always flat and does not bend. And the amount of images per set is variable. Just make sure to cover the whole image area, place the pattern in several distances, rotations, etc.

cjpedc, 2015-03-19:
Unfortunately I deleted those files and I can't see them again, so I will have to perform another calibration and see if I get some similar problem. Although, I can remember that the depth image of those frames were not zero. 

After your first answer I performed a whole new calibration with the OpenCL depth processor and I removed all the files from the calibration folder (/kinect2_bridge/data), in order to use the ones you provided. 

The checker board from the PR2 that I talked about is 7x6x0.108m. I tried once with it and it didn't work as expected. but maybe I didn't collect enough images. Would you recommend to fix the pattern to a surface and instead move the camera?

wiedemeyer, 2015-03-20:
You can move the camera or the pattern, (or both), that does not matter.

closed 2015-05-27 by wiedemeyer