code-iai/iai_kinect2 #63: Beignet without X server as non-root user

avidbot1, 2015-03-17:
Using Ubuntu 14.04, ROS Indigo, and Beignet. I'm trying to run upstart so that kinect2_bridge is launched when my computer is started. It works when I use CPU depth processing, but when I try to use OpenCL depth processing I get this error: 

    /dev/dri/card0 not authenticated
    open("/dev/dri/card1", O_RDWR) failed: No such file or directory
    open("/dev/dri/card2", O_RDWR) failed: No such file or directory 
    open("/dev/dri/card3", O_RDWR) failed: No such file or directory
    open("/dev/dri/card4", O_RDWR) failed: No such file or directory
    open("/dev/dri/card5", O_RDWR) failed: No such file or directory
    open("/dev/dri/card6", O_RDWR) failed: No such file or directory
    open("/dev/dri/card7", O_RDWR) failed: No such file or directory
    open("/dev/dri/card8", O_RDWR) failed: No such file or directory
    open("/dev/dri/card9", O_RDWR) failed: No such file or directory
    open("/dev/dri/card10", O_RDWR) failed: No such file or directory
    open("/dev/dri/card11", O_RDWR) failed: No such file or directory
    open("/dev/dri/card12", O_RDWR) failed: No such file or directory
    open("/dev/dri/card13", O_RDWR) failed: No such file or directory
    open("/dev/dri/card14", O_RDWR) failed: No such file or directory
    open("/dev/dri/card15", O_RDWR) failed: No such file or directory
    Device open failed, aborting...
    cl_get_gt_device(): error, unknown device: ffffffff

The Beignet documentation says "Normally, beignet needs to run under X server environment as normal user. If there isn't X server, beignet provides two alternative to run: * Run as root without X. * Enable the drm render nodes by passing drm.rnodes=1 to the kernel boot args, then you can run beignet with non-root and without X."

I tried enabling the drm render nodes by editing /etc/default/grub so that 
`GRUB_CMDLINE_LINUX_DEFAULT="quiet splash drm.rnodes=1"` but I'm still getting the error. kinect2_bridge runs properly with the X server and when run as root without X. What am I supposed to do to get it to work without X as a non-root user?

Thanks

wiedemeyer, 2015-03-18:
Did you run `update-grub` after adding `drm.rnodes=1`?

avidbot1, 2015-03-18:
Yes. And the output of cat /proc/cmdline is:

    BOOT_IMAGE=/boot/vmlinuz-3.16.0-31-generic.efi.signed root=UUID=57903c34-38c5-4f2f-9d31-    
    8cd599796d58 ro quiet splash drm.rnodes=1 vt.handoff=7

wiedemeyer, 2015-03-18:
I think I got it running once, but I can't remember how. Do you have the `libdrm-intel1` installed? Maybe search for `ubuntu drm intel` or you could ask the beignet developers for help.

ais-hilgert, 2015-04-07:
I am running into the same problem, so that i am only able to use kinect2_bridge in CPU mode on the Intel NUC. Could you please provide the script you used during boot, that made it run with OpenCL?

wiedemeyer, 2015-04-07:
Simply execute kinect2_bridge as root user.
Login to root, source your workspace, run kinect2_bridge.

closed 2015-06-01 by wiedemeyer