code-iai/iai_kinect2 #64: ERROR: clGetDeviceInfo(-30), when using opencl

bg-rad, 2015-03-18:
Hello,

I am having difficulties trying to get the kinect2 to run and publish depth data properly. I was successfull in geting it to publish image data when using the CPU method however in that case it did not publish any depth data what so ever. I installed the latest nvidia drivers (340) and the ocl-icd-opencl-dev headers to replace the default nvidia ones, since those didn't seem to work when trying to make the project.

After the installation of the drivers I uninstalled all repositories which are needed to get beignet running in case of an AMD GPU (just to make sure that there won't be any conflicts occuring) after making a new workspace and installing everything and trying to run kinect2_bridge I get the following output:

parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/pa10/kinekt2/catkin_workspace/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::listDevice]  devices:
[OpenCLDepthPacketProcessor::init] ERROR: clGetDeviceInfo(-30)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetDeviceInfo
Aborted (core dumped)

Anyone know what could be wrong ? The device is detected by the usb and as stated earlier, I did manage to get it running via CPU where it only published image data (which could also be viewed using the simple ROS image_view program)

PS: I also had to include a "fix" in the CMake file when  when trying to make and install protonect since I needed to force the gcc compiler flag to use the sse2 instructions, curiously I did not have to do this the first time I installed this project....

wiedemeyer, 2015-03-18:
Just to clarify things, beignet is only for Intel GPUs, not AMD. AMD has its one OpenCL lib that comes with the catalyst driver. The lastest nvidia driver is 346.47.

For nvidia I just installed `nvidia-346`(from xorg-edgers) and `opencl-headers`. The only issue I encountered is, that the first time after a reboot of the system I have to run some program using opencl  as root (e.g. `kinect2_bridge` or `clinfo`), and afterwards it works without problems. But this could also be a problem related to my multi GPU setup.

Maybe gcc does not know your cpu, so you may need to specify the `-march=...` option. What are the specs of the system are you using? CPU, GPU, OS...

bg-rad, 2015-03-18:
Thanks for the reply,

The CPU is the Intel Core i7 CPU     860  @ 2.80GHz,

For the GPU I'm using the NVIDIA Corporation GT218 [GeForce 310] for which the Nvidia site detects the latest suitable driver to be the 340.76 one which I then got  from xorg-edgers.

I am using the 14.04 Ubuntu (trusty) release, the 32 bit version. Using the 3.16.0-031600-generic kernel (i686).

Iirc the march option is set to native, I didn't tinker around with it much but I'm sure it's no big deal. The CPU method continues to work on this catkin setup if I use the _depth_method:=cpu command after running the bridge.

Is it also maybe possible that I need to install some aditional OpenCL stuff?

wiedemeyer, 2015-03-18:
Actually you don't have to replace any nvidia opencl headers. Try to remove the packages and simply install the two packages I mentioned (`nvidia-340` and then `opencl-headers`). Hope that it works, but I never tried to run anything on a 32 bit machine.

wiedemeyer, 2015-03-20:
And you could try to run it with root rights.

bg-rad, 2015-03-23:
I tried running it with root but it didn't help.

I think I might have a hunch why it is occuring however. I ran the Openclinfo program to try and see if it even finds the OpenCL modules, and it does, however if the platform is specified as NULL it will not pick one and thus likely lead to the -30 error (this supposedly depends on the OpenCL ICD loader) and so the platforms have to specifically be chosen by the applications.

I didn't immerse myself into your code too much yet, however is it possible that in the code you specify the platforms to be opened as "NULL" and then leave it up to the OpenCL ICD loader to do the rest? If so is it possible to edit your code so that you manually choose what platforms need to get run?

bg-rad, 2015-03-23:
Okay so after a fresh install of the trusty Ubuntu, this time the 64 bit version, and after installing the nvidia 340 drivers I managed to get it running through OpenCL. At first I got the -1001 OpenCL error but this did in fact go away when I ran it as root. (I also managed to get the -1001 error instead of the -30 error on the 32bit trusty OS however that was with the 331 driver I believe, when using the 340 drivers I only ever got -30 which does not go away with root)

However upon running the program I still do not get depth registration, here is the output of the program:

parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/k2test/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: GeForce 310 (GPU)[NVIDIA Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: GeForce 310 (GPU)[NVIDIA Corporation]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:2 with serial 093508334347
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 093508334347 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 0b 0a 14 47 4b 47 31 37 32 2e 31    13......GKG172.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        .&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 093508334347
device firmware: 4.1.3911.0.7

default ir camera parameters: 
fx 366.216, fy 366.216, cx 261.026, cy 209.79
k1 0.0913037, k2 -0.266266, p1 0, p2 0, k3 0.0906861

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[366.2164916992188, 0, 261.0255126953125;
  0, 366.2164916992188, 209.7899932861328;
  0, 0, 1]
distortion coefficients ir:
[0.09130371361970901, -0.2662658989429474, 0, 0, 0.09068608283996582]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: GeForce 310 (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::init]  selected device: GeForce 310 (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: GeForce 310 (GPU)[NVIDIA Corporation]
[DepthRegistrationOpenCL::init]  selected device: GeForce 310 (GPU)[NVIDIA Corporation]
starting main loop

[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~3725.51Hz (0.26842ms) publishing rate: ~27.978Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.8252ms -> ~40.2817Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~3937.04Hz (0.253998ms) publishing rate: ~29.9085Hz
.
.
.
.
.
.
.

A user had similar issues and he managed to resolve them by removing the ocl-icd-libopencl1 and installing the nvidia-libopencl1-340. However if I do this I get the following error when trying to run catkin_make (as root) in my catkin_ws:

[ 14%] make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libOpenCL.so', needed by `/home/k2test/catkin_ws/devel/lib/libdepth_registration.so'.  Stop.
make[1]: *** [iai_kinect2/depth_registration/CMakeFiles/depth_registration.dir/all] Error 2
Built target kinect2_calibration
make[1]: *** Waiting for unfinished jobs....
[ 28%] Built target viewer
make: *** [all] Error 2

 I have the following stuff pertaining to openCL installed:
-nvidia-opencl-icd-340
-nvidia-libopencl1-340
-opencl-headers

If I install ocl-icd-libopencl1 it works but no depth gets published as seen above.

wiedemeyer, 2015-03-23:
Is there a `libOpenCL.so` in somewhere in `/usr/lib/` or `/usr/lib/x86_64-linux-gnu/`. For a system with an AMD just the symlink from `libOpenCL.so.1` to `libOpenCL.so` was missing.
Do you get depth images when running `Protonect` from the libfreenect2 library? Is the Kinect2 plugged into an USB 3.0 port? What USB controller are you using?

bg-rad, 2015-03-23:
This is for the 64bit system I'm  using atm where it's runnning through OpenCL (I'll revert back to the 32bit one and try to patch things up there later but I feel that I'm closer to getting it to work here):

I do not seem to find a specific libOpenCL.so file in /usr/lib/ or /usr/lib/x86_64-linux-gnu/ however there are many "cl" files present.

Yes the Kinect2 is plugged in and is recognized by lsusb. I'm using a PCIe usb 3.0 card by ST Lab and the libusb lib.

Here are the BUS devices as listed by lsusb (the ones pertaining to this issue).

Bus 004 Device 002: ID 045e:02c4 Microsoft Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

If I run Protonect I get the following output:

Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:2 with serial 093508334347
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    ............C..A
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    .!3U... ........
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    ................
0x0040:  31 33 00 00 00 0b 0a 14 47 4b 47 31 37 32 2e 31    13......GKG172.1
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        .&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        .&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 093508334347
device firmware: 4.1.3911.0.7
[TurboJpegRgbPacketProcessor] avg. time: 24.27ms -> ~41.2032Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.9968ms -> ~41.6723Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.9965ms -> ~41.6728Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.1955ms -> ~41.33Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.1458ms -> ~41.4151Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.8868ms -> ~41.8642Hz

wiedemeyer, 2015-03-23:
So protonect has the same issue. There is probably something wrong with the USB. It is not able to receive frames from the kinect2. Is it connected as a SuperSpeed device? You can see that in the `dmesg` logs. Maybe the USB controller is not supported, but I don't know. You should look at https://github.com/OpenKinect/libfreenect2/issues for similar issues or open an issue there.

bg-rad, 2015-03-23:
Just to be clear there seem to be seperate issues for the two different Ubuntu installations, however the latter one seems to be closer to working. Thanks for the info, btw what about the above posted issue with trying to make it with nvidia-libopencl1-340 installed because that seemed to work for another user who had pretty much the exact problem as me, however it does not seem to work here since for what ever reason I need ocl-icd-libopencl1 instead...

wiedemeyer, 2015-03-23:
i just have `ocl-icd-libopencl1` and `nvidia-opencl-icd-346` installed and it's working fine. And it also works for you, that no depth images are received is a different problem. For the other one you have to make sure that it finds the right opencl library and links to it, but since the other solution works, you could simply stick to it.

closed 2015-05-27 by wiedemeyer