code-iai/iai_kinect2 #65: Segmentation fault on ./kinect2_bridge

asheshjain399, 2015-03-27:
On running the kinect2_bridge I get a segmentation fault. I'm not sure whether this is because my laptop is not detecting kinect2, or is there a bug in kinect2_bridge. I have connected kinect2 to USB 3.0 port on Ubuntu 12.04 using the latest kernel. Below is the output of "./kinect2_bridge" executable:

ashesh@ashesh-jain:~/catkin_ws/devel/lib/kinect2_bridge$ ./kinect2_bridge 
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ashesh/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

Segmentation fault (core dumped)

asheshjain399, 2015-03-27:
Upon running ./Protonect I get the following error:

ashesh@ashesh-jain:~/Downloads/libfreenect2/examples/protonect/bin$ gdb -ex r --args  ./Protonect 
GNU gdb (Ubuntu/Linaro 7.4-2012.04-0ubuntu2.1) 7.4-2012.04
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
For bug reporting instructions, please see:
<http://bugs.launchpad.net/gdb-linaro/>...
Reading symbols from /home/ashesh/Downloads/libfreenect2/examples/protonect/bin/Protonect...done.
Starting program: /home/ashesh/Downloads/libfreenect2/examples/protonect/bin/Protonect 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeaf18700 (LWP 1215)]
[New Thread 0x7fffea717700 (LWP 1216)]

Program received signal SIGSEGV, Segmentation fault.
glfwGetWindowAttrib (handle=0x0, attrib=139266) at /home/ashesh/Downloads/libfreenect2/depends/glfw_src/src/window.c:560
560	            return window->glMajor;
(gdb) bt
#0  glfwGetWindowAttrib (handle=0x0, attrib=139266) at /home/ashesh/Downloads/libfreenect2/depends/glfw_src/src/window.c:560
#1  0x00007ffff7a17f78 in flextInit (window=0x0, bindings=0x6329b0) at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/flextGL.c:20
#2  0x00007ffff7a1b8cb in libfreenect2::OpenGLDepthPacketProcessorImpl::initialize (this=0x63e630)
    at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:469
#3  0x00007ffff7a19700 in libfreenect2::OpenGLDepthPacketProcessor::OpenGLDepthPacketProcessor (this=0x60e670, parent_opengl_context_ptr=<optimized out>, debug=false)
    at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:783
#4  0x00007ffff7a06826 in libfreenect2::OpenGLPacketPipeline::createDepthPacketProcessor (this=0x622170)
    at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/packet_pipeline.cpp:109
#5  0x00007ffff7a06941 in libfreenect2::BasePacketPipeline::initialize (this=0x622170)
    at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/packet_pipeline.cpp:43
#6  0x00007ffff7a06b02 in libfreenect2::OpenGLPacketPipeline::OpenGLPacketPipeline (this=0x622170, parent_opengl_context=<optimized out>, debug=<optimized out>)
    at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/packet_pipeline.cpp:102
#7  0x00007ffff7a1137e in libfreenect2::createDefaultPacketPipeline () at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/libfreenect2.cpp:631
#8  0x00007ffff7a140d8 in libfreenect2::Freenect2::openDevice (this=0x7fffffffdcf0, idx=0)
    at /home/ashesh/Downloads/libfreenect2/examples/protonect/src/libfreenect2.cpp:669
#9  0x0000000000401b5d in main (argc=<optimized out>, argv=<optimized out>) at /home/ashesh/Downloads/libfreenect2/examples/protonect/Protonect.cpp:58
(gdb)

Is this problem because of OpenGL? I have the following OpenGL version:
"OpenGL version string: 3.0 Mesa 8.0.4"

wiedemeyer, 2015-03-30:
Hi,
I think for the OpenGL processor you need at least OpenGL 3.3. Try out the OpenCL processor.

ipekbocegi, 2015-04-17:
I used to have the segmentation error: "Segmentation fault (core dumped)" when I want to "$ rosrun kinect2_bridge kinect2" I found a solution for my case. Maybe it helps somebody:

$ sudo gedit /etc/default/grub
Here, be sure there is no modification about nomodeset. It should stay as its original default version:
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"
$ sudo update-grub
$ sudo reboot

In my case, I used to changed it to GRUB_CMDLINE_LINUX_DEFAULT="nomodeset" for some reason, therefore kinect2_bridge was not working.

closed 2015-05-27 by wiedemeyer

vishu2287, 2015-09-29:
I am able to run ./Protonect and can view color, ir and depth images with out any error.
When I am trying to run kinect2_bridge from the root, it asks for clients. If I provide kinect2/sd/image_color_rect, it produces the messages but if I provide other topics like kinect2/qhd/image_color, the following error genetares

root@tegra-ubuntu:~/catkin_ws/devel# rosrun kinect2_bridge kinect2_bridge
parameter:
base_name: kinect2
sensor: 
fps_limit: -1
calib_path: /home/ubuntu/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
use_png: false
jpeg_quality: 90
png_level: 1
depth_method: opengl
depth_device: -1
reg_method: default
reg_devive: -1
max_depth: 12
min_depth: 0.1
queue_size: 2
bilateral_filter: true
edge_aware_filter: true
publish_tf: false
base_name_tf: kinect2
worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:5 with serial 017275643547
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
0: 017275643547 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000: 00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41 ............C..A
0x0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x0020: 0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00 .!3U... ........
0x0030: 00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00 .......... .....
0x0040: 31 33 00 00 00 04 0a 0d 47 4d 42 30 31 33 2e 31 13......GMB013.1
0x0050: 58 00 00 00 00 00 00 00 07 00 00 00 X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000: 23 26 00 00 #&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000: 23 26 00 00 #&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[... 328 lines cut ...]
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~127.059Hz (7.87038ms) publishing rate: ~14.6242Hz
Segmentation fault

I am using Jetson TK1 board .

cpu ang gpu performance were altered with maximum efficiency

Why lots of "skipping rgb packet!" and " [TransferPool::submit] failed to submit transfer " ?

why not providing depth data?
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz

Why segmentation fault after producing one message to the topics?