code-iai/iai_kinect2 #67: Black IR images

dvonhausen, 2015-03-30:
I'm getting good color images but my ir images (when running "rosrun kinect2_calibration kinect2_calibration chess5x7x0.0255375 record ir" or "Protonect" are black. I did "rostopic echo kinect2/depth/image" and it looks like I'm getting all zeros. I have two Kinect2 sensors and two cables (Kinect Adaptor for Windows) so I tried all 4 combinations to see if its a bad kinect or cables with same results in all cases. Also tried increasing and decreasing min and max values when calibrating with wame results. Is there something I'm missing?

Dave

wiedemeyer, 2015-03-30:
If protonect is not working, I would guess it's an USB issue. What OS/Kernel/USB 3.0 controller are you using?

dvonhausen, 2015-04-01:
I'm using:
Intel® Core™ i7-3630QM CPU @ 2.40GHz × 8
Ivybridge mobile
Ubuntu 14.04 LTS  64 bit
kernel:  3.16.0-031600

wiedemeyer, 2015-04-02:
That looks fine, but what USB 3.0 controller are you using? You should also look at the libfreenect2 issues, as several people experiences similar problems.

dvonhausen, 2015-04-03:
I have:
USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
and
USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)

I will check on the libfreenect2 thread for others experiencing this problem.

dvonhausen, 2015-04-03:
In this thread "https://github.com/OpenKinect/libfreenect2/issues/31", christiankerl provide a solution when building Protonect that works but I would like to utilize opencl support which doesn't compile using this line from christiankerl solution:
libfreenect2::Freenect2Device *dev = freenect2.openDefaultDevice(new libfreenect2::OpenCLPacketPipeline());

wiedemeyer, 2015-04-07:
did you enabled opencl when you build libfreenect2 with cmake?

dvonhausen, 2015-04-11:
OK it works. For others to benefit I'll recap:

I'm using:
Intel® Core™ i7-3630QM CPU @ 2.40GHz × 8
Ivybridge mobile
Ubuntu 14.04 LTS 64 bit

I originally got good color images but my ir images were black when I calibrated. I went to my libfreenect2 folder to run Protonect and got the same results.

Then I edited ~/libfreenect2/examples/protonect/Protonect.cpp to make the following changes:
		after line 37 added "#include <libfreenect2/packet_pipeline.h>"
		changed
			libfreenect2::Freenect2Device *dev = freenect2.openDefaultDevice();
		to
			libfreenect2::Freenect2Device *dev = freenect2.openDefaultDevice(new libfreenect2::OpenCLPacketPipeline());

Then I used "cmake ../../examples/protonect/ -DENABLE_OPENCL=ON" instead of "cmake ../../examples/protonect/ -DENABLE_CXX11=ON" and now Protonect works with opencl. Note: I'm using Beignet 1.0.2 in my case and went back to using the 3.13.0-44-generic kernel.

closed 2015-04-11 by dvonhausen

dvonhausen, 2015-04-28:
My mistake on the previous message. It should be:
after line 37 I added: "#include <libfreenect2/packet_pipeline.h>"