code-iai/iai_kinect2 #69: undefined reference to symbol 'glReadBuffer'

navid33, 2015-03-30:
Hi,
when I run catkin_make -DCMAKE_BUILD_TYPE="Release" I am getting this errors:

[ 83%] Built target kinect2_bridge_nodelet
Linking CXX executable /home/navidd33/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
/usr/bin/ld: /usr/local/lib/libfreenect2.a(opengl_depth_packet_processor.cpp.o): undefined reference to symbol 'glReadBuffer'
//usr/lib/x86_64-linux-gnu/mesa/libGL.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [/home/navidd33/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Any ideas?
Thanks

wiedemeyer, 2015-03-31:
Why do you have a `libfreenect2.a`? Normally libfreenect2 should create a `.so` file, to which kinect2_bridge links against.

navid33, 2015-03-31:
well I dont no why! I am new to Ubunta and I did everything step by step according to instructions.

wiedemeyer, 2015-03-31:
I really don't know what is going wrong. You should try to figure out why there is no `libfreenect2.so`. What changes did you do to the CMakeLists of libfreenect2?

navid33, 2015-03-31:
I have made 2 changes on CMakeLists.
1)#libUSB same as instructions
2)i have changed ADD_LIBRARY(freenect2 SHARED ${SOURCES})   TO
ADD_LIBRARY(freenect2 STATIC ${SOURCES})

because I had this error and  I fixed it with that. 
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libturbojpeg.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status

wiedemeyer, 2015-03-31:
Ok, don't change it to STATIC. The other error could probably be fixed by adding a symlink `sudo ln -s /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0.0.0 /usr/lib/x86_64-linux-gnu/libturbojpeg.so`, like described in the libfreenect2 install instructions.

navid33, 2015-03-31:
I have already done sudo ln -s /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0.0.0 /usr/lib/x86_64-linux-gnu/libturbojpeg.so . but it didnt fix the error

wiedemeyer, 2015-03-31:
Try a clean build of libfreenect2. Remove the build folder, or make a different one, run cmake, make and make install. You could also check the issue page from libfreenect2 for similar issues. If nothing helps you could open an issue there.

closed 2015-04-29 by navid33