code-iai/iai_kinect2 #70: All topics are in the root namespace

mitchellwills, 2015-04-01:
Here https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L455
A slash is prefixed to the the topic names which causes all of them to end up in the root namespace making it ignore namespaces specified in launch files.

wiedemeyer, 2015-04-01:
I know, this behavior is intended. You can change the name with the `base_name` parameter. Changing it to use the namespace is already on my todos.

mitchellwills, 2015-04-01:
It should just be a matter of not prepending the leading slash. I can submit a PR if you would like

wiedemeyer, 2015-04-02:
It actually not that easy. But I modified it to use namespaces instead of this `base_name`. You can find the code in the devel branch.

mitchellwills, 2015-04-02:
I'm not sure why you need to do anything special to handle namespaces. ros::init will automatically read the namespace and then if you don't prefix a topic name with a slash the namespace will automatically be prefixed. By making it a node parameter it makes it impossible to use the roslaunch ns argument

wiedemeyer, 2015-04-02:
I don't get what you mean. In the devel branch `ros::init` is used. The only thing done there is to add default namespace parameter in the beginning of the parameters so that the node can be run directly without specifying a namespace and works the same way as if it has been run through the launch file where a default namespace is set.
If you want a different namespace just add a `__ns:=...` to the nodes parameter, which will override the default one, or change the `ns` argument of the launch file.

mitchellwills, 2015-04-05:
Sorry for the confusion with ros::init. I just haven't seen a node that adds a default namespace parameter before. To me this behavior is very confusing (although I guess I could understand if you were doing it for backwards compatibility, although this could by solved by simply prefixing all topics with kinect2). Normally if I run a node and and it normally topics A/B, A/C and A/D and then I run the node with a namespace (N) specified I would expect to get topics N/A/B, N/A/C and N/A/D; however, with your modifications I get N/B, N/C and N/D. If the kinect2 was always there, even if a namespace was specified then it would make more sense in my mind (this is what usb_cam does). Additionally, the ns argument in the launch file is not super useful as the launch files can just be called in that namespace.

wiedemeyer, 2015-04-05:
That makes sense. I will make the namespace optional and make the base name changeable through a parameter. So that it is A/B and when a namespace is given, it will be N/A/B.

wiedemeyer, 2015-05-18:
Ok, I finally got time to change it. It should now be A/B or N/A/B if a namespace is given. It is in the devel branch

closed 2015-05-27 by wiedemeyer