code-iai/iai_kinect2 #71: Camera only mode

superslimjimmy, 2015-04-02:
Great work on the the ROS driver for the kinect v2.

Even with OpenCL, using the kinect v2 vs. a standard webcam has a relatively high computation load due to the depth registration. Right now, I only need video and not depth information from the kinect. 

Any thoughts on a good way to run in "camera only" mode with the kinect v2 and save on computation load?

wiedemeyer, 2015-04-02:
Well, kinect2_bridge only compute what is necessary. If you just subscribe to one of the color topics, just processing needed for that topic is done. For example subscribing to `rgb_rect/image/compressed` rectifies the image and compresses it, but nothing else.
The CPU load when not subscribing any topics is created by libfreenect2, it constantly receives images from the sensor. One CPU consuming part are the color images. These are jpeg encoded, so libfreenect2 needs to decode them.

wiedemeyer, 2015-06-01:
Right now a camera only mode is not possible with libfreenect2 without modifications. But the normal use case for the Kinect2 is to have access to color and depth.

closed 2015-06-01 by wiedemeyer