code-iai/iai_kinect2 #72: Recommended graphics card for high resolution point clouds?

bg-rad, 2015-04-08:
Hello,

I managed to get the drivers running with the depth processing being done with opencl and the depth registration being done with cpu (some buffer errors appear if trying to get it to work through opencl), the graphics card I used for this task was the Nvidia 8800 GTS which is quite an old graphics card. I manage to display high/low resolution images with it just fine (most of the time...), the viewer runs on 30 fps using the lower resolution images and around 11 fps when displaying the higher resolution ones but if I try to display the high resolution point clouds everything freezes/crashes and I have to manually reset the computer (the screen turns either green or black and just stays there while 2 lighs flicker on the keyboard).

I assume this has something to do with the graphics card since it is quite old and also has some flickering going on. Do you maybe know what range Nvidia cards would do a decent/good job with it or rather what card are you using if you can get high resolution point cloud to display properly?

wiedemeyer, 2015-04-09:
Hi,
if you look at the images using the viewer on the same machine the kinect is connected to, you might want to add the `raw` parameter of the viewer. This way the images don't get compressed and decompressed, which saves a lot CPU time.
Well I used a nvidia 640, a 770 and a 970. They all do the job fine. What are the timings printed on the console? Just past some of the console output here.
What CPU are you using? How is the CPU load? While running kinect2 you could also look at the cpu load with htop for example.

bg-rad, 2015-04-13:
Never mind, getting a new graphics card fixed all the issues I had at that moment. Got a 960 and things are running quite well for the time being.

closed 2015-04-13 by bg-rad