code-iai/iai_kinect2 #73: cd ../build: No such file or directory in installation on Ubuntu 14.04

ipekbocegi, 2015-04-09:
Hi everyone,

I am trying to download and run iai_kinect2. I follow the steps in Readme file. However I stuck at the very beginning. I am newbie at ubuntu, ros and so on. So any help is appreciated.

1. I installed ROS indigo.
2. I set the environment
3. I have now a folder name "catkin_ws" in home
4. I wrote the followings:
cd ~
sudo apt-get install -y build-essential libturbojpeg libtool autoconf libudev-dev cmake mesa-common-dev freeglut3-dev libxrandr-dev doxygen libxi-dev libopencv-dev
sudo ln -s /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 /usr/lib/x86_64-linux-gnu/libturbojpeg.so
git clone https://github.com/OpenKinect/libfreenect2
5. No problem until here
6. I changed CMakeList.txt as told.
7. I wrote the followings:
cd libfreenect2/depends
./install_ubuntu.sh
8. No problem. Now I have a folder "libfreenect2" in home.
9. Here it comes. When I write "cd ../build" it says:
../build: No such file or directory
Because there is actually no such build folder in libfreenect.
10. There is only one "build" folder which is in libfreenect2/depends/glfw_src/build not in libfreenect2/

My question is:
Should I move on with 
1.
cd depends//glfw_src/build
mkdir linux
cd linux
cmake ../../examples/protonect/ -DENABLE_CXX11=ON
make && sudo make install

or

2.
cd libfreenect2
mkdir build
cd build
mkdir linux
cd linux
cmake ../../examples/protonect/ -DENABLE_CXX11=ON
make && sudo make install

or

3.
What?

Thank you for your patience.

wiedemeyer, 2015-04-09:
It got recently removed from the repo. Just create a new folder named build. I will change that in the README.

ipekbocegi, 2015-04-09:
Thank you for your quick response and help. It works properly now

closed 2015-04-09 by ipekbocegi