code-iai/iai_kinect2 #74: [rospack] Error: package 'kinect2_bridge' not found during installation on Ubuntu 14.04

ipekbocegi, 2015-04-09:
Hi again, 
I still follow readme file for installation as I mention in issue #73 yet there is another problem occured. Any help is appreciated

10. I move on with the suggestion in issue #73  No problem during make and install now.
11. I move on with readme file step 4, by compiling these: 

cd ~/catkin_ws/src/
git clone https://github.com/code-iai/iai_kinect2.git
cd iai_kinect2
rosdep install -r --from-paths .
cd ~/catkin_ws
catkin_make -DCMAKE_BUILD_TYPE="Release"

12. Still no problem. I checked 
camera by testing Protonect by writing:

cd ~/libfreenect2/examples/protonect
./bin/Protonect

It gives me rgb output but not ir or depth. Besides [Freenect2Impl] found valid Kinect v2 @2:14 with serial 508765442542

13. Here is the problem:
When I write "rosrun kinect2_bridge kinect2_bridge" it says 
[rospack] Error: package 'kinect2_bridge' not found

14. I know there exist such bridge, checked the files in catkin_ws. I know usb 3.0 port is connected. (I checked with "lsusb" and also with step 12)
15. I gave permission as told in readme file by creating a rule. Checked the id numbers. It is consistent
16. I have already installed "ocl-icd-opencl-dev"
17. I did not do anything about calibration or kinect2_bridge/data/<serialnumber> or anything about Beignet. 

My questions are:
1. how can I make "rosrun kinect2_bridge kinect2_bridge" work?
2. how can I see ir and depth outputs?

 I checked other issues including closed ones but could not find a cure. Thanks a lot

Note: I am working on an Asus X555LN laptop with Ubuntu 14.04 LTS and kernel 3.16.0-33-generic
I think I am using Intel graphic card not the nvidia. There is no Nvidia in additional drivers. "ubuntu-drivers list" gives me only "bcmwl-kernel-source". When I write "lspci" it says "00:02.0 VGA compatible controller: Intel Corporation Broadwell-U Integrated Graphics (rev 09)"

ipekbocegi, 2015-04-09:
I just repeat step 4 in readme file and realized that when I write "rosdep install -r --from-paths ." it says:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
registration_viewer: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_bridge: Cannot locate rosdep definition for [depth_registration]
kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

But when I move on "catkin_make -DCMAKE_BUILD_TYPE="Release"" it says:
Base path: /home/ipek/catkin_ws
Source space: /home/ipek/catkin_ws/src
Build space: /home/ipek/catkin_ws/build
Devel space: /home/ipek/catkin_ws/devel
Install space: /home/ipek/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ipek/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ipek/catkin_ws/build"
####
[ 14%] Built target kinect2_calibration
[ 57%] Built target depth_registration
[ 71%] Built target viewer
[ 85%] Built target kinect2_bridge_nodelet
[100%] Built target kinect2_bridge

bg-rad, 2015-04-09:
Did you source the catkin workspace properly? You have to run source devel/setup.bash when in the catkin workspace, or edit it to the .bashrc file so it automatically sources it when you open the terminal.

ipekbocegi, 2015-04-09:
* I wrote "sudo nano ~/.bashrc" and add "source ~/catkin_ws/devel/setup.bash" at the very end of the file, under the "source /opt/ros/indigo/setup.bash" command save and exit.

* I restarted the pc. Checked with "echo $ROS_PACKAGE_PATH" and see: 
/home/ipek/catkin_ws/src:/opt/ros/indigo/share:/opt/ros/indigo/stacks

* Opened a new terminal and wrote "rosrun kinect2_bridge kinect2_bridge". It says:

[ERROR] [1428586803.429417348]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

* I stop with CTRL+Z and write "roscore" since it says about master. In a new terminal I wrote "rosrun kinect2_bridge kinect2_bridge" again and here is the new problem:

parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ipek/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)

Now it is about OpenCL and device serial right? Any solutions?

bg-rad, 2015-04-09:
Try running it in root, that fixes the -1001 error in my case (some weird Nvidia permission issue where even if the exceptiosn are written in the file).

Type sudo -i, then go to your catkin workspace folder, source it again (or source the whole .bashrc file again in root ie: source /home/usr_name/.bashrc) and run it with root.

wiedemeyer, 2015-04-09:
And you don't have to use `sudo` when editing your `~/.bashrc`. That file should already belong to your user.

ipekbocegi, 2015-04-09:
Ok here what I wrote:
sudo -i
cd ~
cd ..
cd home/ipek/catkin_ws
source ~/.bashrc
rosrun kinect2_bridge kinect2_bridge

and result is:
rosrun: command not found

Did I miss something very serious? There is no rosrun or roscore

When I write ros and double tab I see these:
rosco                                            rosdistro_reformat
rosdep                                          rosinstall
rosdep-source                              roslocate
rosdistro_build_cache                  rosversion
rosdistro_migrate_to_rep_141     rosws
rosdistro_migrate_to_rep_143

wiedemeyer, 2015-04-09:
you need to source the `.bashrc`of your user, not the one from the root user. `~` always points to the home directory. But if you are logged in as root, it points to `/root` and not to `/home/usr_name`. That is why @bg-rad wrote a absolute path in the example.

ipekbocegi, 2015-04-10:
* Ok I see. Now I tried:
sudo -i
source /home/ipek/.bashrc

It says:
-bash: /root/catkin_ws/devel/setup.bash: No such file or directory

But I manually go there and there exist such file which contains followings:
#!/usr/bin/env bash
# generated from catkin/cmake/templates/setup.bash.in

CATKIN_SHELL=bash

# source setup.sh from same directory as this file
_CATKIN_SETUP_DIR=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)
. "$_CATKIN_SETUP_DIR/setup.sh"

* I wrote exit and become user again. Then wrote "roswtf" it says:
Loaded plugin tf.tfwtf
No package or stack in context
===============================================================================
Static checks summary:

Found 1 error(s).

ERROR Not all paths in PYTHONPATH [/home/ipek/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/indigo/lib/python2.7/dist-packages] point to a directory: 
 * /home/ipek/catkin_ws/devel/lib/python2.7/dist-packages

===============================================================================

ROS Master does not appear to be running.
Online graph checks will not be run.
ROS_MASTER_URI is [http://localhost:11311]

Are they related somehow?

wiedemeyer, 2015-04-10:
open a terminal and start roscore, then open another terminal and do
```
sudo -i
source /home/ipek/catkin_ws/devel/setup.bash
rosrun kinect2_bridge kinect2_bridge
```

ipekbocegi, 2015-04-10:
Sorry for bothering you but..

ipek@ipek:~$ sudo -i
[sudo] password for ipek: 
root@ipek:~# source /home/ipek/catkin_ws/devel/setup.bash
-bash: /home/ipek/catkin_ws/devel/setup.bash: No such file or directory
root@pek:~# rosrun kinect2_bridge kinect2_bridge
rosrun: command not found
root@ipek:~#

bg-rad, 2015-04-10:
do sudo -i then do cd .. then source /home/ipek/catkin_ws/devel/setup.bash

ipekbocegi, 2015-04-10:
ipek@ipek:~$ sudo -i
[sudo] password for ipek: 
root@ipek:~# cd ..
root@ipek:/# source /home/ipek/catkin_ws/devel/setup.bash 
root@ipek:/# rosrun kinect2_bridge kinect2_bridge
modprobe: FATAL: Module nvidia not found.
[ERROR] [1428660466.686761287]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

At least this is something new

bg-rad, 2015-04-10:
If you use Nvidia, do you have the proper drivers installed? Make sure to have at least the 340 ones from xorg-edgers installed (the required OpenCL stuff comes packaged with them), also if you do try re-installing them as root, I remember having to reinstall one of the seperate modules as root because it wouldn't link properly if trying to run things as root user, but I forgot which one.

Also try running modprobe nvidia while in root.

If you use Intel then make sure Beignet + all the required libraries are installed.

ipekbocegi, 2015-04-10:
Last week I tried individually installing NVidia-340, 331,346, 304, prime and so on but all of them resulted with horror. Each one is seemed to be installed properly, but when I reboot, every time I faced with black screen. I reinstalled Ubuntu for 15 times.. Now I afraid installing any Nvidia.
On my laptop it says I have Geforce 840M. Also when I write "lspci" it says "00:02.0 VGA compatible controller: Intel Corporation Broadwell-U Integrated Graphics (rev 09). 
In additional drivers there is no NVidia so I guess I dont have any NVidia now. 
So I move on with Intel and install Beignet and its libraries?
Thank you for help

bg-rad, 2015-04-10:
Try removing the nouveau (default, open-source) Nvidia drivers, maybe they are conflicting with the proprietary one. Also try and find if some kernel modules are screwing it up for you and blacklist them. What desktop environment do you use?

For the drivers make sure to get them from the xorg-edgers repo, do

sudo add-apt-repository ppa:xorg-edgers/ppa 
sudo apt-get update

then install the desired graphics card drivers, my current card only suppords the 340 ones, check the nvidia site to see which ones your card supports and then do sudo apt-get install yourversion. I think there is this bumblebee thing for laptop nvidia graphics cards, maybe check into that.

Yes if you are planning to use the Intel graphics card you need to install Beignet and the required libraries.

ipekbocegi, 2015-04-13:
Hi again,
* I tried installing Beignet but could not succeed. So decided to try Nvidia once more..
* I use default desktop environment.
* I used to have only wifi driver in additional drivers. After I wrote:
sudo add-apt-repository ppa:xorg-edgers/ppa
sudo apt-get update
I get Nvidia drivers 340,346,349 and X.org thing in additional drivers. Please see the attachment. Xorg is selected by default.

* I checked the blacklist.conf file in etc/modprobe.d and it does not says that nouveau is blacklisted. I add this file also in attachment.

* $ grep nouveau /etc/modprobe.d/* gives:
/etc/modprobe.d/nvidia-304-updates_hybrid.conf:blacklist nouveau
/etc/modprobe.d/nvidia-304-updates_hybrid.conf:blacklist lbm-nouveau
/etc/modprobe.d/nvidia-304-updates_hybrid.conf:alias nouveau off
/etc/modprobe.d/nvidia-30
[image]
4-updates_hybrid.conf:alias lbm-nouveau off

* $ ubuntu-drivers list gives:
nvidia-340
bcmwl-kernel-source
nvidia-346
nvidia-349

* Nvidia web site says that I should use version 346.59 

So, I really afraid to chose that one in additional drivers and make apply changes and blacklisting nouveau. I think I will get the black screen when I reboot.

Please guide me what should I do now? Is there also any trick about kernel thing or so??

[image]

bg-rad, 2015-04-13:
Ok, what I would do in your case is to try and block certain kernel-modules which may interfere with the Nvidia drivers working.

Try adding the following lines to the end your modprobe.d file.

blacklist vga16fb
blacklist nouveau
blacklist rivafb
blacklist nvidiafb
blacklist rivatv
blacklist lbm-nouveau

You can also purge the x.org server drivers after installing the Nvidia drivers (you can obviously reinstall them later if the proprietary drivers won't work), simply write

sudo apt-get purge --remove xserver-xorg-video-nouveau

If you have the xorg-edgers properly set up then simple write

sudo apt-get install nvidia-346 or nvidia-340 (I managed to get it to run the 340 drivers but I believe most users use the 346 drivers and those are also your recommended ones)

Now it is possible that you will get a black screen but it should be an easy fix, simply enter the GRUB boot menu, run Ubuntu in safe mode, select the option to boot the terminal. Now I think you will have to manually load your filesystem by writing

sudo mount -o remount, rw/

login into your user name and then simply remove all Nvidia related things by writing

sudo apt-get remove --purge nvidia-*

and now you can simply reinstall the xserver and remove the blocked modules from the blacklist file.

I'm not familiar with the nature of your work and what you are trying to achieve by using the drivers however remember you can also run it through the CPU without using the graphics card (however the frame rate will be rather poor).

Just make sure that you have Eigen installed (if not install it and re- build/make your workspace) and when running the bridge add the flag parameters _reg_method:=cpu and _depth_method:=cpu or you can simply edit the code a bit and see that those are the default parameters when starting.

Good luck

ipekbocegi, 2015-04-14:
@bg-rad thank you for your support but I have bad news.. Black screen again.. I removed Nvidia-346 to get my screen back. Now I have Xorg again. 
I can not run kinect2_bridge (modprobe: FATAL: Module nvidia not found), that is why I feel I have to settle nvidia.
How can I run it through CPU? 
Whether I have Xorg or Nvidia346 it does not matter I can not run kinect2_bridge..
Thank you for your patience

bg-rad, 2015-04-14:
Just to clarify,if I understood correctly earlier you mentioned that you have another workstation where it's using an intel graphics card, on that one you should try and install beignet + the libraries. 

If for what ever reason you do not manage to get your drivers to run just try and running the bridge with the depth and registration operating through the CPU and see if it works then.

rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=cpu

ipekbocegi, 2015-04-14:
Sorry for confusion,
I have only one laptop. It has Intel Broadwell-U and Nvidia Geforce 840m graphic card in it as hardware. It uses intel driver as default. I tried to install Nvidia346 driver but could not survive. I tried Beignet to install but coul not install it..

Having roscore in another terminal, $ rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=cpu gives me:

parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ipek/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: cpu
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:4 with serial 508765442542
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 508765442542 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  03 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 01 12 0f 47 4d 41 39 38 33 2e 31    31332e2e2e2e2e2e474d413938332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 508765442542
device firmware: 4.3.3912.0.7

default ir camera parameters: 
fx 365.64, fy 365.64, cx 259.634, cy 205.005
k1 0.0949731, k2 -0.271987, p1 0, p2 0, k3 0.092491

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
[DepthPacketStreamParser::handleNewData] not all subsequences received 0

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.6398010253906, 0, 259.6340026855469;
  0, 365.6398010253906, 205.0050964355469;
  0, 0, 1]
distortion coefficients ir:
[0.09497310221195221, -0.2719865143299103, 0, 0, 0.09249096363782883]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052;
  0;
  0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
/home/ipek/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: symbol lookup error: /home/ipek/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: undefined symbol: _ZN17DepthRegistration4initERKN2cv3MatERKNS0_5Size_IiEES3_S7_S3_S3_S3_ffi

bg-rad, 2015-04-15:
Maybe some library is missing or it is not linked properly, maybe you replaced something when you tried to install the Beignet libraries?

wiedemeyer, 2015-04-15:
Seems to be a linking issue. I does not find the `init` method of the depth calibration. Maybe it helps to clean your workspace, `rm -rf devel build` in your catkin workspace and then `catkin_make`.

ipekbocegi, 2015-04-15:
cleaned workspace and made catkin_make. I think I see the problem.
Somewhere during process it says:
...
...
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
....
....
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
....
....
It finishes catkin_make without an error though

There is nothing like "sudo apt-get install openni-dev" 

Is it the problem??

wiedemeyer, 2015-04-15:
no, that is not a problem. you don't need openni-dev. So is it running now?

ipekbocegi, 2015-04-15:
Nope..  
$ rosrun kinect2_bridge kinect2_bridge gives still the -1001 error
$ rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=cpu gives continiously something like

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
starting main loop

[kinect2_bridge] depth processing: ~8927.85Hz (0.112009ms) publishing rate: ~3.32639Hz
[kinect2_bridge] color processing: ~4948.01Hz (0.202101ms) publishing rate: ~27.6091Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.9357ms -> ~47.7654Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~9010.32Hz (0.110984ms) publishing rate: ~3.32158Hz
[kinect2_bridge] color processing: ~3720.58Hz (0.268775ms) publishing rate: ~29.562Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!

I did $ rosdep install -r --from-paths .
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
registration_viewer: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_bridge: Cannot locate rosdep definition for [depth_registration]
kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

Is it the problem???

wiedemeyer, 2015-04-15:
no, it is fine, it can't find the packages, because they are all part of iai_kinect2, which is not installed at that point.
And if you look at the output, kinect2_bridge works. it receives color and depth.

ipekbocegi, 2015-04-15:
So are we done here? I live with -1001 error and work with "rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=cpu" for my further studies?? 

I wish those packages would be installed.. Is there anything that I can do for them? Actually I followed readme file so I assumed them to be installed until now?

By the way " $ rosrun registration_viewer viewer kinect2 sd cloud "
topic color: /kinect2/rgb_lowres/image
topic depth: /kinect2/depth_lowres/image
starting receiver...

Just waits like that and does not give me any result. Is it related?

wiedemeyer, 2015-04-15:
The viewer should work, does `rosrun registration_viewer viewer kinect2 sd image` work?
Now you know that it is working, but I would recommend to not use CPU based depth processing. So now you can try to get OpenGL or OpenCL working.

ipekbocegi, 2015-04-15:
Nope it does not work either. Other modes ir or hd also do not work. Just says "starting receiver..." but no viewer.
Ok I will try to get images using regular "rosrun kinect2_bridge kinect2_bridge" by handling that 1001 error. I will give info as soon as I find a solution. 
Now I can get the depth and color data via cpu. 
Thanks for all the help
Best wishes

wiedemeyer, 2015-04-15:
do you receive images through `rostopic hz /kinect2/depth/image` or `rostopic hz /kinect2/rgb/image`?

ipekbocegi, 2015-04-15:
Nope..
$ rostopic hz /kinect2/depth/image
subscribed to [/kinect2/depth/image]
Ctrl+C
$ rostopic hz /kinect2/rgb/image
subscribed to [/kinect2/rgb/image]
Ctrl+C

ipekbocegi, 2015-04-16:
Solved
I format the Ubuntu 14.04
I made software updates and update my kernel from 3.16.0.30 to 3.16.0.34
Did not install or change anything about Nvidia or Beignet
Installed Ros
Followed Readme file
Install libfreenect2, iai_kinect2, give permission in udev
Did not externally install PCL or OpenCV or anything else.

$ rosrun kinect2_bridge kinect2_bridge
...
[kinect2_bridge] depth processing: ~9027.77Hz (0.110769ms) publishing rate: ~29.9997Hz
[kinect2_bridge] color processing: ~4492.67Hz (0.222585ms) publishing rate: ~29.9997Hz
[OpenGLDepthPacketProcessor] avg. time: 4.29621ms -> ~232.764Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.9252ms -> ~47.7892Hz
[kinect2_bridge] depth processing: ~8536.58Hz (0.117143ms) publishing rate: ~29.9612Hz
[kinect2_bridge] color processing: ~4187.23Hz (0.238821ms) publishing rate: ~29.9612Hz
[OpenGLDepthPacketProcessor] avg. time: 4.20771ms -> ~237.659Hz
...

It is very slow but at least I get the output.

Thank you everybody

closed 2015-04-16 by ipekbocegi

wiedemeyer, 2015-04-16:
why slow, the output looks fine.

ipekbocegi, 2015-04-16:
One more update: 
No need to format. It gave the same error when I close the terminal and open new one. 
Solution:
$ cd catkin_ws
$ source devel/setup.bash
$ roscore
In a new terminal
$ rosrun kinect2_bridge kinect2_bridge
Now works.

or
$ sudo gedit ~/.bashrc
write this at the very end:
source ~/catkin_ws/devel/setup.bash
save and close
$ roscore
In a new terminal
$ rosrun kinect2_bridge kinect2_bridge
Now works.

Best wishes

manolovaza, 2015-04-21:
I am running a kernel 3.13.0-49-generic, could that be the reason why I cannot view the IR and depth images with ./Protonect and get always this message: 

[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)

ipekbocegi, 2015-04-21:
I don't think it is related with kernel version. I think it is about source issue. What do you see at the very bottom of your bashrc file? You can check with 
$ cd ~
$ sudo nano .bashrc
There should be these lines:
source /opt/ros/indigo/setup.bash
source ~/catkin_ws/devel/setup.bash
( I assume/guess you installed everything in catkin_ws )

manolovaza, 2015-04-21:
@ipekbocegi  have it like these:

source /opt/ros/indigo/setup.bash
source ~/ROS_workspace/devel/setup.bash
export ROS_HOME=~/ROS_workspace/roshome

ipekbocegi, 2015-04-21:
What does "export | grep ROS" give?

manolovaza, 2015-04-21:
@ipekbocegi I have this output:

research@toshiba:~$ export | grep ROS
declare -x CATKIN_TEST_RESULTS_DIR="/home/research/ROS_workspace/build/test_results"
declare -x CMAKE_PREFIX_PATH="/home/research/ROS_workspace/devel:/opt/ros/indigo"
declare -x CPATH="/home/research/ROS_workspace/devel/include:/opt/ros/indigo/include"
declare -x LD_LIBRARY_PATH="/home/research/ROS_workspace/devel/lib:/home/research/ROS_workspace/devel/lib/x86_64-linux-gnu:/opt/ros/indigo/lib/x86_64-linux-gnu:/opt/ros/indigo/lib"
declare -x PATH="/home/research/ROS_workspace/devel/bin:/opt/ros/indigo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games"
declare -x PKG_CONFIG_PATH="/home/research/ROS_workspace/devel/lib/pkgconfig:/home/research/ROS_workspace/devel/lib/x86_64-linux-gnu/pkgconfig:/opt/ros/indigo/lib/x86_64-linux-gnu/pkgconfig:/opt/ros/indigo/lib/pkgconfig"
declare -x PYTHONPATH="/home/research/ROS_workspace/devel/lib/python2.7/dist-packages:/opt/ros/indigo/lib/python2.7/dist-packages"
declare -x ROSLISP_PACKAGE_DIRECTORIES="/home/research/ROS_workspace/devel/share/common-lisp"
declare -x ROS_DISTRO="indigo"
declare -x ROS_ETC_DIR="/opt/ros/indigo/etc/ros"
declare -x ROS_HOME="/home/research/ROS_workspace/roshome"
declare -x ROS_MASTER_URI="http://localhost:11311"
declare -x ROS_PACKAGE_PATH="/home/research/ROS_workspace/src:/opt/ros/indigo/share:/opt/ros/indigo/stacks"
declare -x ROS_ROOT="/opt/ros/indigo/share/ros"
declare -x ROS_TEST_RESULTS_DIR="/home/research/ROS_workspace/build/test_results"

wiedemeyer, 2015-04-21:
I guess you have an nvidia card, right? Either try to run it as root with sudo or try to replace the name in `/etc/OpenCL/vendors/nvidia.icd` by the full path to that file. If you don't know where it is, run `find /usr -name "libnvidia-opencl.so.1"`.

ipekbocegi, 2015-04-21:
ROS declerations seem ok. I think problem is about "export ROS_HOME=~/ROS_workspace/roshome"
I am not sure but maybe you may try this "export ROS_HOME=~/ROS_workspace/roshome:$ROS_HOME"

manolovaza, 2015-04-21:
@wiedemeyer in by setting I have the next graphic card and in vendor Intel64:
-Graphics Intel® Ivybridge Mobile 
-research@toshiba:/etc/OpenCL/vendors$ ls
intel64.icd  intel.icd

@ipekbocegi I tried it, but I have the same output:
research@toshiba:~$ rosrun kinect2_bridge kinect2_bridge 
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/research/ROS_workspace/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)

manolovaza, 2015-04-21:
@ipekbocegi 
@wiedemeyer  
But I it runs perfect with:
rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu _depth_method:=cpu

But I read it is not recommended, and when I tried the calibration I couldn't save the images, maybe it s overloading.

wiedemeyer, 2015-04-21:
did you tried running it with sudo?
you should look at libfreenect2 issues. There are some for Intel GPU and OpenCL, maybe there are some similar issues described.

ipekbocegi, 2015-04-21:
My problem solved with proper source but yours..
I do not know, maybe you may try somehow get rid of extra path that you described in bashrc and combine all catkin workspace in one place and make it. Why do you need that path? 
You may recheck whether you define that path correctly or not. Like try this one:  "export ROS_HOME = $ROS_HOME: ~/ROS_workspace/roshome" 
You may also try restart your pc and try sudo apt-get update. 
I used to format ubuntu when I was out of choice and reinstalled everything then it worked. Maybe you miss something very simple or you delete unconsciously something. 
By the way, you did not changed anything in /etc/default/grub ? If you set the "quick splash" to "nomodeset" it causes problem too

manolovaza, 2015-04-21:
@wiedemeyer i did these, but I will keep checkint libfreenect issues:
sudo -i
source /home/ipek/catkin_ws/devel/setup.bash
rosrun kinect2_bridge kinect2_bridge

@ipekbocegi I tried "export ROS_HOME = $ROS_HOME: ~/ROS_workspace/roshome" but doesn't work, maybe you are right and I missed something during installation. I haven't touch the grub, I just tried updating the kernel to 3.16 because I thought it was the problem, it didnt work, and I errased it becuase it made a mess.

manolovaza, 2015-04-22:
@ipekbocegi I have `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"`, I wouldn't like to format Ubuntu unless is completely necessary. I will try to follow all the installation steps again to see if I forgot a step.
@wiedemeyer I have checked the libfreenect2 issues but so far I haven't found a solution. I will keep trying, for now I am running the kinect2 it with the cpu and I can visualize everything but it seems slow and overloaded.

manolovaza, 2017-03-26:
Hello Timo,

thank you very much for your nodelet iai and sorry to write you an email 
to your university account, I saw the /*Do not write emails*/ message 
after I did so. Your nodelet has very useful for my research. I am 
citing it on a paper that I am currently writing. Just wanted to ask you 
a question since the process from a 2.D to 3D image was a black box for 
me and now I am having problems to describe it.

Your nodelet performs the pinhole projection and the radial camera 
distortion correction (barrel lens distortion), but not the alignment of 
the depth an color outputs? The alignment is done with the calibration 
as well as the distance difference errors mainly at the corners of the 
depth image, am I right?

Hope not to disturb you. Best regards,