code-iai/iai_kinect2 #75: Reducing CPU load when num subscribers = 0

linknum23, 2015-04-09:
Are any plans either to put the libfreekinect2 interface into a "standby" state when there are no subscribers? 
I have the (possibly odd) use case of only needing several Kinect2 frames every minute. The libfreekinect2 seems to use an entire processor just receiving Kinect2 data that doesn't go anywhere.

linknum23, 2015-04-09:
I tried adding some code to stop and start libfreekinect2 device in Kinect2Bridge run(). Sadly it looks like there is a problem (probably in libfreekinect2) where it doesn't handle multiple stop/start sequences properly. It looks like the depth processor gets out of sync from all of the messages like those below: **[DepthPacketStreamParser::handleNewData] working buffer full, resetting it!**
**[DepthPacketStreamParser::handleNewData] remaining data should be 8 bytes, but is 5784**

Below is the attempted code. Does anyone have any ideas on how to do this properly?
```c
std::vector<Status> status(COUNT, UNSUBCRIBED);

for(; running && ros::ok();)
{
    // lets keep the kinect running only when there are subscribers
    bool hasSubscribers = updateStatus(status);
    if(inStandby && hasSubscribers)
    {
	     // start the kinect interface back up
	     std::cout << "[kinect2_bridge] restarting kinect"<< std::endl << std::flush;
	     device->start();
	     inStandby = false;
    }
    else if(!inStandby && !hasSubscribers)
    {
	     // shut the kinect interface down
	     std::cout << "[kinect2_bridge] stopping kinect"<< std::endl << std::flush;
	     device->stop();
	     inStandby = true;
    }

     // loop code below...
```

wiedemeyer, 2015-04-10:
Hi,
right now I have plenty of things to do, but I could look at it later. But it might be an issue of libfreenect2, so open an issue there could be a good idea. Meanwhile you could try to create a simple example using `Protonect` to show the issue. Modify the code to do something like start, wait, stop, wait, restart. So that the devs can reproduce the problem.

linknum23, 2015-04-10:
I will try submitting a bug.  Thanks:)

linknum23, 2015-04-13:
Just for reference, here is the [issue](http://github.com/OpenKinect/libfreenect2/issues/206)

wiedemeyer, 2015-05-21:
I added the start/stop feature to the devel branch.

closed 2015-05-27 by wiedemeyer