code-iai/iai_kinect2 #76: version OPENCL_1.2 not defined in file libOpenCL.so.1

pammirato, 2015-04-10:
hi

I have been playing with this for a few days, messing around with nvidia drivers and other things, but can not get it to work.

****I CAN run libfreenect2 on its own fine.     ./bin/Protonect works.

when i run rosrun kinect2_bridge kinect2_bridge   I get:

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
/home/ammirato/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: relocation error: /home/ammirato/catkin_ws/devel/lib/libdepth_registration.so: symbol clRetainDevice, version OPENCL_1.2 not defined in file libOpenCL.so.1 with link time reference

Heres some more of the output:

base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ammirato/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

thanks!!

pammirato, 2015-04-10:
When i tried to install some nvidia-opencl-dev stuff i got this

ammirato@bvision4:~$ sudo apt-get install nvidia-opencl-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 nvidia-opencl-dev : Depends: nvidia-libopencl1 (>= 319.37) but it is not installable or
                              nvidia-libopencl1-331 but it is not going to be installed or
                              nvidia-libopencl1-331-updates but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

thanks again

closed 2015-04-12 by pammirato

wiedemeyer, 2015-04-13:
Hi,
did you found a solution that fixed you issue?

giacomodabisias, 2015-07-08:
I get the same problem

pammirato, 2015-07-08:
i just don't use opencl now, i set the parameters _reg_method:=cpu _depth_method:=opengl