code-iai/iai_kinect2 #77: Multiple kinects Serial Number

dvonhausen, 2015-04-11:
When I run: 

rosrun kinect2_bridge kinect2_bridge _sensor:=014101745147

I get:

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:13 with serial 014101745147
[Freenect2Impl] found valid Kinect v2 @4:15 with serial 011843445147
[Freenect2Impl] found 2 devices
Kinect2 devices found: 
  0: 014101745147
  1: 011843445147
Error: Device with serial '14101745147' not found!

Notice that the leading 0 in the serial number is deleted and its looking for "14101745147", not "014101745147". Its as if the string is read in as a number in the code.

wiedemeyer, 2015-04-13:
I know this one. I will push a fix for it.

dvonhausen, 2015-04-27:
Works now.

dvonhausen, 2015-05-12:
I did some further testing and there still appears to be a problem. Both my kinects starts with a serial number of 0 and if you don't have any kinects that have this property then it may be difficult for you to test. I applied the changes of #81 commit by hand. When I plug both kinects into my pc, it always takes the first one regardless of what serial number I give it. Looking at the code, it may be the case that this line:
<param name="sensor"            type="double" value="$(arg sensor)"/>
still removes the leading 0 in the number.

I'm using
rosrun kinect2_bridge kinect2_bridge _sensor:="011843445147"

I'll look into it further soon.

dvonhausen, 2015-05-12:
Didn't mean to close this issue just now. Still investigating.

dvonhausen, 2015-05-12:
OK - it does work - my mistake. In the launch file I needed to use type="str", not type="double". Still works.

closed 2015-05-12 by dvonhausen

dvonhausen, 2015-05-12:
One other thing:
When I use roslaunch, both sensors work ok (except I now get "image data too short!"). However, if I use rosrun... sensor=..., I get "LIBUSB_ERROR_BUSY" error and both terminate.