code-iai/iai_kinect2 #78: rostopic echo /kinect2/depth/image is full of zeros

ipekbocegi, 2015-04-17:
Hi everyone,

Following commands return me full of zeros:
$ rostopic echo /kinect2/depth/image
$ rostopic echo /kinect2/depth_lowres/image
$ rostopic echo /kinect2/ir/image
$ rostopic echo /kinect2/rgb/image

But these return meaningful pixel values:
$ rostopic echo /kinect2/depth/image/compressedDepth
$ rostopic echo /kinect2/depth_lowres/image/compressedDepth
$ rostopic echo /kinect2/ir/image/compressedDepth
$ rostopic echo /kinect2/rgb/image/compressedDepth

I did not try the rest or the topics. I think problem is about /image. 

Does anyone has faced with the same or similar issue? Why it gives me black?

What I have is:
* rosrun kinect2_bridge kinect2_bridge works:
             ...
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
[DepthPacketStreamParser::handleNewData] not all subsequences received 0
camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.6398010253906, 0, 259.6340026855469;
  0, 365.6398010253906, 205.0050964355469;
  0, 0, 1]
distortion coefficients ir:
[0.09497310221195221, -0.2719865143299103, 0, 0, 0.09249096363782883]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
starting main loop
[kinect2_bridge] depth processing: ~9411.05Hz (0.106258ms) publishing rate: ~27.9092Hz
[kinect2_bridge] color processing: ~4947.9Hz (0.202106ms) publishing rate: ~20.5997Hz
[OpenGLDepthPacketProcessor] avg. time: 4.37858ms -> ~228.384Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.8897ms -> ~47.8704Hz

* "$ rosrun registration_viewer viewer kinect2 sd both" shows the output only in image window(blueish image) but not the cloud window(it is black, only some number is counted countiniously. I don't know maybe it is supposed to be black?)

* libfreenect2 is working properly. It shows depth, ir, rgb images.
...
[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 508765442542
device firmware: 4.3.3912.0.7
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: 22.9463ms -> ~43.5801Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.545ms -> ~48.6737Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: 25.7769ms -> ~38.7945Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.6907ms -> ~42.2107Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.2182ms -> ~52.0339Hz
[TurboJpegRgbPacketProcessor] avg. time: 18.8ms -> ~53.1916Hz

Any help is appreciated.. 

Best wishes

wiedemeyer, 2015-04-17:
If the depth images are zeros, the compressed topics will output a compressed representation of the zeros.
From your output it looks like libfreenect2 is not working correctly, there are no messages regarding the depth processor timings.

ipekbocegi, 2015-04-17:
Thank you for quick response

Ok please help me to understand. This is screenshot from ./Protonect

[image]

So sample example does not prove that the libfreenect2 actually works properly? Is it true?

For example "$ rostopic echo /kinect2/depth/image/compressedDepth" gives
header: 
  seq: 57
  stamp: 
    secs: 1429283854
    nsecs: 636259781
  frame_id: kinect2_ir_optical_frame
format: 16UC1; compressedDepth

data: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 73, 42, 0, 120, 30, 0, 0, 128, 0, 32, 80, 56, 36, 22, 13, 7, 132, 66, 97, 80, 184, 100, 54, 29, 15, 136, 68, 98, 81, 56, 164, 86, 45, 23, 140, 70, 99, 81, 184, 228, 118, 61, 31, 144, 72, 100, 82, 57, 36, 150, 77, 39, 148, 74, 101, 82, 185, 100, 182, 93, 47, 152, 76, 102, 83, 57, 164, 214, 109, 55, 156, 78, 103, 83, 185, 228, 246, 125, 63, 160, 80, 104, 84, 58, 37, 22, 141, 71, 164, 82, 105, 84, 186, 101, 54, 157, 79, 168, 84, 106, 85, 58, 165, 86, 173, 87, 172, 86, 107, 85, 186, 229, 118, 189, 95, 176, 88, 108, 86, 59, 37, 150, 205, 103, 180, 90, 109, 86, 187, 101, 182, 221, 111, 184, 92, 110, 87, 59, 165, 214, 237, 119, 188, 94, 111, 87, 187, 229, 246, 253, 127, 160, 64, 64, 128, 0, 32, 80, 56, 36, 22, 13, 7, 132, 66, 97, 80, 184, 100, 54, 29, 15, 136, 68, 98, 81, 56, 164, , ........bunch of numbers.......  , 0, 0, 160, 4, 0, 0, 51, 5, 0, 0, 198, 5, 0, 0, 89, 6, 0, 0, 236, 6, 0, 0, 127, 7, 0, 0, 18, 8, 0, 0, 165, 8, 0, 0, 56, 9, 0, 0, 203, 9, 0, 0, 94, 10, 0, 0, 241, 10, 0, 0, 132, 11, 0, 0, 23, 12, 0, 0, 170, 12, 0, 0, 61, 13, 0, 0, 208, 13, 0, 0, 99, 14, 0, 0, 246, 14, 0, 0, 137, 15, 0, 0, 28, 16, 0, 0, 175, 16, 0, 0, 66, 17, 0, 0, 213, 17, 0, 0, 104, 18, 0, 0, 251, 18, 0, 0, 142, 19, 0, 0, 33, 20, 0, 0, 180, 20, 0, 0, 71, 21, 0, 0, 218, 21, 0, 0, 109, 22, 0, 0, 0, 23, 0, 0, 147, 23, 0, 0, 38, 24, 0, 0, 185, 24, 0, 0, 76, 25, 0, 0, 223, 25, 0, 0, 114, 26, 0, 0, 5, 27, 0, 0, 152, 27, 0, 0, 43, 28, 0, 0, 190, 28, 0, 0, 81, 29, 0, 0, 228, 29, 0, 0]

but "$ rostopic echo /kinect2/ir/image" gives full of zeros..

I can not understand where those numbers and that protonect image come from if libfreenect2 is not working.. Also I don't understand how can compressed values exist where the image is absent.  

Where is the source of the problem exactly? How can I fix it?

wiedemeyer, 2015-04-17:
kinect2_bridge gets an image filled with zeros from libfreenect2. Compression does not care if the data contains just zeros.
But from your screenshot libfreenect2 seems to work, but kinect2_bridge isn't. It could be some linking issue. Is kinect2_bridge linked against the same libraries as Protonect? Is the `libfreenect2.so` file in your libfreenect2 folder the same as the one installed in `/usr/local/lib` (or whereever you installed libfreenect2)?

wiedemeyer, 2015-04-20:
could it be the same as in issue #79?

ipekbocegi, 2015-04-20:
Checking

ipekbocegi, 2015-04-20:
Any diagnosis?

ipekbocegi, 2015-04-20:
Ok I have got some news. I got images. 
I used "$ rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu" instead of "$ rosrun kinect2_bridge kinect2_bridge"
Then I tested image topics. They all work. Besides, "rosrun registration_viewer viewer kinect2 sd both" also gives images.
So, problem is solved
Thanks

closed 2015-04-20 by ipekbocegi