code-iai/iai_kinect2 #79: Error: no Kinect2 devices found!

flobotics, 2015-04-17:
hi,
i have ubuntu 14.04 and indigo installed, when i run  rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu i get:

parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ros/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @9:3!
[Freenect2Impl] found 0 devices
Error: no Kinect2 devices found!

when i run tail -f /var/log/syslog while plug-in-and-out the kinect2 from usb3.0 port it returns:

Apr 17 20:00:01 rospc kernel: [  706.563785] usb 9-2: USB disconnect, device number 2
Apr 17 20:00:01 rospc kernel: [  706.563790] usb 9-2.1: USB disconnect, device number 4
Apr 17 20:00:01 rospc kernel: [  706.695508] usb 8-2: USB disconnect, device number 2
Apr 17 20:00:11 rospc kernel: [  716.106291] usb 8-2: new high-speed USB device number 3 using xhci_hcd
Apr 17 20:00:11 rospc kernel: [  716.125877] usb 8-2: New USB device found, idVendor=045e, idProduct=02d9
Apr 17 20:00:11 rospc kernel: [  716.125880] usb 8-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Apr 17 20:00:11 rospc kernel: [  716.125882] usb 8-2: Product: NuiSensor Adaptor      
Apr 17 20:00:11 rospc kernel: [  716.125884] usb 8-2: Manufacturer: Microsoft Corporation  
Apr 17 20:00:11 rospc kernel: [  716.127034] hub 8-2:1.0: USB hub found
Apr 17 20:00:11 rospc kernel: [  716.127499] hub 8-2:1.0: 1 port detected
Apr 17 20:00:11 rospc kernel: [  716.571781] usb 9-2: new SuperSpeed USB device number 5 using xhci_hcd
Apr 17 20:00:13 rospc kernel: [  717.903652] usb 9-2: Parent hub missing LPM exit latency info.  Power management will be impacted.
Apr 17 20:00:14 rospc kernel: [  719.550352] usb 9-2: New USB device found, idVendor=045e, idProduct=02d9
Apr 17 20:00:14 rospc kernel: [  719.550356] usb 9-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Apr 17 20:00:14 rospc kernel: [  719.550358] usb 9-2: Product: NuiSensor Adaptor      
Apr 17 20:00:14 rospc kernel: [  719.550359] usb 9-2: Manufacturer: Microsoft Corporation  
Apr 17 20:00:14 rospc kernel: [  719.551618] hub 9-2:1.0: USB hub found
Apr 17 20:00:14 rospc kernel: [  719.551966] hub 9-2:1.0: 1 port detected
Apr 17 20:00:15 rospc kernel: [  720.018711] usb 9-2.1: new SuperSpeed USB device number 6 using xhci_hcd
Apr 17 20:00:15 rospc kernel: [  720.034084] usb 9-2.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
Apr 17 20:00:15 rospc kernel: [  720.037082] usb 9-2.1: New USB device found, idVendor=045e, idProduct=02c4
Apr 17 20:00:15 rospc kernel: [  720.037086] usb 9-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
Apr 17 20:00:15 rospc kernel: [  720.037088] usb 9-2.1: Product: Xbox NUI Sensor
Apr 17 20:00:15 rospc kernel: [  720.037089] usb 9-2.1: Manufacturer: Microsoft
Apr 17 20:00:15 rospc kernel: [  720.037090] usb 9-2.1: SerialNumber: 111452734247

could it be, that it searches for the wrong device number or anything like that?

thanks
flo

wiedemeyer, 2015-04-17:
Hi,
did you set the permission for the device? https://github.com/code-iai/iai_kinect2#permissions-to-access-the-kinect-one

flobotics, 2015-04-18:
hi,
thanks for your hint, now i can run "rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu"  but after some information printed on screen, it prints always:

[kinect2_bridge] depth processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz

When i open "rosrun rviz rviz" , select an image-display and connect it to /kinect2/rgb/image  i see only a black image, because of the 0Hz publishing rate i think, but i dont know what todo now ?

thanks for help
flo

wiedemeyer, 2015-04-18:
Does `Protonect` from libfreenect2 work? Which USB controller are you using?

flobotics, 2015-04-18:
Protonect run without any parameters returns: segfault.  With kinect2_bridge i need to add _depth_method:=cpu to not have a segfault.

what do you mean with "which usb controller" ? a usb3.0 sitecom pci-express card

flobotics, 2015-04-18:
02:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)

wiedemeyer, 2015-04-18:
The once from Renesas usually work well, but I don't know if all models work. Protonect uses OpenGL 3.3, maybe that is not supported by your GPU.
I guess the problem is related to USB, because you are not receiving anything. You should try to get Protonect working. Replace the default package pipeline by the CPU package pipeline in the source code of Protonect and try if it then works. If not, you could also ask on the issues page for libfreenect2 for help, or search there for similar issues, which might contain a working solution.

flobotics, 2015-04-18:
with glxinfo i get version 3.0, so i installed the drivers from http://support.amd.com/en-us/download/desktop?os=Ubuntu%20x86%2064 now i got version 4.4 and Protonect runs fine with ir, depth and rgb-image displays. It prints:

[TurboJpegRgbPacketProcessor] avg. time: 20.05ms -> ~49.8753Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~66.5498Hz (15.0263ms) publishing rate: ~29.9137Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.9062ms -> ~50.2355Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~83.3762Hz (11.9938ms) publishing rate: ~29.9882Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.8104ms -> ~50.4785Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~4322.8Hz (0.231332ms) publishing rate: ~30.2536Hz

When i run "rosrun kinect2_bridge kinect2_bridge" it runs, and in rviz i can see rgb and mono images with an image-display, but on console it prints:

[TurboJpegRgbPacketProcessor] avg. time: 20.5127ms -> ~48.7502Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~62.6877Hz (15.9521ms) publishing rate: ~29.9709Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.05ms -> ~49.8753Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~66.5498Hz (15.0263ms) publishing rate: ~29.9137Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.9062ms -> ~50.2355Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~83.3762Hz (11.9938ms) publishing rate: ~29.9882Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.8104ms -> ~50.4785Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~4322.8Hz (0.231332ms) publishing rate: ~30.2536Hz

it seemed to publish no depth images ?

wiedemeyer, 2015-04-18:
seems to be similar to #78. Try the things i mentioned in the last post there.

flobotics, 2015-04-18:
yes, they are the same. I rebuild libfreenect2 and iai_kinect2 and on my whole system i only got

/usr/local/lib/libfreenect2.so  and
/home/ros/libfreenect2/examples/protonect/lib/libfreenect2.so

both have the date of some minutes (when i rebuild), but i still get

[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~6176.37Hz (0.161907ms) publishing rate: ~29.9161Hz

wiedemeyer, 2015-04-18:
did you ran it with `_depth_method:=cpu` or `_depth_method:=opengl`, try both.

flobotics, 2015-04-18:
if i use cpu or opengl it returns with both:

[DepthRegistration::New] Using CPU registration method!

[TurboJpegRgbPacketProcessor] avg. time: 19.8291ms -> ~50.4309Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~5168.23Hz (0.19349ms) publishing rate: ~29.9151Hz

wiedemeyer, 2015-04-18:
depth_method is to select the depth processor of libfreenect2, reg_method is for depth registration in kinect2_bridge.
So kinect2_bridge is not receiving any frames from libfreenect2. Right now I have no clue, what the problem could be. What is the whole output of kinect2_bridge?

flobotics, 2015-04-18:
rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/ros/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] found valid Kinect v2 @9:3 with serial 111452734247
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 111452734247 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 0b 08 14 47 4b 43 36 33 33 2e 31    31332e2e2e2e2e2e474b433633332e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[... 103 lines cut ...]
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[Freenect2DeviceImpl] stopped
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] releasing usb interfaces...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] already closed, doing nothing

wiedemeyer, 2015-04-18:
`[TransferPool::submit] failed to submit transfer` this means there is a problem with the USB transfer. Is kinect2_bridge linking to the same libusb as libfreenect2? Did you add the modification https://github.com/code-iai/iai_kinect2#modifications-to-upstream-libfreenect2 to libfreenect2?

flobotics, 2015-04-18:
no, i thought they were included. But i dont know in which CMakeListst.txt file, there are many in libfreenect2 folder

wiedemeyer, 2015-04-18:
The files is linked in the description of the modifications.

flobotics, 2015-04-18:
sorry, the line "Install libfreenect2 with the described modifications:"  takes me to think, that it contains the modifications

i put it in and rebuild it, now it works

[TurboJpegRgbPacketProcessor] avg. time: 20.4297ms -> ~48.9483Hz
[OpenGLDepthPacketProcessor] avg. time: 25.4539ms -> ~39.2867Hz
[kinect2_bridge] depth processing: ~7680.28Hz (0.130204ms) publishing rate: ~29.2926Hz
[kinect2_bridge] color processing: ~4703.42Hz (0.212611ms) publishing rate: ~29.9583Hz
[TurboJpegRgbPacketProcessor] avg. time: 20.4484ms -> ~48.9036Hz
[OpenGLDepthPacketProcessor] avg. time: 25.5947ms -> ~39.0706Hz
[kinect2_bridge] depth processing: ~7043.06Hz (0.141984ms) publishing rate: ~29.3118Hz
[kinect2_bridge] color processing: ~3917.06Hz (0.255293ms) publishing rate: ~29.978Hz

thanks alot for your help

closed 2015-04-20 by wiedemeyer

ipekbocegi, 2015-04-20:
@flobotics happy to hear you get depth values now. Will you please share "rostopic echo /kinect2/dept/image" result when you try? Is it zeros like mine or works properly? Thanks

flobotics, 2015-11-11:
hi,i firstly checked my emails here, sorryi hope you did get it managed, at the moment i dont do anything with kinect2.

ipekbocegi, 2015-11-11:
No problem I am done with kinect.
Sincerely,