code-iai/iai_kinect2 #80: when calibrate ir, calibration pattern not detected

flobotics, 2015-04-18:
hi,
i want to calibrate the kinect2 and i can run:
rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color

which shows me the calibration pattern over the chess pattern and i can record a image with space-bar. Also i can calibrate color.

But when i run:

rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record ir

it does not show the calibration pattern over the chess pattern ? I tried to tune it a little bit with the l,h,1,2,3,4 keyboard shortcuts, but no calibration pattern is showing up ?

what do i wrong ?
flo

pammirato, 2015-04-19:
did you input the correct pattern?  i.e is your board 5x7?  
remember this not the number of squares but the number of intersections.

pammirato, 2015-04-19:
also my calibration efforts are not going well.

I also have trouble detecting the pattern, and even when i do the depth is still misaligned. 
I get errors of about .2 when calibrating the color,ir,and sync. But when I run the registration viewer the results are not good.

pammirato, 2015-04-19:
hahah nevermind when i was running registration viewer I had on the 'raw' option. woops

flobotics, 2015-04-20:
i use the orginal 5x7x0.03  and i measured it and it is exactly 3x3cm squares.

The color (which is mono) detects it directly, but not the ir. I only see a white paper on ir image ?

wiedemeyer, 2015-04-20:
The ir emitter is pretty strong. Move the pattern further away from the camera. 50 to 150 cm. Once the pattern is recognized the calibration tool should adjust the brightness automatically.

flobotics, 2015-04-20:
it does not work with ir. 

In the image that is shown when running "rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record ir" i by myself can never see the black squares, even how far or near i am at the kinect ?  Is that normal ?

wiedemeyer, 2015-04-20:
No, never seen that. could you add a screenshot?

flobotics, 2015-04-20:
[image]

wiedemeyer, 2015-04-20:
I just talked to a colleague. He said some inks might be transparent to IR. Are you using an ink printer? Could you try it on a laser printer? If that's true, what type of printer and ink are you using? (just out of curiosity)

flobotics, 2015-04-20:
yes, it was an ink-printer. it was not mine, so i dont know the printer/ink info.  I will try it with an laser printer, but that will take until tomorrow.

flobotics, 2015-04-21:
yes, seemed  so.
i printed it on a laser printer and now it directly gets it
[image]

thanks for the tip

closed 2015-04-22 by wiedemeyer