code-iai/iai_kinect2 #83: about point cloud topics

ipekbocegi, 2015-04-20:
Hi,
I can not find point cloud topics when I rosrun kinect2_bridge. I am searching for these:
/kinect2/depth_lowres/points
/kinect2/depth_highres/points
Did I miss something?
referance: https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge#point-cloud-topics

I need cloud data so that I can process image and extract objects and peoples and so on. 

What we have right now is depth data right? If so, how can I obtain pointcloud data from them. I mean what do those numbers stand for when I get using /kinect2/depth/image? 

They look like 
....... , 2, 33, 2, 33, 2, 32, 2, 32, 2, 31, 2, 32, 2, 31, 2, 30, 2, 30, 2, 30, 2, 30, 2, 31, 2, 0, 0, 29, 2, 0, 0, 0, 0, 28, 2, 28, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 35, 2, 34, 2, 36, 2, 36, 2, 38, 2, 38, 2, 0, 0, 39, 2, 40, 2, 39, 2, 39, 2, 39, 2, 40, 2, 40, 2, 42, 2, 41, 2, 40, 2, 0, 0, 40, 2, 40, 2, 39, 2, 40, 2, 40, 2, 41, 2, 41, 2, 40, 2, 40, 2, 40, 2, 41, 2, 42, 2, 42, 2, 42, 2, 44, 2, 44, 2, 44, 2, 42, 2, 43, 2, 42, 2, 43, 2, 42, 2, 47, 2, 58, 2, 83, 2, 99, 2, 105, 2, 104, 2, 107, 2, 109, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 104, 2, 94, 2, 91, 2, 88, 2, 0, 0, 0, 0, 0, 0, 179, 7, 190, 7, 193, 7, 195, 7, 193, 7, 191, 7, 187, 7, 188, 7, 191, 7, 194, 7, 186, 7, 176, 7, 172, 7, 169, 7, 170, 7, 168, 7, 167, 7, 163, 7, 160, 7, 157, 7, 152, 7, 149, 7, 145, 7, 143, 7, 138, 7, 135, 7, 133, 7, 129, ......

Any help would be great. Thank you

wiedemeyer, 2015-04-20:
Hi,
that's simple, just launch it via launch file. [The README of kinect2_bridge states](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge#point-cloud-topics): `Only available if kinect2_bridge.launch is launched.`

ipekbocegi, 2015-04-21:
Oops, sorry missed it
$ roslaunch kinect2_bridge kinect2_bridge.launch worked for sure
Now with $ rostopic echo /kinect2/depth_lowres/points, I have something like:
..... 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 37, 31, 32, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 41, 35, 36, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 114, 109, 110, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 98, 93, 94, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 255, 253, 254, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 255, 253, 253, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 254, 254, 254, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 0, 0, 254, 254, 254 ....

They are continuous RGB pixel elements of 960 x 540 continuous flowing images right? So I have 518400 elements for each frame? But what is the coordinate relation? I mean in .launch file it says "xyzrgb" so how can I interpret the distance? I have been searching on web but could not find any illuminating source regarding this format or conversation. Thank you

wiedemeyer, 2015-04-21:
You should look at PCL documentation for information. http://pointclouds.org/documentation/tutorials/
Point clouds consists of points in space, xyz are coordinates in space and rgb is the color of that point.

ipekbocegi, 2015-04-21:
Thank you so much
Best wishes

closed 2015-04-21 by ipekbocegi

summer1216, 2015-06-16:
Hello, I have the similar problem.  Since I don't have a kinect at moment, I want to do the tests with rosbag (view the point cloud of the rosbag). When I'm playing the rosbag and run the kinect2_viewer, it gets stuck with the statement  "" starting receiver...""

 the topic list is showed as follows:

/clicked_point
/clock
/color/sd/camera_info
/color/sd/image_color_rect
/color/sd/image_depth_rect
/initialpose
/kinect2/bond
/kinect2/depth_lowres/camera_info
/kinect2/depth_lowres/image/compressed
/kinect2/ir/camera_info
/kinect2/ir/image/compressed
/kinect2/ir_rect/camera_info
/kinect2/ir_rect/image/compressed
/kinect2/rgb_lowres/camera_info
/kinect2/rgb_lowres/image/compressed
/kinect2/sd/points
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

I think it's also lack of 

/kinect2/depth_lowres/points
/kinect2/depth_highres/points

right?

Thank you very much if you could give me some help!

ipekbocegi, 2015-06-17:
@summer1216 /points will be available if you launch kinect2_bridge in a new terminal. Just do "$ roscore" in a terminal and "$ roslaunch kinect2_bridge kinect2_bridge.launch" in another terminal. Then in the third terminal list the rostopics and you will see .../points.

summer1216, 2015-06-17:
Thanks a lot, but my corporation above is already based on kinect2_bridge.lauch. The topics from that are /kinect2/bond and /kinect2/sd/points.. And kinect2_view can not work...

ipekbocegi, 2015-06-17:
ok try this in a new terminal while it roslauch still says starting receiver. "rostopic echo /kinect2/depth_lowres/points"

ipekbocegi, 2015-06-17:
and try this "rosrun registration_viewer viewer kinect2 sd both" in a new terminal. Keep "roscore"rosrun kinect2_bridge kinect2_bridge" in another terminals

wiedemeyer, 2015-06-17:
@summer1216 is that the full list of topics? Normally there are more topics. A complete list is given [here](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge)
It seems that you are using an old version of iai_kinect2. Please pull from the current master branch and also update libfreenect2 to the latest version.

summer1216, 2015-06-18:
Thanks a lot @ipekbocegi  and @wiedemeyer . Now it's solved. Yes, the problem is indeed because of the version. Although I used the new version of iat_kinect2, but the topic names of the rosbag I used were according to the old version.. which the new version could not recognize. I use new recorded bags and now it's fine. Thanks!