code-iai/iai_kinect2 #84: Cloud Viewer show nothing ,use Intel CPU/GPU  and OpenCL

YuweiGuo, 2015-04-21:
hallo , 
when I run cloud viewer with intel GPU, I have a problem , the cloud viewer show nothing . this is my computer :
system : Ubuntu 14.04
cpu : Intel® Core™ i7-3630QM CPU @ 2.40GHz × 8 
graphics : Nvidia Quadro K3000M and Intel(R) HD Graphics IvyBridge M GT2 . 

when I use the Nvidia card and OpenCL , the programm run gut . 
then I install the Beignet for Intel GPU . and then I start the Kinect_bridge the second time.

the following is my steps:

1----Nvidia Karte driver 331 , opencl-icd-331 , cuda-dev ...

2----reboot

3----OpenCV

4----PCL

5----ROS

7----iai-kinectv2   run gut  , image and cloud

8----switch the Graphic cart to  Intel  , can not find opencl device 

9----Intel opencl install , Beignet 1.0.0

10---- run gut.

11----reboot

11---- and then I switch back to Nvidia card

the viewer show it self gut , but the programm show that , it use the Intel card as a OpenCL device. 
I get the following output (note : I still use Nvidia card now):
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
modprobe: ERROR: could not insert 'nvidia_331_updates': No such device
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/yuwei/Groupprojekt/ros_kinectV2_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl     // different
     depth_device: -1
       reg_method: opencl    
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:5 with serial 039374243547
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 039374243547 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

[... 388 lines cut ...]

/////////////////////////////////////////////////////////

the following is my steps:

0----ubuntu 14.04 new install

1----install opencv

2----install pcl

3----install ros

4----install libfreenect2 , rgb run gut , but ir and cloud show nothing

5----install iai_kinect2 , bridge show  that  using CPU registration method , image run gut but cloud show nothing ,

6----install intel OpenCL , beignet 1.0.0 , image run gut but cloud show nothing 

7----compile iai_kinect2 onece mal , image run gut but cloud show nothing 

what have I make false? 
thank you!

yuwei

wiedemeyer, 2015-04-21:
Is `Protonect` working?
So if you use OpenCL as `depth_method` it works, but with `OpenGL` it doesn't. So just use OpenCL, `depth_method:=opencl` as a parameter.

YuweiGuo, 2015-04-21:
The Protonect is working , but just for the nvidia card . Intel not work.
Yes , I think so too, how can I fix it?

YuweiGuo, 2015-04-21:
Rewrite the kinect2_bridge.cpp?

wiedemeyer, 2015-04-21:
libfreenect2 has to be compiled with OpenCL support. And kinect2_bridge has to be compiled after it. Then it should use OpenCL as default.
You can change it on start via parameters, for example:
`roslaunch kinect2_bridge kinect2_bridge depth_method:=opencl`

closed 2015-06-01 by wiedemeyer