code-iai/iai_kinect2 #89: kinect2_bridge cv Exception

rastaxe, 2015-04-30:
My computer is Intel NUC Ubuntu 14.04 (kernel 3.16), ROS indigo. When I launch the calibration or the viewer program, kinect2_bridge crashes:
```
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/nuc1/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell Ultrabook GT2 Mobile (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics Haswell Ultrabook GT2 Mobile (GPU)[Intel]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 8 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:5 with serial 013375243747
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 013375243747 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 05 0b 14 47 4d 46 36 30 33 2e 31    31332e2e2e2e2e2e474d463630332e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 26 00 00                                        7f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 013375243747
device firmware: 4.3.3913.0.7

default ir camera parameters: 
fx 365.817, fy 365.817, cx 252.354, cy 200.07
k1 0.0957173, k2 -0.274653, p1 0, p2 0, k3 0.0932564

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.
[DepthPacketStreamParser::handleNewData] not all subsequences received 0

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.8168029785156, 0, 252.3535003662109;
  0, 365.8168029785156, 200.0697937011719;
  0, 0, 1]
distortion coefficients ir:
[0.09571728855371475, -0.2746531963348389, 0, 0, 0.09325640648603439]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell Ultrabook GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics Haswell Ultrabook GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell Ultrabook GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics Haswell Ultrabook GT2 Mobile (GPU)[Intel]
starting main loop

[kinect2_bridge] depth processing: ~7338.81Hz (0.136262ms) publishing rate: ~27.998Hz
[kinect2_bridge] color processing: ~3881.27Hz (0.257648ms) publishing rate: ~23.9983Hz
[OpenCLDepthPacketProcessor] avg. time: 16.4621ms -> ~60.7456Hz
[TurboJpegRgbPacketProcessor] avg. time: 29.7237ms -> ~33.6432Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~7462.73Hz (0.133999ms) publishing rate: ~29.9898Hz
[kinect2_bridge] color processing: ~3985.56Hz (0.250906ms) publishing rate: ~29.3234Hz
[OpenCLDepthPacketProcessor] avg. time: 16.5312ms -> ~60.4919Hz
[TurboJpegRgbPacketProcessor] avg. time: 29.8694ms -> ~33.4791Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~7463.47Hz (0.133986ms) publishing rate: ~29.9412Hz
[kinect2_bridge] color processing: ~4157.49Hz (0.240529ms) publishing rate: ~27.2798Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[OpenCLDepthPacketProcessor] avg. time: 16.7707ms -> ~59.6278Hz
[TurboJpegRgbPacketProcessor] avg. time: 30.379ms -> ~32.9175Hz
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] image data too short!
[DepthPacketStreamParser::handleNewData] not all subsequences received 1021
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
OpenCV Error: Assertion failed (s >= 0) in setSize, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp, line 116
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp:116: error: (-215) s >= 0 in function setSize
```

Protonect works (I can see RGB, depth and IR images). I compiled libfreenect2 with: cmake ../../examples/protonect/ -DENABLE_CXX11=ON -DENABLE_OPENCL:BOOL=ON -DENABLE_OPENGL:BOOL=OFF

wiedemeyer, 2015-05-12:
Hi,
could you compile it in debug and check where this exception is thrown.

wiedemeyer, 2015-06-01:
Any update or can I close this issue?

closed 2015-06-01 by wiedemeyer

rastaxe, 2015-06-03:
Sorry, I am very busy and I did not have time to try...