code-iai/iai_kinect2 #90: ERROR: clEnqueueWriteBuffer(-5)

UnforgiveNX, 2015-05-04:
Hello,

When I run `roslaunch kinect2_bridge kinect2_bridge.launch` with these arguments:
```
  <arg name="base_name"         default="kinect2"/>
  <arg name="sensor"            default="-1" />
  <arg name="publish_tf"        default="false" />
  <arg name="base_name_tf"      default="$(arg base_name)" />
  <arg name="fps_limit"         default="8.0"/>
  <arg name="calib_path"        default="$(find kinect2_bridge)/data/"/>
  <arg name="use_png"           default="false"/>
  <arg name="jpeg_quality"      default="75"/>
  <arg name="png_level"         default="1"/>
  <arg name="depth_method"      default="default"/>
  <arg name="depth_device"      default="-1"/>
  <arg name="reg_method"        default="default"/>
  <arg name="reg_devive"        default="-1"/>
  <arg name="max_depth"         default="12.0"/>
  <arg name="min_depth"         default="0.1"/>
  <arg name="queue_size"        default="2"/>
  <arg name="bilateral_filter"  default="true"/>
  <arg name="edge_aware_filter" default="true"/>
  <arg name="worker_threads"    default="4"/>
  <arg name="machine"           default="localhost" />
  <arg name="nodelet_manager"   default="$(arg base_name)" />
  <arg name="start_manager"     default="true" />
  <arg name="use_machine"       default="true" />
```

it starts publishing topics normally. But after when I run, `rosrun registration_viewer viewer sd image`

The output is this:

```
[kinect2_bridge] depth processing: ~47.5939Hz (21.0111ms) publishing rate: ~8.31667Hz
[kinect2_bridge] color processing: ~71.2506Hz (14.035ms) publishing rate: ~8.31667Hz
[OpenCLDepthPacketProcessor] avg. time: 18.0317ms -> ~55.4578Hz
[DepthRegistrationOpenCL::registerDepth] ERROR: clWaitForEvents(-9999)
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[TurboJpegRgbPacketProcessor] avg. time: 23.3276ms -> ~42.8677Hz
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[DepthRegistrationOpenCL::registerDepth] ERROR: clEnqueueWriteBuffer(-5)
[kinect2_bridge] depth processing: ~43.0744Hz (23.2157ms) publishing rate: ~7.98564Hz
[kinect2_bridge] color processing: ~67.5529Hz (14.8032ms) publishing rate: ~7.98564Hz
```

and the viewer shows:
[image]

Sometimes this error is shown immediately after registration_viewer, but sometimes after a few seconds of normal operation.

In my findings it is told that -5 error is CL_OUT_OF_RESOURCES error. Is it related with GeForce GT 540M I have in my laptop?

Thanks

wiedemeyer, 2015-05-12:
oh, never seen something like that. Maybe it is too much for your GPU. I will look into it.

wiedemeyer, 2015-05-12:
ok, could you try to run it and watch the GPU info with `watch -n 1 nvidia-smi`.

wiedemeyer, 2015-05-12:
And you can try to run it and then listen to the registered depth topic with `rostopic hz /kinect2/depth_highres/image`

wiedemeyer, 2015-06-03:
Any update or can we close this issue?

closed 2015-06-08 by wiedemeyer