code-iai/iai_kinect2 #91: libusb segmentation fault: "libusb error -7: LIBUSB_ERROR_TIMEOUT"

christianmueller, 2015-05-08:
Hello,

Thanks alot for the detailed HowTo and for driver! I followed your Readme on https://github.com/code-iai/iai_kinect2 . 

I am running it on Ubuntu 12.04 with ROS-hydro.

When I run:

rosrun kinect2_bridge kinect2_bridge 

I get the following output:
---------------------------->
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/.../src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 16 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:14 with serial 008754441947
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 008754441947 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 00 15 47 4b 43 37 30 31 2e 31    31332e2e2e2e2e2e474b433730312e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[CommandTransaction::receive] bulk transfer failed! libusb error -7: LIBUSB_ERROR_TIMEOUT
[CommandTransaction::execute] missing response complete!
Segmentation fault (core dumped)
<----------------------------

Or when I try in libfreenect2

 ./Protonect 

with gdb I get the following out:
---------------------------->
Starting program: /home/.../libfreenect2/examples/protonect/bin/Protonect 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe9e53700 (LWP 2392)]
[New Thread 0x7fffe9652700 (LWP 2393)]
[New Thread 0x7fffe6494700 (LWP 2394)]
[New Thread 0x7fffe5c93700 (LWP 2395)]
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 16 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:14 with serial 008754441947
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 00 15 47 4b 43 37 30 31 2e 31    31332e2e2e2e2e2e474b433730312e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[CommandTransaction::receive] bulk transfer failed! libusb error -7: LIBUSB_ERROR_TIMEOUT
[CommandTransaction::execute] missing response complete!

Program received signal SIGSEGV, Segmentation fault.
libfreenect2::Freenect2DeviceImpl::start (this=0x61fcd0) at /home/.../libfreenect2/examples/protonect/src/libfreenect2.cpp:461
461	  ir_camera_params_.fx = ir_p->fx;
(gdb) bt
#0  libfreenect2::Freenect2DeviceImpl::start (this=0x61fcd0) at /home/.../libfreenect2/examples/protonect/src/libfreenect2.cpp:461
#1  0x0000000000401c17 in main (argc=<optimized out>, argv=<optimized out>) at /home/.../libfreenect2/examples/protonect/Protonect.cpp:74
(gdb) f 1
#1  0x0000000000401c17 in main (argc=<optimized out>, argv=<optimized out>) at /home/.../libfreenect2/examples/protonect/Protonect.cpp:74
74	  dev->start();
(gdb) l 74
69	  libfreenect2::SyncMultiFrameListener listener(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
70	  libfreenect2::FrameMap frames;
71	
72	  dev->setColorFrameListener(&listener);
73	  dev->setIrAndDepthFrameListener(&listener);
74	  dev->start();
75	
76	  std::cout << "device serial: " << dev->getSerialNumber() << std::endl;
77	  std::cout << "device firmware: " << dev->getFirmwareVersion() << std::endl;
78	
(gdb) f 0
#0  libfreenect2::Freenect2DeviceImpl::start (this=0x61fcd0) at /home/.../libfreenect2/examples/protonect/src/libfreenect2.cpp:461
(gdb) l 461
456	  }
457	
458	  command_tx_.execute(ReadDepthCameraParametersCommand(nextCommandSeq()), result);
459	  DepthCameraParamsResponse *ir_p = reinterpret_cast<DepthCameraParamsResponse *>(result.data);
460	
461	  ir_camera_params_.fx = ir_p->fx;
462	  ir_camera_params_.fy = ir_p->fy;
463	  ir_camera_params_.cx = ir_p->cx;
464	  ir_camera_params_.cy = ir_p->cy;
465	  ir_camera_params_.k1 = ir_p->k1;

<----------------------------

Apparently it dies when the device is supposed to start? Has anyone else experienced this behavior? Thanks in advance for any comments on this issue.

UnforgiveNX, 2015-05-11:
Hello,

Did you change the necessary parts in CMakeList?

https://github.com/code-iai/iai_kinect2#modifications-to-upstream-libfreenect2

christianmueller, 2015-05-12:
Thanks  for the comment!

I already changed the CMakeList. After some investigation, I found out that the Kinect is not probably connected over the  USB 3.0. Since I can see the device over lsusb:

Bus 002 Device 004: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 002 Device 005: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
Bus 002 Device 006: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 003 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 004 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 004 Device 003: ID 045e:02c4 Microsoft Corp. 

However  "dmesg | grep 045e" or "dmesg | grep kinect" or  "dmesg | grep Micro" does not return anything. I think this also causes that libusb receives  LIBUSB_ERROR_TIMEOUT error.

dmesg returns after plugged in Kinect:

[ 1475.029715] usb 1-1.3.4: new high-speed USB device number 6 using ehci_hcd
[ 1475.170230] Initializing USB Mass Storage driver...
[ 1475.170426] scsi4 : usb-storage 1-1.3.4:1.2
[ 1475.170505] usbcore: registered new interface driver usb-storage
[ 1475.170507] USB Mass Storage support registered.
[ 1475.326471] usb 1-1.3.4: USB disconnect, device number 6
[ 1476.078700] usb 1-1.3.4: new high-speed USB device number 7 using ehci_hcd
[ 3214.214996] usb 1-1.3.4: USB disconnect, device number 7
[ 6391.606364] usb 3-3: new high-speed USB device number 2 using xhci_hcd
[ 6391.625252] hub 3-3:1.0: USB hub found
[ 6391.625546] hub 3-3:1.0: 1 port detected
[ 6392.066259] usb 4-3: new SuperSpeed USB device number 2 using xhci_hcd
[ 6395.036722] hub 4-3:1.0: USB hub found
[ 6395.036909] hub 4-3:1.0: 1 port detected
[ 6395.500241] usb 4-3.1: new SuperSpeed USB device number 3 using xhci_hcd
[ 6395.563195] usbcore: registered new interface driver snd-usb-audio

But it does NOT show that a new USB device was found like for instance:
[ 7500.603642] usb 4-2.1: New USB device found, idVendor=045e, idProduct=02c4
[ 7500.603652] usb 4-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4

Did someone else have similar problems before?

Thanks!

christianmueller, 2015-05-12:
Update:

dmesg returns an error after plugged in Kinect:
[ 8179.972783] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802dd3cf300
[ 8179.972787] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802dd3cf340
[ 8179.972789] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802dd3cf3c0
[ 8179.972791] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802dd3cf380
[ 8180.037633] usb 4-4.1: usbfs: usb_submit_urb returned -121
[ 8714.492018] usb 4-4: USB disconnect, device number 6
[ 8714.492021] usb 4-4.1: USB disconnect, device number 7
[ 8714.619681] usb 3-4: USB disconnect, device number 4
[ 8729.409348] usb 3-4: new high-speed USB device number 5 using xhci_hcd
[ 8729.427443] hub 3-4:1.0: USB hub found
[ 8729.427731] hub 3-4:1.0: 1 port detected
[ 8729.869272] usb 4-4: new SuperSpeed USB device number 8 using xhci_hcd
[ 8732.839958] hub 4-4:1.0: USB hub found
[ 8732.840162] hub 4-4:1.0: 1 port detected
[ 8733.303117] usb 4-4.1: new SuperSpeed USB device number 9 using xhci_hcd

The kernel version which I am using is 3.2.0-83-generic

wiedemeyer, 2015-05-12:
Try kernel 3.13 or 3.16.

christianmueller, 2015-05-12:
It works with 3.13 :-) ! - unfortunately not with 3.2. 

Thanks alot for the help!

wiedemeyer, 2015-05-12:
USB 3.0 is not working well in older kernels.

closed 2015-05-12 by wiedemeyer

schelian, 2016-01-27:
@akifh, can you please tell me which edits I need to make to the CMake or Makefile?  I don't see any instructions for that on the github repo of libfreenect2 but I seem to be running into this issue where the Protonect executeable is linking to /lib/x86_64-linux-gnu/libusb-1.0.so.0 as opposed to what's in libfreenect2/depends/libusb/lib.

wiedemeyer, 2016-01-27:
Those changes are not necessary any more. You have to follow the libfreenect2 install instructions:
https://github.com/OpenKinect/libfreenect2/blob/master/README.md#debianubuntu-1404
Step 3 describes how to install libusb.

schelian, 2016-01-27:
Hi Thiemo, I got IAI Kinect2 working on a computer then through a series of upgrades got it un-working (:-P).  I'm following the instructions and when I call CMake it's linking to /usr/lib/x86_64-linux-gnu/libusb-1.0.so, not what's in libfreenect2/depends/libusb/lib.  Is that OK?

I'm running in to the LIBUSB_ERROR_TIMEOUT error I got on the previous thread (https://github.com/code-iai/iai_kinect2/issues/203) using Protonect and the Kinect2_Viewer.

For reference, I'm  running Ubuntu 14.04 LTS (3.19.0-47-generic; upgraded from last time) with ROS Indigo (1.11.16). lsusb shows the Kinect is plugged in.

wiedemeyer, 2016-01-27:
by default libfreenect2 uses now the libusb from the system, but you have to make sure that it is the correct one. Here are the instructions:
Install libusb. `sudo apt-get install libusb-1.0-0-dev`. The version must be >=1.0.20. You may use `sudo apt-add-repository ppa:floe/libusb && sudo apt-get update && sudo apt-get upgrade` if version 1.0.20 is not available.