code-iai/iai_kinect2 #92: problem to install the dependencies for ROS

Wissou, 2015-05-13:
hello everyone ,
 Protonect example runs perfectly for me with the described modifications , but when im trying to install iai_kinect2 in my catkin workspace I got this when I run "  rosdep install -r --from-paths ." :

wissem@wissem-SVE14A15FFB:~/catkin_ws/src/iai_kinect2$ rosdep install -r --from-paths .
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
registration_viewer: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_bridge: Cannot locate rosdep definition for [depth_registration]
kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

please help me !!

wiedemeyer, 2015-05-13:
That's not a real problem. rosdep can't locate these packages because they are all part of the iai_kinect2 package. So everything worked fine.

Wissou, 2015-05-13:
ok I proceeded like you said , but when I run " rosrun kinect2_bridge kinect2_bridge "  windows does not came out , in the terminal everything works fine :
wissem@wissem-SVE14A15FFB:~/catkin_ws$ rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/wissem/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:8 with serial 507199442542
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 507199442542 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 0a 07 17 47 4d 53 33 33 35 2e 31    31332e2e2e2e2e2e474d533333352e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  0b 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[... 70 lines cut ...]
  0, 0, 1]
distortion coefficients ir:
[0.09528505802154541, -0.2790214121341705, 0, 0, 0.09984645247459412]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
starting main loop

[kinect2_bridge] depth processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~2148.35Hz (0.465473ms) publishing rate: ~12.9889Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~2145.99Hz (0.465986ms) publishing rate: ~14.983Hz
[TurboJpegRgbPacketProcessor] avg. time: 27.6729ms -> ~36.1364Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~2577.94Hz (0.387907ms) publishing rate: ~14.9827H

what's wrong ??

wiedemeyer, 2015-05-13:
`kinect2_bridge` is not displaying any windows, it just published the images through ROS topics.
There is a problem with the USB: `[TransferPool::submit] failed to submit transfer`. There are a lot of issues to this problem, you could look at those to find a solution. In general, make sure you have a kernel >= 3.13, check with USB 3.0 controller you are using. Make sure that `ldd libfreenect2.so` points to the libusb in `libfreenect2/depeds/libusb`.

And GitHub is sending me notification whenever someone adds or comments on an issue, so there is no need in writing mails to inform me, that you have added a new comment.

KahnShi, 2015-05-17:
Actually I think I have the same problem with @Wissou. 
When I run "rosrun kinect2_bridge kinect2_bridge", there is a problem that "[TransferPool::submit] failed to submit transfer" and "[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz".
My kernel is 3.13 and USB 3.0 seems all right.

$ lsusb 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 005: ID 045e:02c4 Microsoft Corp. 
Bus 004 Device 004: ID 045e:02d9 Microsoft Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 045e:02d9 Microsoft Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 04f2:b289 Chicony Electronics Co., Ltd 
Bus 001 Device 004: ID 18d1:4ee1 Google Inc. Nexus 4
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

For libfreenect2.so, part of the results are as follows.
$ ldd libfreenect2.so 
	linux-vdso.so.1 =>  (0x00007ffdab585000)
	libusb-1.0.so.0 => /home/libfreenect2/examples/protonect/../../depends/libusb/lib/libusb-1.0.so.0 (0x00007ff81e761000)
......

I don`t know how to solve my problems, could you give me some help?
Thank you so much!

wiedemeyer, 2015-05-18:
Protonect is working and not printing any `[TransferPool::submit] failed to submit transfer` errors and kinect2_bridge is printing those errors, right?
To which libfreenect2.so does kinect2_bridge link? And does that one also link to `/home/libfreenect2/examples/protonect/../../depends/libusb/lib/libusb-1.0.so.0 `?

KahnShi, 2015-05-18:
Oh, I think you found the problem.
For "Protonect is working and not printing any [TransferPool::submit] failed to submit transfer errors and kinect2_bridge is printing those errors, right?", my answer is yes.

I checked $ldd ~/iai_kinect2/devel/lib/kinect2_bridge, it linked to /usr/local/lib/libfreenect2.so.
Then I $ldd /usr/local/lib/libfreenect2.so, it shows that 
libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0

Is there anything wrong with it or should it link to other location? THX!

wiedemeyer, 2015-05-18:
Did you added the modifications to the cmakelists.txt from libfreenect2? If so, just `make` and `sudo make install` libfreenect2 and try again.

KahnShi, 2015-05-18:
@wiedemeyer you are right. The problem is that I forgor to modify the CMakelists.txt out of my carelessness. Maybe you have the same mistake @Wissou and you can check it.

In the end, I have one more question. In rviz, if I use "Image", now I could get the visulization of all of the topics. But if I want to use "PointCloud2" or "Camera", they failed. Is it because I don`t set the correct Fixed Frame? I searched a lot and made some tries but they all failed.
Looking forwards to your reply!

wiedemeyer, 2015-05-18:
For getting a point cloud, you need to start it using roslaunch. The launch file starts the nodes responsible for creating the point clouds.
As frame you have to select `kinect2_link` if I remember correctly.

KahnShi, 2015-05-18:
Sorry it`s not kinect2_link...
BTW, do you know where we could get the information of correct fixed frame?
I mean it seems that information of fixed frame couldn`t be found in Cmakelists.txt or package.xml. I

wiedemeyer, 2015-05-18:
`kinect2_rgb_optical_frame` should work

KahnShi, 2015-05-18:
Yes, it works!
Thank you so much for your help!

Wissou, 2015-05-23:
hello , 

sorry I was in travel , here what I got when I ldd .I did the modifications you mention above to Cmakelistes.txt from the beginning and my is kernel > 3.13 , but I still have the same problem with  [TransferPool::submit] failed to submit transfer :
wissem@wissem-SVE14A15FFB:~$ ldd /home/wissem/libfreenect2/examples/protonect/lib/libfreenect2.so
	linux-vdso.so.1 =>  (0x00007ffed61b7000)
	libusb-1.0.so.0 => /home/wissem/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f6a8b5d3000)

can you help me ...

wiedemeyer, 2015-05-26:
To which libfreenect2.so does kinect2_bridge link and to which libusb does that libfreenect2.so link?

wiedemeyer, 2015-06-03:
Any update or can we close this issue?

closed 2015-06-08 by wiedemeyer