code-iai/iai_kinect2 #93: Could pointcloud2 be available for rviz?

KahnShi, 2015-05-14:
Hi, first I want to say thank you for your great work!

Now, I have a question that when I use rviz, I could not make Pointcloud2 and Camera show the data of topics they subscribe. 
Among the ways of visualization, it seems that only Image could work and show the data of /Kinect2/rgb/image.
I don`t know how to deal with it. Could you give me a hand?
Thank you in advance!
[image]

wiedemeyer, 2015-05-14:
Could you copy and past the terminal output of kinect2_bridge here? Does the registration_viewer work?

KahnShi, 2015-05-18:
It solved with the help of @wiedemeyer. So many thanks to you!

For me, there is a mistake that I forgot to modify the cmakelists.txt. For more details, please check
https://github.com/code-iai/iai_kinect2/issues/92

What`s more, if you want to use rviz, please set Fixed Frame as kinect2_rgb_optical_frame.

closed 2015-05-18 by KahnShi