code-iai/iai_kinect2 #94: Runtime Segfault

hbadino, 2015-05-14:
Hi developers,

I've successfully installed the driver but I get a segmentation fault error when running kinect2_bridge.

The output is:

$rosrun kinect2_bridge kinect2_bridge
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/hbadino/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

Segmentation fault (core dumped)

gdb shows this:

Program received signal SIGSEGV, Segmentation fault.
glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266) at /home/hbadino/ros/libfreenect2/depends/glfw_src/src/window.c:560
560	/home/hbadino/ros/libfreenect2/depends/glfw_src/src/window.c: No such file or directory.
(gdb) bt
#0  glfwGetWindowAttrib (handle=handle@entry=0x0, attrib=attrib@entry=139266) at /home/hbadino/ros/libfreenect2/depends/glfw_src/src/window.c:560
#1  0x00007ffff56cd894 in flextInit (window=0x0, bindings=0x72bd00) at /home/hbadino/ros/libfreenect2/examples/protonect/src/flextGL.c:20
#2  0x00007ffff56d0ef0 in libfreenect2::OpenGLDepthPacketProcessorImpl::initialize (this=this@entry=0x728e10)
    at /home/hbadino/ros/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:472
#3  0x00007ffff56cef59 in libfreenect2::OpenGLDepthPacketProcessor::OpenGLDepthPacketProcessor (this=0x6b2780, parent_opengl_context_ptr=<optimized out>, 
    debug=<optimized out>) at /home/hbadino/ros/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:786
#4  0x00007ffff56bce66 in libfreenect2::OpenGLPacketPipeline::createDepthPacketProcessor (this=0x6ae870)
    at /home/hbadino/ros/libfreenect2/examples/protonect/src/packet_pipeline.cpp:109
#5  0x00007ffff56bcf15 in libfreenect2::BasePacketPipeline::initialize (this=this@entry=0x6ae870)
    at /home/hbadino/ros/libfreenect2/examples/protonect/src/packet_pipeline.cpp:43
#6  0x00007ffff56bd383 in libfreenect2::OpenGLPacketPipeline::OpenGLPacketPipeline (this=0x6ae870, parent_opengl_context=<optimized out>, debug=<optimized out>)
    at /home/hbadino/ros/libfreenect2/examples/protonect/src/packet_pipeline.cpp:102
#7  0x000000000041efaf in Kinect2Bridge::initialize() ()
#8  0x000000000041f3e9 in Kinect2Bridge::run() ()
#9  0x000000000040af09 in main ()

Upon connection of the Kinect, the kernel buffer shows:

[  328.473633] usb 4-1: new SuperSpeed USB device number 2 using xhci_hcd
[  328.492429] usb 4-1: New USB device found, idVendor=045e, idProduct=02c4
[  328.492434] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  328.492437] usb 4-1: Product: Xbox NUI Sensor
[  328.492439] usb 4-1: Manufacturer: Microsoft
[  328.492441] usb 4-1: SerialNumber: 006928351047
[  328.547135] usbcore: registered new interface driver snd-usb-audio

And when I get the segfault:

[  343.564198] kinect2_bridge[10450]: segfault at 1c0 ip 00007f90a8d784f8 sp 00007fffac3357e0 error 4 in libglfw.so.3.0[7f90a8d71000+12000]

Your help would be greatly appreciated.

Hernan

wiedemeyer, 2015-05-15:
The OpenGL processor from libfreenect2 requires OpenGL 3.3. Could it be that your driver only supports 3.0?
You could try to use the OpenCL processor instead.

hbadino, 2015-05-15:
Hi Thiemo,

thanks for replying.

How do I install/enabled/configure the OpenCL processor?

Hernan

wiedemeyer, 2015-05-18:
You need to install OpenCL. When running cmake for libfreenect2 it should enable the OpenCL depth processor when it finds the OpenCL libraries.
Installation of OpenCL depends on the graphics card you are using. For Intel you need beignet, there are some instructions in the readme. For AMD and Nvidia you just need to install their newest drivers and `opencl-headers`.

wiedemeyer, 2015-06-03:
Any updates or can we close this one?

hbadino, 2015-06-03:
You can close it. Thanks.

closed 2015-06-03 by wiedemeyer

vishu2287, 2015-09-29:
I am able to run ./Protonect and can view color, ir and depth images with out any error.
When I am trying to run kinect2_bridge from the root, it asks for clients. If I provide kinect2/sd/image_color_rect, it produces the messages but if I provide other topics like kinect2/qhd/image_color, the following error genetares

root@tegra-ubuntu:~/catkin_ws/devel# rosrun kinect2_bridge kinect2_bridge
parameter:
base_name: kinect2
sensor: 
fps_limit: -1
calib_path: /home/ubuntu/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
use_png: false
jpeg_quality: 90
png_level: 1
depth_method: opengl
depth_device: -1
reg_method: default
reg_devive: -1
max_depth: 12
min_depth: 0.1
queue_size: 2
bilateral_filter: true
edge_aware_filter: true
publish_tf: false
base_name_tf: kinect2
worker_threads: 4

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:5 with serial 017275643547
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
0: 017275643547 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000: 00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41 ............C..A
0x0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
0x0020: 0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00 .!3U... ........
0x0030: 00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00 .......... .....
0x0040: 31 33 00 00 00 04 0a 0d 47 4d 42 30 31 33 2e 31 13......GMB013.1
0x0050: 58 00 00 00 00 00 00 00 07 00 00 00 X...........

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000: 23 26 00 00 #&..

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000: 23 26 00 00 #&..

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[... 328 lines cut ...]
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~127.059Hz (7.87038ms) publishing rate: ~14.6242Hz
Segmentation fault

I am using Jetson TK1 board .

cpu ang gpu performance were altered with maximum efficiency

Why lots of "skipping rgb packet!" and " [TransferPool::submit] failed to submit transfer " ?

why not providing depth data?
[kinect2_bridge] depth processing: ~nanHz (nanms) publishing rate: ~0Hz

Why segmentation fault after producing one message to the topics?