code-iai/iai_kinect2 #95: fatal error: depth_registration.h: No such file or directory

lakehanne, 2015-05-23:
Hi Thiemo,

I was the one who contacted you via email earlier in the week about running your code in hydro. So I have ROS set up in hydro and while doing "catkin_make -DCMAKE_BUILD_TYPE="Release", I got the following error:
    

   /home/servicelab/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:50:32: fatal error: depth_registration.h: No such file or directory/home/servicelab/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:50:32: fatal error: depth_registration.h: No such file or directory
compilation terminated.

compilation terminated.
make[2]: make[2]: *** [CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] Error 1*** [CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] Error 1

make[1]: *** [CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2

I looked into /iai_kinect2/depth_registration/include folder and I see the depth_registratiuon file is there. 
I am deploying this on an intel Core i7 x86_64 GNU/Linux running ubuntu 12.04, the VGA controller is Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09). 
I was wondering if you have a suggestion for me on how I can tackle this problem?
Thank you!

wiedemeyer, 2015-05-26:
Never saw an issue like that. I guess it's a ROS setup problem, but I am not sure. 
First of all, are there any warnings or erros prior to this? Could you past the full catkin_make output (but first remove devel and build)?
When compiling you only sourced `/opt/ros/hydro/setup.bash` right? Did you ran `sudo rosdep init` and `rosdep update` after installation?

lakehanne, 2015-05-28:
Yes, I did. 
I am working from my own system now as the previous system was a fellow lab member's own. Anyways, I installed Ubuntu 14.04 trusty and I got Indigo setup and ros environment properly setup.

In step 4. of your ReadMe.Md file, I entered 
    rosdep install -r --from-paths .
and I got 
    ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
    registration_viewer: No definition of [libpcl-all] for OS version []
    kinect2_bridge: No definition of [cv_bridge] for OS version []
    kinect2_calibration: No definition of [compressed_depth_image_transport] for OS version []
    depth_registration: No definition of [cv_bridge] for OS version []
    Continuing to install resolvable dependencies...

I know you fixed this issue for hydro in (https://github.com/code-iai/iai_kinect2/commit/670cd895e0d065b354291b8e95d598d1235d6a88). 

Did you ever fix the indigo one?

wiedemeyer, 2015-05-28:
For your previous problem: Either you didn't executed `./install_ubuntu.sh` in `libfreenect2/deps` or the build process of libusb failed.
For your recent problem: Did you executed `sudo rosdep init` and `rosdep update` after installing ROS?

lakehanne, 2015-05-28:
Okay, I went back to the Ubuntu 12.04 , went through the installation all over again, executed `rosdep init`, exceuted `rosdep update` and did `rosdep install -r --from-paths .` again and everything seemed honky dory.
But when I do `catkin_make_isolated -DCMAKE_BUILD_TYPE="Release"`, I get the following, 

`
==> Processing catkin package: 'interactive_marker_tutorials'
==> Building with env: '/home/servicelab/catkin_ws/devel_isolated/interactive_markers/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/servicelab/catkin_ws/build_isolated/interactive_marker_tutorials'
==> make -j4 -l4 in '/home/servicelab/catkin_ws/build_isolated/interactive_marker_tutorials'
[ 14%] [ 28%] Built target basic_controls
Built target cube
[ 42%] [ 57%] Built target point_cloud
Built target menu
[ 71%] Built target pong
[ 85%] [100%] Built target selection
Built target simple_marker
<== Finished processing package [215 of 265]: 'interactive_marker_tutorials'

==> Processing catkin package: 'kinect2_bridge'
==> Building with env: '/home/servicelab/catkin_ws/devel_isolated/interactive_marker_tutorials/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/servicelab/catkin_ws/build_isolated/kinect2_bridge'
==> make -j4 -l4 in '/home/servicelab/catkin_ws/build_isolated/kinect2_bridge'
[100%] [100%] Building CXX object CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
Building CXX object CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
/home/servicelab/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:50:32: fatal error: depth_registration.h: No such file or directory/home/servicelab/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:50:32: fatal error: depth_registration.h: No such file or directory

compilation terminated.
compilation terminated.
make[2]: *** [CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] Error 1
make[1]: *** [CMakeFiles/kinect2_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] Error 1
make[1]: *** [CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'kinect2_bridge': 
  Command '/home/servicelab/catkin_ws/devel_isolated/interactive_marker_tutorials/env.sh make -j4 -l4' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/servicelab/catkin_ws/build_isolated/kinect2_bridge && /home/servicelab/catkin_ws/devel_isolated/interactive_marker_tutorials/env.sh make -j4 -l4

`

wiedemeyer, 2015-05-29:
So in #102 it looks fine. But on your hydro system it is still not working, right?
I would suggest that you do the following:
- only source `/opt/ros/hydro/setup.bash`
- create a new empty catkin workspace
- git clone iai_kinect into the src folder
- run catkin_make

horxianfeng, 2015-05-31:
@lakehanne  u try the comment with rosdep install -r iai_kinect2
It work for me in ubuntu 14.04, indigo

wiedemeyer, 2015-06-03:
Did this solved the issue?

lakehanne, 2015-06-06:
No.

wiedemeyer, 2015-06-08:
so you still have the issue with the header file not found?

closed 2015-06-12 by wiedemeyer