code-iai/iai_kinect2 #96: [OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001) terminate called after throwing an instance of 'cl::Error'   what():  clGetPlatformIDs

kws1130, 2015-05-26:


wiedemeyer, 2015-05-26:
Try `./Protonect cl` that should give the same error. Then you could try `sudo ./Protonect cl`, if that works, try to run kinect2_bridge again. If not you could try to run kinect2_bridge with OpenGL depth_method and CPU reg_method instead.

kws1130, 2015-05-26:
i tried urs suggestion using the ./Protonect cl, as ur prediction, i get the same error,
then i tried sudo ./Protonect cl but i get these errors..
[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
anything goes wrong?? I reali duno......

kws1130, 2015-05-26:
... And one thing more... when i folow ur instruction during make n sudo make install the protonect folder, ithe compilation stopped due to #include < libusb.h> not found n i changed all the headers to #include < libusb-1.0/libusb.h> n the compilation success. However the result is the problem i'm facing now, i wonder if this action cause these problems???

wiedemeyer, 2015-05-27:
Did you installed the dependencies in `libfreenect2/depeds`?
What GPU are you using and what did you installed for opencl?

kws1130, 2015-05-27:
u mean the readme in depend? if yes, i did install the kernel 3.0 and also the new libusbx, for the opencl i folowed some website n download the opencl headers, i think is 14.2 smtg, but about gpu i dun have any idea with it, can u provide some ways or website to me?? Ur help wil be apreciated a lot!!

kws1130, 2015-05-27:
but.... i juz now realise tat the opencl-runtime 14.2 i manualy download is not in tar form and i cant even untar it..==... it is in somekinda of setup.msi form??!!...

wiedemeyer, 2015-05-28:
For the dependencies of libfreenect2 you should read their [README](https://github.com/OpenKinect/libfreenect2/blob/master/README.md). You have to execute `./install_ubuntu.sh` in `libfreenect2/depends`. You don't need to install libusb by hand.

Which Ubuntu version are you using? And which kernel?

For OpenCL you need to install `opencl-headers` using `apt-get` and the drivers for your GPU, depending on which GPU you have, probably either INTEL, AMD, or NVIDIA. There are some instructions in the [README](https://github.com/code-iai/iai_kinect2#opencl-with-amd).

A setup.msi is an executable for Windows, but I guess your are using linux.

*Note: This is not a chat in a game and you don't have to hurry. You can take as much time as you want to write an answer. You don't need to shorten the words. That would make it easier to read. I would really appreciate that.*

kws1130, 2015-05-28:
sorry for those shorten words for disturbing you, really sorry for that. 

Erm.... I'm using ubuntu 14.04, and i had read the readme and do execute the ./install_ubuntu.sh, I also check the kernel version yesterday using "uname -r", the version is 3.16.0-34-generic. Moreover, my pc use Gallium 0.4 on AMD TURKS for GPU, and i've install
AMD-APP-SDK-v3.0-0.113.50-Beta-linux64.tar.bz2 ---> amd gpu
AMD Catalyst™ 14.12 Proprietary Ubuntu 14.04 x86 Video Driver for Graphics Accelerators
AMD Catalyst™ 14.12 Proprietary Ubuntu 14.04 x86_64 Video Driver for Graphics Accelerators these drivers since i need both 32 and 64 bit driver to run my 64 bit linux
guided by this link ---> https://code4geeks.wordpress.com/2013/01/05/ubuntu-12-10-installing-amd-catalyst-and-opencl-sdk-nightmare/
sudo apt-get install opencl-headers to get opencl
I even install the opencl with intel gpu ....
I followed this webpage for beignet : https://gist.github.com/spiralray/cae0bc235509e495fec1

And..i try to run ./bin/Protonect again , it detects the kinect but failed to open it.... so sad...
I also tried the rosrun kinect2_bridge kinect2_bridge and the result is......
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/kws/ROS/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)

wiedemeyer, 2015-05-29:
I guess something is misconfigured somewhere. I also have a system with an AMD in use and there I only installed `opencl-headers` and the 14.12 catalyst drivers and OpenCL works fine.
There is also a new amd drivers in the ubuntu repositories now. I just installed that one and it works too.

I would suggest to uninstall beignet, the amd drivers and the amd sdk, and then install the amd drivers from the official ubuntu repository.

You can also try out `./Protonect gl` and `./Protonect cpu`. If the opengl version works, you just need to tell kinect2_bridge to use opengl for depth processing and cpu for depth registration.

SamTew, 2015-05-30:
The very very sad case is..... the next day i did these all stuff, my ubuntu crashed and even cant enter comment line.
I think it is either the kernel's fault but later i found out it is impossible,
so it must be the other issue:
During install the amd-app-sdk, the system told me that i 'm lack of a dependency, the flgrx-core, and i use superuser mode to install the external file and fuse it with the driver, i think it is the factor causing this problem, and now i have to reboot my ubuntu again and hope all these thing will goes well.....
By the way, thanks for ur help~

SamTew, 2015-05-30:
I forget my id and password, this is my fren's id <----------

wiedemeyer, 2015-06-04:
Try to just install `sudo apt-get install opencl-headers fglrx`.

closed 2015-06-12 by wiedemeyer