code-iai/iai_kinect2 #97: [News] iai_kinect2/libfreenect2 at ICRA 2015

xlz, 2015-05-27:
Three teams at [Amazon Picking Challenge](http://amazonpickingchallenge.org/teams.shtml) on the first day used iai_kinect2/libfreenect2 for their manipulation tasks.

JSK, University of Tokyo reported overall success with less than ideal performance of pointcloud framerate being 10Hz with a GTX 780.
[image]

MIT reported flawless success with a GTX Titan X and dual Kinect v2.
[image]

Alten the Netherlands had linking issues with the binary ROS package and could not enable GPU processing. But they were able to work it through using CPU only processing.
[image]

PR2 maintainer Clearpath Robotics also sets to integrate Kinect v2 in its indigo upgrade kit for PR2. They will have two people working on contributing to the driver.

wiedemeyer, 2015-05-27:
Thanks for sharing. It's really nice to see that people from all over the world are using this software. Also interesting to see how much the individual setups differ.

lakehanne, 2015-05-28:
Uber cool! Congrats Thiemo!

closed 2015-10-26 by wiedemeyer