code-iai/iai_kinect2 #98: [Poll] Topic structure

wiedemeyer, 2015-05-27:
I would to change the topic names to be more ROS conform and to reduce the total number of advertised topics, but before I would like to ask you for your opinions and remarks on this.

Right now each image has its own base topic, for example the rectified color image is `\kinect2\rgc_rect\image` and `\kinect2\rgc_rect\camera_info`. But the camera_info for all high res images is the same.

To reduce the number of advertised topics, the color/mono images can be grouped by the sensor like this:
```
/kinect2/rgb/camera_info
/kinect2/rgb/image
/kinect2/rgb/image_rect
/kinect2/rgb/image_color
/kinect2/rgb/image_color_rect
```
But then another topic for the registered depth image is needed and it is inconsistent to publish the monochrome image under a base topic named "rgb".

Another idea would be to group the images based on the resolution, for example `highres` or `hd` and also advertise the registered depth image in there:
```
/kinect2/hd/camera_info
/kinect2/hd/image
/kinect2/hd/image_rect
/kinect2/hd/image_color
/kinect2/hd/image_color_rect
/kinect2/hd/image_depth_rect
```
The same can be done for `lowres` or `qhd` (quater hd) or `sd`.
```
/kinect2/qhd/camera_info
/kinect2/qhd/image
/kinect2/qhd/image_rect
/kinect2/qhd/image_color
/kinect2/qhd/image_color_rect
/kinect2/qhd/image_depth_rect
```

`ir` and `depth` could be also grouped together:
```
/kinect2/ir/camera_info
/kinect2/ir/image
/kinect2/ir/image_rect
/kinect2/ir/image_depth
/kinect2/ir/image_depth_rect
```

This would reduce the advertised topics (excluding compressed topics) from 24 to 17. This also clarifies which depth image belongs to which other image.

So what do you think of this structure and which base names should these groups have?

Tabjones, 2015-05-28:
I would go for the second layout, it looks clearer to me, but i would add a further distinction based on sensors. For example:
`/kinect2/rgb/hd`
`/kinect2/d/hd`
First one would include all topics that originate from the rgb sensor, so for example mono images and rgb images
Second one would include all topics from the infrared sensor: ir images and depth images.
This way all the topics in these groups should have the same camera info, i think.
Same goes for low res:
`/kinect2/rgb/qhd`
`/kinect2/d/qhd`

As of the point clouds i would group them separated like this:
`/kinect2/cloud/hd`
`/kinect2/cloud/qhd`
or something like that, since they don't actually generate from one single sensor but from both.

cheers

wiedemeyer, 2015-05-28:
So you suggest a 3 level topic structure, but then I would name the first group for the resolution and the second for the sensor so that it is clear that the camera_info is valid for all the sub topics, like:
```
/kinect2/hd/camera_info
/kinect2/hd/rgb/image
/kinect2/hd/rgb/image_rect
/kinect2/hd/rgb/image_color
/kinect2/hd/rgb/image_color_rect
/kinect2/hd/depth/image_depth_rect
/kinect2/hd/cloud/points
```
But I think this extra level is obsolete because the topic names include the information, `.../image` should be the mono image `...\image_color` should be the color image, `.../points` should give point clouds, etc.

To make it more clear for the mono image we could name it `.../image_mono`, then it would be similar to [image_proc](http://wiki.ros.org/image_proc).
```
/kinect2/hd/camera_info
/kinect2/hd/image_mono
/kinect2/hd/image_mono_rect
/kinect2/hd/image_color
/kinect2/hd/image_color_rect
/kinect2/hd/image_depth_rect
/kinect2/hd/points
```

Tabjones, 2015-05-28:
Ok, i like it, your solution seems more elegant.

But what about IR ? 
`/kinect2/hd/image_ir`
`/kinect2/hd/image_ir_rect`
grouped with the rest of the hd topics ?

Aren't they under a different `../camera_info` ?

Maybe you could introduce different camera info topics in the same grouping, like:
`/kinect2/hd/rgb_camera_info`
`/kinect2/hd/ir_camera_info`
but then it is unclear which topic is originated from which camera info...

Umh i don't know, maybe leave a different subgroup for ir alone ? (like it is now, but with rect and raw togheter)

wiedemeyer, 2015-05-28:
There are no hd ir images, only depth.

wiedemeyer, 2015-05-28:
I commited a version to the devel branch where the topics are changed, so everyone could try it out. The structure is like this:
```
/kinect2/bond
/kinect2/hd/camera_info
/kinect2/hd/image_color
/kinect2/hd/image_color/compressed
/kinect2/hd/image_color_rect
/kinect2/hd/image_color_rect/compressed
/kinect2/hd/image_depth_rect
/kinect2/hd/image_depth_rect/compressed
/kinect2/hd/image_mono
/kinect2/hd/image_mono/compressed
/kinect2/hd/image_mono_rect
/kinect2/hd/image_mono_rect/compressed
/kinect2/hd/points
/kinect2/ir/camera_info
/kinect2/ir/image_depth
/kinect2/ir/image_depth/compressed
/kinect2/ir/image_depth_rect
/kinect2/ir/image_depth_rect/compressed
/kinect2/ir/image_ir
/kinect2/ir/image_ir/compressed
/kinect2/ir/image_ir_rect
/kinect2/ir/image_ir_rect/compressed
/kinect2/qhd/camera_info
/kinect2/qhd/image_color
/kinect2/qhd/image_color/compressed
/kinect2/qhd/image_color_rect
/kinect2/qhd/image_color_rect/compressed
/kinect2/qhd/image_depth_rect
/kinect2/qhd/image_depth_rect/compressed
/kinect2/qhd/image_mono
/kinect2/qhd/image_mono/compressed
/kinect2/qhd/image_mono_rect
/kinect2/qhd/image_mono_rect/compressed
/kinect2/qhd/points
```

Tabjones, 2015-05-28:
Good! Looks neat

wiedemeyer, 2015-06-01:
I pushed those changes to the master, because no other complains or remarks where posted.

closed 2015-06-01 by wiedemeyer

BHAASKARR, 2017-03-10:
@wiedemeyer  
I would like to see the depth values of picture (Distance data from camera to image).
so which of the following will give distance data from camera to image
/kinect2/sd/image_depth
/kinect2/sd/image_depth/compressed
/kinect2/sd/image_depth_rect
/kinect2/sd/image_depth_rect/compressed
/kinect2/sd/image_ir
/kinect2/sd/image_ir/compressed
/kinect2/sd/image_ir_rect
/kinect2/sd/image_ir_rect/compressed

and where can i see the source code for above files ?
what represents each value of below collected data row (EX: data:[73,73,42,0,76,235,2,0,128,0,0,11,......etc..]) ?

// Below data is from the /kinect2/sd/image_depth_rect/compressed topic //
header: 
  seq: 2437
  stamp: 
    secs: 1489174612
    nsecs: 963329202
  frame_id: kinect2_ir_optical_frame
format: 16UC1; tiff compressed
data: [73, 73, 42, 0, 76, 235, 2, 0, 128, 0, 0, 11, 80, 65, 45, 238, 241, 4, 3, 222, 240, 40, 100, 54, 29, 13, 7, 2, 29, 79, 240, 67, 225, 240, 7, 138, 195, 227, 80, 198, 56, 28, 114, 248, 142, 199, 227, 114, 56, 115, 8, 14, 3, 0, 1, 128, 3, 87, 195, 68, 14, 20, 0, 6, 223, 17, 183, 104, 28, 72, 0, 109, 191, 221, 207, 240, 112, 1, 254, 255, 16, 0, 39, 96, 16, 0, 124, 0, 5, 0, 57, 31, 228, 80, 5, 17, 240, 255, 12, 0, 4, 64, 6, 243, 254, 166, 19, 0, 74, 7, 64, 0, 64, 1, 210, 255, 89, 63, 221, 143, 240, 120, 1, 57, 11, 179, 2, 12, 143, 116, 224, 32, 34, 0, 125, 63, 203, 47, 113, 184, 36, 24, 0, 94, 189, 137, 192, 144, 224, 1, 214, 255, 114, 191, 233, 5, 240, .....................some values .......... 0, 238, 138, 2, 0, 209, 155, 2, 0, 242, 171, 2, 0, 75, 188, 2, 0, 169, 203, 2, 0, 39, 219, 2, 0]
---